gbionics/bipedal-locomotion-framework #347: TimeVaryingDCMPlannerUnitTests segs fault on macOS

GiulioRomualdi, 2021-06-22:
I noticed that `TimeVaryingDCMPlannerUnitTests` fails on macOS with a segmentation fault

 https://github.com/dic-iit/bipedal-locomotion-framework/pull/346/checks?check_run_id=2884619713#step:20:62

GiulioRomualdi, 2021-06-22:
I tried to replicate the error with https://github.com/GiulioRomualdi/blf-fastmac but I'm not able to replicate the error 

@traversaro

GiulioRomualdi, 2021-06-23:
Issue investigated in https://github.com/dic-iit/bipedal-locomotion-framework/pull/348

GiulioRomualdi, 2021-06-23:
Since #348 is not failing. I think that this bug can be solved invalidating the cache and recompile the dependencies in macOS

closed 2021-06-23 by GiulioRomualdi

traversaro, 2021-06-26:
Just to speculate on what happened, I think that the problem is that ABI changes in Homebrew-provided dependencies are not accounted for in the cache, so it is possible that the Homebrew dependencies ABI changed in a new build job, but the cache was not invalidated. To be honest, I am not sure what is the simple solution to account for that. In theory, we should include the versions of all homebrew packages used by the cached dependencies in the cache key, but that probably would result in many more rebuilt. Probably for now can just manually invalidate the cache whenever this happens.

PR #349 (2021-06-23): Invalidate the cache in GitHub CI
This fixes #347