gbionics/bipedal-locomotion-framework #350: Fail if a component is explicitly enabled but CMake dependencies are not found

diegoferigo, 2021-06-29:
Currently, configuring the project enabling explicitly a component, e.g. with `-DFRAMEWORK_COMPILE_Contact:BOOL=ON`, does not fail if the component dependencies are not met. In fact, the component is silently disabled. In my opinion, it would be better if the configuration would fail since the user explicitly asked the component to be built.

Currently, the project fails only if another component is explicitly enabled and links again a target of the missing component:

```
2021-06-29T09:02:55.8878878Z   CMake Error at cmake/AddBipedalLocomotionLibrary.cmake:67 (add_library):
2021-06-29T09:02:55.8880033Z     Target "Planners" links to target "BipedalLocomotion::Contacts" but the
2021-06-29T09:02:55.8880939Z     target was not found.  Perhaps a find_package() call is missing for an
2021-06-29T09:02:55.8881679Z     IMPORTED target, or an ALIAS target is missing?
2021-06-29T09:02:55.8882309Z   Call Stack (most recent call first):
2021-06-29T09:02:55.8883038Z     src/Planners/CMakeLists.txt:9 (add_bipedal_locomotion_library)
```

cc @GiulioRomualdi @S-Dafarra

S-Dafarra, 2021-06-29:
The logic in this case is https://github.com/dic-iit/bipedal-locomotion-framework/blob/74c9bae880b1118b0a98b3cbbb136a08e648f9b9/cmake/BipedalLocomotionFrameworkFindDependencies.cmake#L77-L126

I think that somehow we can add a check if the value is set by the user, but the dependencies are not satisfied. In that case, it should print an error.

Personal opinion, the part in https://github.com/dic-iit/bipedal-locomotion-framework/blob/74c9bae880b1118b0a98b3cbbb136a08e648f9b9/cmake/BipedalLocomotionFrameworkFindDependencies.cmake#L110-L123, drove me crazy a couple of times :sweat_smile:

diegoferigo, 2021-06-29:
Thanks for the details @S-Dafarra. Unfortunately, this issue covers one of curses of CMake: functions / macros like this one with regexp and similar, could be fully understood only from the original author :sweat_smile: It seems another alphabet to me at first look.

> In that case, it should print an error.

To be clear, it's not just printing an error and continuin, I meant that the `cmake` call should return raising a `FATAL_ERROR` with a proper message.

diegoferigo, 2021-06-29:
For instance, check this output and try to figure out what's missing (good luck xD)

```
2021-06-29T09:22:49.2883354Z   -- Created target ContinuousDynamicalSystem for export BipedalLocomotionFramework.
2021-06-29T09:22:49.2900425Z   -- Created target TextLogging for export BipedalLocomotionFramework.
2021-06-29T09:22:49.2926789Z   -- Created target GenericContainer for export BipedalLocomotionFramework.
2021-06-29T09:22:49.2956331Z   -- Created target YarpUtilities for export BipedalLocomotionFramework.
2021-06-29T09:22:49.3013346Z   -- Created target ParametersHandlerYarpImplementation for export BipedalLocomotionFramework.
2021-06-29T09:22:49.3030691Z   -- Created target ParametersHandler for export BipedalLocomotionFramework.
2021-06-29T09:22:49.3055881Z   -- Created target CommonConversions for export BipedalLocomotionFramework.
2021-06-29T09:22:49.3062315Z   -- Created target ManifConversions for export BipedalLocomotionFramework.
2021-06-29T09:22:49.3103662Z   -- Created target Estimators for export BipedalLocomotionFramework.
2021-06-29T09:22:49.3128408Z   -- Created target FloatingBaseEstimators for export BipedalLocomotionFramework.
2021-06-29T09:22:49.3178144Z   -- Created target SystemYarpImplementation for export BipedalLocomotionFramework.
2021-06-29T09:22:49.3184582Z   -- Created target System for export BipedalLocomotionFramework.
2021-06-29T09:22:49.3207423Z   -- Created target Planners for export BipedalLocomotionFramework.
2021-06-29T09:22:49.3246633Z   -- Created target ContactModels for export BipedalLocomotionFramework.
2021-06-29T09:22:49.3292045Z   -- Created target RobotInterfaceYarpImplementation for export BipedalLocomotionFramework.
2021-06-29T09:22:49.3318365Z   -- Created target RobotInterface for export BipedalLocomotionFramework.
2021-06-29T09:22:49.3349498Z   -- Created target Math for export BipedalLocomotionFramework.
2021-06-29T09:22:49.3425327Z   --  [ ] Plugin: FloatingBaseEstimatorDevice (ENABLE_FloatingBaseEstimatorDevice)
2021-06-29T09:22:49.3536340Z   -- Generating the Urdf Model from generate_stairs_urdf
2021-06-29T09:22:49.3610178Z   -- Added files for bindings named System in BipedalLocomotionFramework.
2021-06-29T09:22:49.3622262Z   -- Added files for bindings named ParametersHandlerBindings in BipedalLocomotionFramework.
2021-06-29T09:22:49.3629291Z   -- Added files for bindings named ParametersHandlerYarpImplementationBindings in BipedalLocomotionFramework.
2021-06-29T09:22:49.3649209Z   -- Added files for bindings named PlannersBindings in BipedalLocomotionFramework.
2021-06-29T09:22:49.3660736Z   -- Added files for bindings named RobotInterfaceBindings in BipedalLocomotionFramework.
2021-06-29T09:22:49.3672320Z   -- Added files for bindings named MathBindings in BipedalLocomotionFramework.
2021-06-29T09:22:49.3685454Z   -- Added files for bindings named FloatingBaseEstimatorsBindings in BipedalLocomotionFramework.
2021-06-29T09:22:49.3750304Z   -- Created target application joint-position-tracking.
2021-06-29T09:22:49.3857370Z   -- Configuring done
2021-06-29T09:22:49.4332093Z   CMake Error at cmake/FindPython/Support.cmake:3128 (add_library):
2021-06-29T09:22:50.2046606Z     Target "pybind11_blf" links to target "BipedalLocomotion::Contacts" but the
2021-06-29T09:22:50.2054644Z     target was not found.  Perhaps a find_package() call is missing for an
2021-06-29T09:22:50.2057616Z     IMPORTED target, or an ALIAS target is missing?
2021-06-29T09:22:50.2059793Z   Call Stack (most recent call first):
2021-06-29T09:22:50.2063025Z     cmake/FindPython3.cmake:393 (__Python3_add_library)
2021-06-29T09:22:50.2066242Z     /tmp/pip-build-env-qyt0gj6o/overlay/lib/python3.7/site-packages/pybind11/share/cmake/pybind11/pybind11NewTools.cmake:196 (python3_add_library)
2021-06-29T09:22:50.2069503Z     bindings/python/CMakeLists.txt:26 (pybind11_add_module)
2021-06-29T09:22:50.2070455Z 
2021-06-29T09:22:50.2070936Z 
2021-06-29T09:22:50.2071668Z   CMake Error at cmake/FindPython/Support.cmake:3128 (add_library):
2021-06-29T09:22:50.2072794Z     Target "pybind11_blf" links to target "BipedalLocomotion::Contacts" but the
2021-06-29T09:22:50.2076048Z     target was not found.  Perhaps a find_package() call is missing for an
2021-06-29T09:22:50.2076996Z     IMPORTED target, or an ALIAS target is missing?
2021-06-29T09:22:50.2077743Z   Call Stack (most recent call first):
2021-06-29T09:22:50.2078891Z     cmake/FindPython3.cmake:393 (__Python3_add_library)
2021-06-29T09:22:50.2080748Z     /tmp/pip-build-env-qyt0gj6o/overlay/lib/python3.7/site-packages/pybind11/share/cmake/pybind11/pybind11NewTools.cmake:196 (python3_add_library)
2021-06-29T09:22:50.2082494Z     bindings/python/CMakeLists.txt:26 (pybind11_add_module)
2021-06-29T09:22:50.2083245Z 
2021-06-29T09:22:50.2083670Z 
2021-06-29T09:22:50.2084589Z   CMake Error at cmake/AddBipedalLocomotionLibrary.cmake:67 (add_library):
2021-06-29T09:22:50.2085994Z     Target "FloatingBaseEstimators" links to target
2021-06-29T09:22:50.2087006Z     "BipedalLocomotion::Contacts" but the target was not found.  Perhaps a
2021-06-29T09:22:50.2088022Z     find_package() call is missing for an IMPORTED target, or an ALIAS target
2021-06-29T09:22:50.2088769Z     is missing?
2021-06-29T09:22:50.2089375Z   Call Stack (most recent call first):
2021-06-29T09:22:50.2090195Z     src/Estimators/CMakeLists.txt:17 (add_bipedal_locomotion_library)
2021-06-29T09:22:50.2091369Z 
2021-06-29T09:22:50.2092437Z 
[... cut ...]
2021-06-29T09:22:50.2104110Z     find_package() call is missing for an IMPORTED target, or an ALIAS target
2021-06-29T09:22:50.2104860Z     is missing?
2021-06-29T09:22:50.2105462Z   Call Stack (most recent call first):
2021-06-29T09:22:50.2106842Z     utilities/joint-position-tracking/CMakeLists.txt:7 (add_bipedal_locomotion_application)
2021-06-29T09:22:50.2108279Z 
2021-06-29T09:22:50.2108560Z 
2021-06-29T09:22:50.2109213Z   -- Generating done
2021-06-29T09:22:50.2109966Z   CMake Generate step failed.  Build files cannot be regenerated correctly.
```

S-Dafarra, 2021-06-29:
> Thanks for the details @S-Dafarra. Unfortunately, this issue covers one of curses of CMake: functions / macros like this one with regexp and similar, could be fully understood only from the original author  It seems another alphabet to me at first look.

It had been copied from https://github.com/robotology/yarp/blob/636c0abe43166b56d002c963b9bb5a09a74540e8/cmake/YarpFindDependencies.cmake#L424-L473

AFAIK, feel free to try to simplify it.

> To be clear, it's not just printing an error and continuin, I meant that the `cmake` call should return raising a `FATAL_ERROR` with a proper message.

I agree