gbionics/bipedal-locomotion-framework #351: Avoid to keep the robot urdf model in the test folders

GiulioRomualdi, 2021-07-01:
I noticed that in the floating base estimator testing folder we currently store the model.urdf of a robot (I assume iCub) 

[image]

I tried to remove it and I implemented a simple logic that uses fetch content. However, I noticed that  the model is different from the one stored in icub-models https://github.com/robotology/icub-models/releases/tag/v1.20.0

In particular, the link named `root_link_imu_frame` is missing in icub-models urdf. 

<details><summary>Click me if you are interested in the difference between the current model stored in blf and model of the iCubGazeboV2_5 stored in icub-models</summary>
<p>

```diff
4c4
<       <origin xyz="0.0248821 0.000103947 -0.0440806" rpy="1.57079632679 0 -1.57079632679"/>
---
>       <origin xyz="0.02488209999999999 0.00010394699999999017 -0.04408060000000001" rpy="1.5707963267948961 0 -1.5707963267948961"/>
9c9
<       <origin xyz="-0.0364 0 0.032" rpy="1.57079632679 0 -1.57079632679"/>
---
>       <origin xyz="-0.036399999999999995 0 0.032" rpy="1.5707963267948961 0 -1.5707963267948961"/>
18c18
<       <origin xyz="-0.0364 0 0.032" rpy="1.57079632679 0 -1.57079632679"/>
---
>       <origin xyz="-0.036399999999999995 0 0.032" rpy="1.5707963267948961 0 -1.5707963267948961"/>
24,30d23
<   <link name="root_link_imu_frame"/>
<   <joint name="root_link_imu_frame_fixed_joint" type="fixed">
<     <origin xyz="0.085155 -0.011 -0.112309" rpy="-2.09439521059 0 -1.57079632679"/>
<     <parent link="root_link"/>
<     <child link="root_link_imu_frame"/>
<     <dynamics damping="0.1"/>
<   </joint>
33c26
<       <origin xyz="-0.0005145 -0.0380753 -7.1e-05" rpy="1.57079632679 0 1.57079632679"/>
---
>       <origin xyz="-0.0005145000000000149 -0.038075299999999986 -7.099999999998774e-05" rpy="1.5707963267948961 0 1.5707963267948961"/>
38c31
<       <origin xyz="0.0428515000001 0.026 0.151913" rpy="1.57079632679 0 1.57079632679"/>
---
>       <origin xyz="0.04285150000011213 0.025999999999999232 0.1519129999999679" rpy="1.5707963267948961 0 1.5707963267948961"/>
47c40
<       <origin xyz="0.0428515000001 0.026 0.151913" rpy="1.57079632679 0 1.57079632679"/>
---
>       <origin xyz="0.04285150000011213 0.025999999999999232 0.1519129999999679" rpy="1.5707963267948961 0 1.5707963267948961"/>
54,55c47,48
<     <origin xyz="0.0064515 0.026 -0.119913" rpy="0 0 -3.14159265359"/>
<     <axis xyz="-2.22044604925e-16 -1.0 0.0"/>
---
>     <origin xyz="0.006451499999999992 0.025999999999999968 -0.11991299999999999" rpy="0 0 -3.141592653589793"/>
>     <axis xyz="-2.220446049250313e-16 -0.9999999999999998 0.0"/>
58,59c51,52
<     <limit effort="55.5" lower="-0.785398163397" upper="2.33874119767" velocity="5.1"/>
<     <dynamics damping="0.223" friction="0.0"/>
---
>     <limit effort="55.5" lower="-0.7853981633974483" upper="2.3387411976724017" velocity="5.1"/>
>     <dynamics damping="1.0" friction="0.0"/>
63,65c56,58
<       <origin xyz="-0.0013754 3.49999999998e-06 -0.044282" rpy="1.57079632679 0 1.57079632679"/>
<       <mass value="0.418361"/>
<       <inertia ixx="0.000487866" ixy="-2.70231e-07" ixz="1.85338e-07" iyy="0.000286099" iyz="1.22793e-05" izz="0.000315174"/>
---
>       <origin xyz="-0.0009315000000000018 8.09999999998312e-06 -0.04480899999999996" rpy="1.5707963267948961 0 1.5707963267948961"/>
>       <mass value="0.413242"/>
>       <inertia ixx="0.00046914" ixy="-4.76597e-08" ixz="7.12496e-08" iyy="0.000277108" iyz="3.3401e-06" izz="0.000303891"/>
68c61
<       <origin xyz="0.0428514998094 0.0701 0.151913" rpy="1.57079632679 0 1.57079632679"/>
---
>       <origin xyz="0.04285149980937101 0.07010000000000424 0.15191299999996632" rpy="1.5707963267948961 0 1.5707963267948961"/>
77c70
<       <origin xyz="0.0428514998094 0.0701 0.151913" rpy="1.57079632679 0 1.57079632679"/>
[... 2988 lines cut ...]
>       <update_rate>100</update_rate>
>       <pose>0.0040001999999999885 -0.008600000000000003 -0.045282000000000044 -2.220446049250314e-16 2.220446049250313e-16 1.5707963267948966</pose>
>     </sensor>
>   </gazebo>
>   <sensor name="l_foot_ft_acc_3b13" type="accelerometer">
>     <parent link="l_ankle_2"/>
>     <origin rpy="-2.220446049250314e-16 2.220446049250313e-16 1.5707963267948966" xyz="0.0040001999999999885 -0.008600000000000003 -0.045282000000000044"/>
>   </sensor>
>   <gazebo reference="l_ankle_2">
>     <sensor name="l_foot_ft_gyro_3b13" type="imu">
>       <always_on>1</always_on>
>       <update_rate>100</update_rate>
>       <pose>0.0040001999999999885 -0.008600000000000003 -0.045282000000000044 -2.220446049250314e-16 2.220446049250313e-16 1.5707963267948966</pose>
>     </sensor>
>   </gazebo>
>   <sensor name="l_foot_ft_gyro_3b13" type="gyroscope">
>     <parent link="l_ankle_2"/>
>     <origin rpy="-2.220446049250314e-16 2.220446049250313e-16 1.5707963267948966" xyz="0.0040001999999999885 -0.008600000000000003 -0.045282000000000044"/>
>   </sensor>
```

</p>
</details>

cc @prashanthr05

prashanthr05, 2021-07-02:
> In particular, the link named `root_link_imu_frame` is missing in icub-models urdf.

The relevant IMU frame corresponding to `iCubGenova04` should be `root_link_imu_acc`. In the tests, we are looking for this frame https://github.com/dic-iit/bipedal-locomotion-framework/blob/74c9bae880b1118b0a98b3cbbb136a08e648f9b9/src/Estimators/tests/FloatingBaseEstimators/InvariantEKFBaseEstimatorTest.cpp#L132.

And this is a custom frame added specific to the `iCubGenova04` robot. So an upstream modification is difficult.

traversaro, 2021-07-02:
You can also just add any additional frame on the top of a model that you load from URDF via [`iDynTree::addAdditionalFrameToLink`](https://robotology.github.io/idyntree/master/classiDynTree_1_1Model.html#ab67ec3bb2074412c1002befff50e2f57). As the link frame of `root_link` is explicitly defined, this should be stable even if the icub model is re-generated.

GiulioRomualdi, 2023-06-26:
All the `urdf` files have been removed

closed 2023-06-26 by GiulioRomualdi