gbionics/bipedal-locomotion-framework #353: Conda CI fails in ubuntu

GiulioRomualdi, 2021-07-02:
This is the error
```
CasADi - 2021-07-02 12:18:49 WARNING(".../casadi/core/plugin_interface.hpp:292: Assertion "handle!=nullptr" failed:
PluginInterface::load_plugin: Cannot load shared library 'libcasadi_nlpsol_ipopt.so': 
   (
    Searched directories: 1. casadipath from GlobalOptions
                          2. CASADIPATH env var
                          3. PATH env var (Windows)
                          4. LD_LIBRARY_PATH env var (Linux)
                          5. DYLD_LIBRARY_PATH env var (osx)
    A library may be 'not found' even if the file exists:
          * library is not compatible (different compiler/bitness)
          * the dependencies are not found
   )
  Tried '' :
    Error code: libipopt.so.1: cannot open shared object file: No such file or directory
  Tried '.' :
    Error code: ./libcasadi_nlpsol_ipopt.so: cannot open shared object file: No such file or directory") [.../casadi/core/plugin_interface.hpp:148]
Error: 7-02 12:18:49.232] [thread: 7630] [blf] [error] [TimeVaryingDCMPlanner::initialize] Unable to solve the optimization problem. The following exception has been thrown by the solver: Error in Opti::solve [OptiNode] at .../casadi/core/optistack.cpp:159:
.../casadi/core/plugin_interface.hpp:387: Plugin 'ipopt' is not found..
```

https://github.com/dic-iit/bipedal-locomotion-framework/pull/352/checks?check_run_id=2972139384#step:10:53

There seems that `casadi` pakage doesn't provide anymore the ipopt plugin. cc @traversaro

traversaro, 2021-07-02:
From https://github.com/dic-iit/bipedal-locomotion-framework/actions/workflows/conda-forge-ci.yml?query=event%3Aschedule it seems that it started failing ~4 days ago.

traversaro, 2021-07-02:
I quickly compare the last good build (left) and the first failed build (right), and this is the diff:
[image]

What happend is that ipopt 3.14 was released (https://github.com/conda-forge/ipopt-feedstock/pull/63, not sure why it was only picked 4 days ago), and ipopt 3.14 is not ABI compatible with 3.13 that is the version against which the latest casadi build was compiled, so the conda solver needs either to keep ipopt to 3.13 and use the latest casadi, or to use the latest ipopt 3.14, and find a version of casadi that does not depends on ipopt, that apparently it is the case for the old casadi 3.4.5 .

This problems are usually solved at the conda-forge level by automatically having ABI migrations, to rebuild downstream projects whenever an ABI-incompatible library is released (see https://conda-forge.org/status/#other_migrations for all the ongoing migrations). However, ABI migration need to be explicitly enabled (see https://github.com/conda-forge/conda-forge-pinning-feedstock/pull/1606 where we recently added a migrator for Tinyxml2 after a similar problem) and apparently no one ever set it up for the ipopt library, even because it has a few downstream ports that depend on it.

GiulioRomualdi, 2021-07-02:
Fixed by https://github.com/dic-iit/bipedal-locomotion-framework/pull/354

closed 2021-07-02 by GiulioRomualdi