gbionics/bipedal-locomotion-framework #358: Modify the `calibration-delta-updater` to consider hard limits

GiulioRomualdi, 2021-07-07:
The[`calibration-delta-updater` ](https://github.com/dic-iit/bipedal-locomotion-framework/tree/master/utilities/calibration-delta-updater) is a really useful tool to speed up the calibration procedure.

The application is based on the assumption that the robot has to be manually moved in the zero configuration. Discussing with @pattacini and @S-Dafarra., we realized that a possible solution is to use the hard limits on the robot. 

Considering that we can modify the  `calibration-delta-updater` to move the robot towards the hardware limit, automatically compute the offset and finally automatically update the configuration file. To reach the hardware limit we can use the PWM interface (already supported in blf #346) and then we can use the `ISensorBridge` to read the joint encoder. 

cc @traversaro @isorrentino @DanielePucci

GiulioRomualdi, 2021-07-07:
cc @dic-iit/blf-developers

S-Dafarra, 2021-07-07:
Rather than considering it a special case, a possibility could be to set the individual "expected" values. Those could be all zeros, or the expected encoder measurements when at the limits

GiulioRomualdi, 2021-07-07:
> Rather than considering it a special case, a possibility could be to set the individual "expected" values. Those could be all zeros, or the expected encoder measurements when at the limits. I would avoid having an automated way to reach those limits. I think it is safer if a person reaches that configuration by hand.

In that case is super trivial to implement

S-Dafarra, 2021-07-07:
Using the open loop control may make sense for a subset of joints though, like the ankles or the knee. In any case, I think it is not trivial to understand when a joint has actually hit the limit, or instead it is blocked by gravity for example. Also, I am not sure about how to define the signed value of PWM to apply 🤔

GiulioRomualdi, 2021-07-07:
In #361 I implemented what you suggested in https://github.com/dic-iit/bipedal-locomotion-framework/issues/358#issuecomment-875637690

closed 2021-07-09 by GiulioRomualdi