gbionics/bipedal-locomotion-framework #360: Add the possibility to set exogenous feedback in IK component

GiulioRomualdi, 2021-07-07:
A long long time ago @DanielePucci @S-Dafarra and I had a discussion about the possibility to use the IMU on the feet in the IK instead of using the computation done with iDynTree. 
Given the current infrastructure the feedback (e.g. is taken from KinDynComputation) however we may modify the SE3Task adding the possibility to take an exogenous feedback. What follows is a possible approach that we can use (@S-Dafarra @traversaro @prashanthr05 @isorrentino and @paolo-viceconte please comment if you have other ideas):
- [ ] Add an optional parameter for the `SE3Task` called `use_exogenous_feedback` (Default value `false`). If true the user must set the feedback. Otherwise, the value taken from iDynTree will be used.
- [ ] implement  `bool setExogenousFeedback(const manif::SE3d& feedback);`

traversaro, 2021-07-07:
> Otherwise, the value taken from iDynTree will be used.

I probably asked this already 200 times, but the value taken by iDynTree but using which joint values? The one read from the robot or the one obtaining by integrating the IK solution?

GiulioRomualdi, 2021-07-07:
On the robot, we use the solution of the IK integrated

closed 2022-09-22 by GiulioRomualdi