gbionics/bipedal-locomotion-framework #363: Restructure TSID and IK

GiulioRomualdi, 2021-07-09:
I started to implement QPTSID class and I realized that the implementation is really similar to `QPIntegrationBaseIK` already implemented in the framework. Not only for the interface but also for the actual code. 

So I was thinking to merge TSID and IK together. 

Here it's what I've in mind (please help me to find better names 😄 )
1. `IFloatingBaseSolver` is a virtual class it contains only the common interface. An example of the interface can be found [here](https://github.com/dic-iit/bipedal-locomotion-framework/blob/71e37a822c9c7aa1bf23a91eeabceb6f66ba3165/src/TSID/include/BipedalLocomotion/TSID/TaskSpaceInverseDynamics.h#L50)
2. `QPFloatingBase` implements the floating base algorithm. The implementation is exactly the same for IK and TSID. The real job is done by the task definitions
3. `QPIntegrationBaseIK` and `QPTSID` are the template specialization of the `QPFloatingBase`
4. `QPFixedBaseTSID` is just a specialization  `QPTSID`. 

[image]

@dic-iit/blf-developers, do you have any suggestions?

traversaro, 2021-07-09:
Why the **floating base** in  `IFloatingBaseSolver` and `QPFloatingBase` ? What they have of floating base specific?

GiulioRomualdi, 2021-07-09:
actually nothing. Is the task itself that is floating base specific

isorrentino, 2021-07-12:
@GiulioRomualdi I like the idea!

S-Dafarra, 2021-07-12:
I guess that instead of ``FloatingBase``, a more adequate keyword could be ``StackOfTasks``

GiulioRomualdi, 2021-07-12:
@traversaro suggested me ILinearTaskSolver.

Furthermore while implementing I realised that keeping the same implementation  for TSID and IK may not be a wise choice (at least at che current stage).

In this morning I will open a PR where I merged the two interfaces and we can discuss better there. (Also for the names)

GiulioRomualdi, 2021-07-12:
For the time being we can consider this task achieved

closed 2021-07-12 by GiulioRomualdi