gbionics/bipedal-locomotion-framework #376: SensorBridge crashes if a data is required and the stream is disabled

GiulioRomualdi, 2021-07-22:
I was with @isorrentino and we noticed that `getMotorsCurrent` segs fault if `stream_motor_state` is not set in the config file. 

The function should return false and not crashes. 
The same logic is applied in all the methods of SensorBridge 

cc @prashanthr05

prashanthr05, 2021-07-22:
Looks like it for all the [control board related measurements](https://github.com/dic-iit/bipedal-locomotion-framework/blob/bbafe597b4a417ea8eea2b64e9eb2f1763736b1a/src/RobotInterface/YarpImplementation/include/BipedalLocomotion/RobotInterface/YarpSensorBridgeImpl.h#L111-L136), while the other sensors, for example a cartesian wrench stream has a [check for valid sensor](https://github.com/dic-iit/bipedal-locomotion-framework/blob/4cb787053aeac00c6a53531823dc0f8e62c9b5f5/src/RobotInterface/YarpImplementation/src/YarpSensorBridge.cpp#L457-L462).

We might have to add a simple check on validating that the relevant configuration flag is turned on, relevant interface (eg., `IEncoders` ) is not a nullptr and the measurement buffer size is non-zero.

GiulioRomualdi, 2021-07-23:
Yep something of that is enough 😄

prashanthr05, 2021-07-23:
I can do it. Will open a PR soon.

prashanthr05, 2021-07-23:
I suggest something like this,

<details>
<summary>Check Method</summary>

``` cpp
template <typename ControlBoardInterface>
    bool checkControlBoardSensor(const std::string logPrefix,
                                 ControlBoardInterface* interface,
                                 const bool& streamConfig,
                                 Eigen::Ref<const Eigen::VectorXd> measureBuffer)
    {
        if (!checkValid(logPrefix))
        {
            return false;
        }

        if (!streamConfig)
        {
            log()->error("{} Configuration flag set to {}."
                         "Please set this flag to true before calling this method.",
                         logPrefix, streamConfig);
            return false;
        }

        if (interface == nullptr)
        {
            log()->error("{} Failed to attach to relevant drivers."
                         "Unable to retrieve measurements.",
                         logPrefix);
        }

        if (measureBuffer.size() == 0)
        {
            log()->error("{} Measurement buffers seem empty."
                         "Unable to retrieve measurements.",
                         logPrefix, streamConfig);
        }

        return true;
    }
```

</details>

<details>
<summary>Use case</summary>

``` cpp
bool YarpSensorBridge::getJointPositions(Eigen::Ref<Eigen::VectorXd> jointPositions,
                                         OptionalDoubleRef receiveTimeInSeconds)
{
    if (!m_pimpl->checkControlBoardSensor("[YarpSensorBridge::getJointPositions]",
                                          m_pimpl->controlBoardRemapperInterfaces.encoders,
                                          m_pimpl->metaData.bridgeOptions.isJointSensorsEnabled,
                                          m_pimpl->controlBoardRemapperMeasures.jointPositions))
    {
        return false;
    }
    jointPositions = m_pimpl->controlBoardRemapperMeasures.jointPositions;
    if (receiveTimeInSeconds)
        receiveTimeInSeconds.value().get()
            = m_pimpl->controlBoardRemapperMeasures.receivedTimeInSeconds;
    return true;
}
```

</details>

Does this look fine?

PR #378 (2021-07-23): YarpSensorBridge: add checks in control board getter methods
Fixes #376
closed 2021-07-23 by GiulioRomualdi