gbionics/bipedal-locomotion-framework #409: Possible bug in `getGroup` in IParameter handler implementations

GiulioRomualdi, 2021-09-03:
Let's create a parameter handler using python (the same is valid in c++)

```python
import bipedal_locomotion_framework.bindings as blf 

# Set the group
handler1 = blf.parameters_handler.StdParametersHandler()                
group1 = blf.parameters_handler.StdParametersHandler()                  
group1.set_parameter_int("param1", 10)                                  
handler1.set_group("GROUP_1", group1)                                   
```

If I call `print(handler1)` I get `GROUP_1` and this is perfectly fine Indeed we the main handler contains only a group.
Using `getGroup` it is possible to retrieve the group. Since the `getGroup` function returns a `std::weak_ptr<IParameterHandler>` it is possible to change the content of the group.
```python
group2 = handler1.get_group("GROUP_1")                                  
group2.set_parameter_int("param2", 105)                                 
```
Calling `print(group2)` I get `param2 param1` and this is perfectly fine since the first parameter has been added at the beginning.

However, since we are working with pointers also the group contained inside `handler1` is modified
```python
group3 = handler1.get_group("GROUP_1")                                 

print(group3)                                                          
param2 param1 
```

In my opinion, this is not wanted and we should avoid this, a possible solution is to change the signature of 
```cpp
    /**
     * Get a Group from the handler.
     * @param name name of the group
     * @return A pointer to IParametersHandler, if the group is not found the weak pointer cannot
     * be locked
     * @warning Please implement the specific version of this method in the Derived class. Please
     * check YarpImplementation::getGroup
     */
    virtual td::weak_ptr<IParameterHandler> getGroup(const std::string& name) const = 0;
```

into
```cpp
    /**
     * Get a Group from the handler.
     * @param name name of the group
     * @return A pointer to IParametersHandler, if the group is not found the weak pointer cannot
     * be locked
     * @warning Please implement the specific version of this method in the Derived class. Please
     * check YarpImplementation::getGroup
     */
    virtual std::weak_ptr<const IParameterHandler> getGroup(const std::string& name) const = 0;
```

@traversaro @S-Dafarra

S-Dafarra, 2021-09-03:
I don't think this is a bug at all. Indeed, it was part of the initial idea described in the (long) comment https://github.com/ami-iit/bipedal-locomotion-framework/issues/20#issuecomment-590813717. The idea is that you can allow methods to fill in a group, without knowing that it is a group. The problem is rather on the bindings that cannot reflex the original interface I suppose.

GiulioRomualdi, 2021-09-03:
The main problem here is that you can get a group with `getGroup` that is a `const` function. Modify the group but since the memory is shared while you modify the group you are actually modifying also the handle that is containing the group. This happens also in c++

for instance the following code compiles (bu it shouldn't)
```cpp
void foo(std::shared_ptr<const ParameterHandler::StdImplementation> handler)
{
    auto group = my_group.getGroup("GROUP").lock();
    group->setParameter("parameter", 12);
}
```
Indeed this changes the content of `handler`. Indeed notice that `group` points to the memory stored in `handler`

S-Dafarra, 2021-09-03:
I see. Then we could have two methods, one const that returns a pointer to a const object, and another without any const.

GiulioRomualdi, 2021-09-03:
Yep. I'm opening a PR so we can discuss later on with the code in front of us 😄

GiulioRomualdi, 2024-04-05:
I closed https://github.com/ami-iit/bipedal-locomotion-framework/pull/410 without merging it. No need to work on this anymore

closed 2024-04-05 by GiulioRomualdi