gbionics/bipedal-locomotion-framework #411: Add the possibility to set the contact phase in  SwingFootPlanner and FixedFootDetector without resetting the internal time

GiulioRomualdi, 2021-09-06:
In the fixed foot detector and in SwingFootPlanner the initial time is reset every time a new set of contact is passed to the classes

https://github.com/ami-iit/bipedal-locomotion-framework/blob/252e9308b91f034bafc90388755336a447002801/src/Planners/src/SwingFootPlanner.cpp#L126-L129

https://github.com/ami-iit/bipedal-locomotion-framework/blob/252e9308b91f034bafc90388755336a447002801/src/Contacts/src/FixedFootDetector.cpp#L202-L203

This is actually a problem in the case you want to change only the position of a contact in the contact list and reuse the very same instance of the class. I know that this can be really dangerous if someone does not know what is actually doing. 

Since I need this feature for enabling the step recovery for the CentroidalMPC I will try to implement something in a quick and dirty manner. Then I think we should discuss some more elegant approach with @S-Dafarra

GiulioRomualdi, 2021-09-06:
By the way, probably the `setContactPhase` functions should not reset the internal clock of the advanceable. I think that the initial clock should be set by an ad-hoc method. 

cc @traversaro

S-Dafarra, 2021-09-08:
As discussed quickly F2F, I guess that we can have two different methods, one that initializes everything and does all the resets (like ``setInitialContactList``), and another to update only specific quantities (like ``updateContactList``).

GiulioRomualdi, 2023-06-26:
This has been faced thanks to several PR I opened:
- **SwingFootPlanner** https://github.com/ami-iit/bipedal-locomotion-framework/pull/637 and https://github.com/ami-iit/bipedal-locomotion-framework/pull/688
   
   https://github.com/ami-iit/bipedal-locomotion-framework/blob/ba95103f95745ae604ff14dc4d191cd07e18ca6e/src/Planners/src/SwingFootPlanner.cpp#L262-L265

- **Fixed foot detector** https://github.com/ami-iit/bipedal-locomotion-framework/pull/624
  
   https://github.com/ami-iit/bipedal-locomotion-framework/blob/747fb929fb082576eb5916fc60baf086962b6ec1/src/Contacts/src/FixedFootDetector.cpp#L246-L249

Indeed both the classes have a function to reset the time

closed 2023-06-26 by GiulioRomualdi