gbionics/bipedal-locomotion-framework #412: Possible problems while using `getPresentContact()`

GiulioRomualdi, 2021-09-11:
I was using the `getPresentContact` method to get the active contact at a given time. 

```cpp
const double nextPlannedContactTime = m_pimpl->currentTime + m_pimpl->optiSettings.samplingTime * index;
auto nextPlannedContact = m_pimpl->contactPhaseList.lists().at(key).getPresentContact(nextPlannedContactTime);
```

I noticed that given some numerical error the output of the `getPresentContact` was wrong. Indeed in the case of   `index = 8`, `samplingTime = 0.1` and `currentTime = 9.7`  the `nextPlannedContactTime` computed by my laptop is not `10.5` but
```
nextPlannedContactTime = 10.499999999999982
```

Since the `getPresentContact` search for the latest contact whose activation time is greater is  `nextPlannedContactTime` 

https://github.com/ami-iit/bipedal-locomotion-framework/blob/e7d4addad5395674ae6a114fd647046fbc403a56/src/Contacts/src/ContactList.cpp#L214-L215

the function will return a wrong contact if the contact I'm looking for has an activation time is equal to 10.5. 

Furthermore, I think that the function should return the `end()` iterator if the   `nextPlannedContactTime` does not belong to an active contact domain. Indeed, in my case the `getPresentContact` returns a contact whose time domain is `[9.0, 9.5]` that is actually wrong

cc @S-Dafarra @traversaro

S-Dafarra, 2021-09-13:
Regarding the rounding problem, I guess it has to be handled at the user side. Since no uncertainty is considered at the planning level, the input time is considered as it is, without tolerances. I would suggest rounding the input number to the number of digits you need.

Regarding
> Furthermore, I think that the function should return the `end()` iterator if the `nextPlannedContactTime` does not belong to an active contact domain. Indeed, in my case the `getPresentContact` returns a contact whose time domain is `[9.0, 9.5]` that is actually wrong

I was aware of this limit, and it is documented https://github.com/ami-iit/bipedal-locomotion-framework/blob/e7d4addad5395674ae6a114fd647046fbc403a56/src/Contacts/include/BipedalLocomotion/Contacts/ContactList.h#L205. If I remember correctly, the reason why I decided to keep it that way was to distinguish clearly the case in which we were in a moment between two contacts, or we asked for a time lower than the first activation time. In addition, if we query a time greater than the last activation time, we still get the last step. The check on the deactivation time is something that can be done by the user. 

Maybe we can edit the signature to make it clearer?

GiulioRomualdi, 2024-04-05:
This has been completely solved starting using `std::chrono::nanoseconds` to handle the time in the framework 🎉 https://github.com/ami-iit/bipedal-locomotion-framework/pull/630

closed 2024-04-05 by GiulioRomualdi