gbionics/bipedal-locomotion-framework #413: Compilation issues due to failure when locating nlohman-json library

isorrentino, 2021-09-16:
This is the error printed while compiling the robotlogy-superbuild. Note that here I had already changed branch of `manif` repo from `(HEAD detached at 0.0.4.1)` to `devel` as the `robotology-superbuild` installation is done from scratch.

```shell
[ 59%] Performing configure step for 'bipedal-locomotion-framework'
Not searching for unused variables given on the command line.
loading initial cache file /usr/local/src/robot/robotology-superbuild/build/src/bipedal-locomotion-framework/CMakeFiles/YCMTmp/bipedal-locomotion-framework-cache-Release.cmake
-- Found YARP: /usr/local/src/robot/robotology-superbuild/build/install/lib/cmake/YARP (found version "3.4.6")
-- Setting up installation of BipedalLocomotionFramework.ini to /usr/local/src/robot/robotology-superbuild/build/install/share/yarp/config/path.d folder.
-- Created target ContinuousDynamicalSystem for export BipedalLocomotionFramework.
-- Created target TextLogging for export BipedalLocomotionFramework.
-- Created target GenericContainer for export BipedalLocomotionFramework.
-- Created target YarpUtilities for export BipedalLocomotionFramework.
-- Created target ParametersHandlerYarpImplementation for export BipedalLocomotionFramework.
-- Created target ParametersHandler for export BipedalLocomotionFramework.
-- Created target CommonConversions for export BipedalLocomotionFramework.
-- Created target ManifConversions for export BipedalLocomotionFramework.
-- Created target matioCppConversions for export BipedalLocomotionFramework.
-- Created target Estimators for export BipedalLocomotionFramework.
-- Created target FloatingBaseEstimators for export BipedalLocomotionFramework.
-- Created target SystemYarpImplementation for export BipedalLocomotionFramework.
-- Created target System for export BipedalLocomotionFramework.
-- Created target Planners for export BipedalLocomotionFramework.
-- Created target ContactModels for export BipedalLocomotionFramework.
-- Created target AutoDiffCppAD for export BipedalLocomotionFramework.
-- Created target RobotInterfaceYarpImplementation for export BipedalLocomotionFramework.
-- Created target RobotInterface for export BipedalLocomotionFramework.
-- Created target Math for export BipedalLocomotionFramework.
-- Created target TSID for export BipedalLocomotionFramework.
-- Created target IK for export BipedalLocomotionFramework.
-- Created target SimplifiedModelControllers for export BipedalLocomotionFramework.
--  [ ] Plugin: FloatingBaseEstimatorDevice (ENABLE_FloatingBaseEstimatorDevice)
-- Generating the Urdf Model from generate_stairs_urdf
-- Created target application joint-position-tracking.
-- Created target application joint-trajectory-player.
-- Created target application mas-imu-test.
-- Configuring done
CMake Error at cmake/AddBipedalLocomotionLibrary.cmake:67 (add_library):
  Target "FloatingBaseEstimators" links to target
  "BipedalLocomotion::Contacts" but the target was not found.  Perhaps a
  find_package() call is missing for an IMPORTED target, or an ALIAS target
  is missing?
Call Stack (most recent call first):
  src/Estimators/CMakeLists.txt:17 (add_bipedal_locomotion_library)

CMake Error at cmake/AddBipedalLocomotionLibrary.cmake:67 (add_library):
  Target "Planners" links to target "BipedalLocomotion::Contacts" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?
Call Stack (most recent call first):
  src/Planners/CMakeLists.txt:9 (add_bipedal_locomotion_library)

CMake Error at cmake/AddBipedalLocomotionLibrary.cmake:67 (add_library):
  Target "TSID" links to target "BipedalLocomotion::Contacts" but the target
  was not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?
Call Stack (most recent call first):
  src/TSID/CMakeLists.txt:9 (add_bipedal_locomotion_library)

CMake Error at cmake/AddBipedalLocomotionApplication.cmake:30 (add_executable):
  Target "joint-position-tracking" links to target
  "BipedalLocomotion::Contacts" but the target was not found.  Perhaps a
  find_package() call is missing for an IMPORTED target, or an ALIAS target
  is missing?
Call Stack (most recent call first):
  utilities/joint-position-tracking/CMakeLists.txt:7 (add_bipedal_locomotion_application)

CMake Error at cmake/AddBipedalLocomotionApplication.cmake:30 (add_executable):
  Target "joint-trajectory-player" links to target
  "BipedalLocomotion::Contacts" but the target was not found.  Perhaps a
  find_package() call is missing for an IMPORTED target, or an ALIAS target
  is missing?
Call Stack (most recent call first):
  utilities/joint-trajectory-player/CMakeLists.txt:7 (add_bipedal_locomotion_application)

CMake Error at cmake/AddBipedalLocomotionLibrary.cmake:67 (add_library):
  Target "FloatingBaseEstimators" links to target
  "BipedalLocomotion::Contacts" but the target was not found.  Perhaps a
  find_package() call is missing for an IMPORTED target, or an ALIAS target
  is missing?
Call Stack (most recent call first):
  src/Estimators/CMakeLists.txt:17 (add_bipedal_locomotion_library)

CMake Error at cmake/AddBipedalLocomotionLibrary.cmake:67 (add_library):
  Target "Planners" links to target "BipedalLocomotion::Contacts" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?
Call Stack (most recent call first):
  src/Planners/CMakeLists.txt:9 (add_bipedal_locomotion_library)

CMake Error at cmake/AddBipedalLocomotionLibrary.cmake:67 (add_library):
  Target "TSID" links to target "BipedalLocomotion::Contacts" but the target
  was not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?
Call Stack (most recent call first):
  src/TSID/CMakeLists.txt:9 (add_bipedal_locomotion_library)

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.
CMakeFiles/bipedal-locomotion-framework.dir/build.make:109: recipe for target 'src/bipedal-locomotion-framework/CMakeFiles/YCMStamp/bipedal-locomotion-framework-configure' failed
make[2]: *** [src/bipedal-locomotion-framework/CMakeFiles/YCMStamp/bipedal-locomotion-framework-configure] Error 1
CMakeFiles/Makefile2:1281: recipe for target 'CMakeFiles/bipedal-locomotion-framework.dir/all' failed
make[1]: *** [CMakeFiles/bipedal-locomotion-framework.dir/all] Error 2
Makefile:94: recipe for target 'all' failed
make: *** [all] Error 2
```

@S-Dafarra @GiulioRomualdi @traversaro

isorrentino, 2021-09-16:
Note that we are on the icub-head and it has installed Ubuntu 18. Thanks to @traversaro we know now that the nlohmann-json library needs to be installed by hand.

GiulioRomualdi, 2021-09-16:
Blf doesn't compile on ubuntu 18.04.
You need to install spdlog by hand

S-Dafarra, 2021-09-16:
Besides that, I guess we have to avoid these errors to appear
```
CMake Error at cmake/AddBipedalLocomotionLibrary.cmake:67 (add_library):
  Target "FloatingBaseEstimators" links to target
  "BipedalLocomotion::Contacts" but the target was not found.  Perhaps a
  find_package() call is missing for an IMPORTED target, or an ALIAS target
  is missing?
Call Stack (most recent call first):
  src/Estimators/CMakeLists.txt:17 (add_bipedal_locomotion_library)

CMake Error at cmake/AddBipedalLocomotionLibrary.cmake:67 (add_library):
  Target "Planners" links to target "BipedalLocomotion::Contacts" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?
Call Stack (most recent call first):
  src/Planners/CMakeLists.txt:9 (add_bipedal_locomotion_library)

CMake Error at cmake/AddBipedalLocomotionLibrary.cmake:67 (add_library):
  Target "TSID" links to target "BipedalLocomotion::Contacts" but the target
  was not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?
Call Stack (most recent call first):
  src/TSID/CMakeLists.txt:9 (add_bipedal_locomotion_library)

CMake Error at cmake/AddBipedalLocomotionApplication.cmake:30 (add_executable):
  Target "joint-position-tracking" links to target
  "BipedalLocomotion::Contacts" but the target was not found.  Perhaps a
  find_package() call is missing for an IMPORTED target, or an ALIAS target
  is missing?
Call Stack (most recent call first):
  utilities/joint-position-tracking/CMakeLists.txt:7 (add_bipedal_locomotion_application)

CMake Error at cmake/AddBipedalLocomotionApplication.cmake:30 (add_executable):
  Target "joint-trajectory-player" links to target
  "BipedalLocomotion::Contacts" but the target was not found.  Perhaps a
  find_package() call is missing for an IMPORTED target, or an ALIAS target
  is missing?
Call Stack (most recent call first):
  utilities/joint-trajectory-player/CMakeLists.txt:7 (add_bipedal_locomotion_application)

CMake Error at cmake/AddBipedalLocomotionLibrary.cmake:67 (add_library):
  Target "FloatingBaseEstimators" links to target
  "BipedalLocomotion::Contacts" but the target was not found.  Perhaps a
  find_package() call is missing for an IMPORTED target, or an ALIAS target
  is missing?
Call Stack (most recent call first):
  src/Estimators/CMakeLists.txt:17 (add_bipedal_locomotion_library)

CMake Error at cmake/AddBipedalLocomotionLibrary.cmake:67 (add_library):
  Target "Planners" links to target "BipedalLocomotion::Contacts" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?
Call Stack (most recent call first):
  src/Planners/CMakeLists.txt:9 (add_bipedal_locomotion_library)

CMake Error at cmake/AddBipedalLocomotionLibrary.cmake:67 (add_library):
  Target "TSID" links to target "BipedalLocomotion::Contacts" but the target
  was not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?
Call Stack (most recent call first):
  src/TSID/CMakeLists.txt:9 (add_bipedal_locomotion_library)
```

GiulioRomualdi, 2021-09-16:
Opening a PR in blf. Sorry for that

GiulioRomualdi, 2021-09-16:
Here the PR https://github.com/ami-iit/bipedal-locomotion-framework/pull/414

GiulioRomualdi, 2021-09-16:
This PR https://github.com/ami-iit/bipedal-locomotion-framework/pull/415 block the compilation in case of a wrong version of spdlog  https://github.com/ami-iit/bipedal-locomotion-framework/issues/413#issuecomment-920903787

GiulioRomualdi, 2021-09-16:
I think we can close the issue since the PRs have been merged. @isorrentino please reopen it again in case of problems

closed 2021-09-16 by GiulioRomualdi