gbionics/bipedal-locomotion-framework #442: Move the idyntree bindings outside from `floating_base_estimators` in the bindings

GiulioRomualdi, 2021-11-05:
At the current stage if you want to load a KinDynConputations object in `blf` you should use the `floating_base_estimators` submodule

```python
import bipedal_locomotion_framework as blf                              
kindyn = blf.floating_base_estimators.KinDynComputations()
```

However, the `FloatingBaseEstimator` library is compiled only if 

https://github.com/ami-iit/bipedal-locomotion-framework/blob/d793ac71d80ace4fc3b793db94707f7d6f8113ea/cmake/BipedalLocomotionFrameworkDependencies.cmake#L150-L152

Furthermore, the `KinDyn` object can be used also outside the submodule.

I would suggest moving it to a new submodule called `idyntree`, the submodule will contain all the future bindings related to iDynTree

In this case, it will be possible to instantiate a kindyn object with 

```python
import bipedal_locomotion_framework as blf                              
kindyn = blf.idyntree.KinDynComputations()
```

what do you think? @ami-iit/blf-developers

traversaro, 2021-11-05:
Good idea!

prashanthr05, 2021-11-05:
Related issue: https://github.com/ami-iit/bipedal-locomotion-framework/issues/293

GiulioRomualdi, 2021-11-05:
Ops! Sorry! I close this one in favour of #293

closed 2021-11-05 by GiulioRomualdi