gbionics/bipedal-locomotion-framework #458: If a robot is removed from the config files in YarpRobotLoggerDevice the installation fails

isorrentino, 2021-11-17:
I compiled `bipedal-locomotion-framework` and in particular `YarpRobotLoggerDevice` including `iCubGazeboV3` among the available robots.
Then, I switched to another branch that was not containing the config files for this robot. I run `make uninstall` and then `make install` but I got this error:

```bash
CMake Error at devices/YarpRobotLoggerDevice/app/cmake_install.cmake:45 (file):
file INSTALL cannot find
"/home/isorrentino/dev/robotology-superbuild/src/bipedal-locomotion-framework/devices/YarpRobotLoggerDevice/app/robots/iCubGazeboV3":
No such file or directory.
Call Stack (most recent call first):
devices/YarpRobotLoggerDevice/cmake_install.cmake:70 (include)
devices/cmake_install.cmake:47 (include)
cmake_install.cmake:59 (include)
```

Maybe there is a problem with `make uninstall`.

cc @GiulioRomualdi @traversaro

traversaro, 2021-11-17:
Did you run `cmake` after you switched branch? Those part of CMake uses `GLOB` (not sure where), and if you use that command you also need to run `cmake` whenever the files on disk change.

isorrentino, 2021-11-17:
Running `cmake` the error disappeared. Thanks a lot @traversaro. Closing.

closed 2021-11-17 by isorrentino