gbionics/bipedal-locomotion-framework #499: YarpRobotLoggerDevice not running at the specified frequency

isorrentino, 2022-01-17:
Acquiring some datasets by running the `YarpRobotLoggerDevice` @GiulioRomualdi @paolo-viceconte and I noticed that the data were not logged at the specified frequency.
Investigating this, I found the source of the problem.
When the option `stream_pids` (https://github.com/ami-iit/bipedal-locomotion-framework/blob/c2c7d6e481c4487ca2ce9547dfdc1094682dd000/devices/YarpRobotLoggerDevice/app/robots/iCubGenova09/yarp-robot-logger.xml#L32) is true the `advance` (https://github.com/ami-iit/bipedal-locomotion-framework/blob/c2c7d6e481c4487ca2ce9547dfdc1094682dd000/devices/YarpRobotLoggerDevice/src/YarpRobotLoggerDevice.cpp#L336-L339) of the SensorBridge called in the `run` of the device, takes more than expected. 
@S-Dafarra suggested that the possible reason could be that the port streaming the PIDs is an RPC.

cc @S-Dafarra @GiulioRomualdi @paolo-viceconte @prashanthr05

GiulioRomualdi, 2022-01-25:
Cc @traversaro

traversaro, 2022-01-25:
I am not sure which YARP methods are called if `stream_pids` is enabled, but in general I think this is expected, see the related comment in https://github.com/robotology/yarp-telemetry/pull/125#issuecomment-828267644 : 

> An issue that unfortunately it is not clear from the YARP interface level and that is instead implementation specific is that while some quantities are already available in physical device such [`embObjMotionControl`](https://github.com/robotology/icub-main/tree/master/src/libraries/icubmod/embObjMotionControl), so calling the related method is not expensive is just copying some buffers, while some other methods are blocking and generate traffic at the firmware level to retrieve online the quantities from the firmware, so having a lot of those calls could be problematic (as far as I know which quantities are cached in `embObjMotionControl` and which one are asked on the fly to the firmware is not documented, but it is probably better to ask to Marco Accame) . 

In a nutshell, the custom Ethernet protocol used inside the iCub robot does not stream all the variables available in the firmware to the `icub-head` . In the YARP interface, more variables than the one that are continously streamed are available, but those that are not streamed are available via RPC call between the `icub-head` and the firmware, so it is quite expensive in terms of time.

GiulioRomualdi, 2022-01-27:
Given that we should decide how to proceed. 
Currently, the PID infos are retrieved with `getPidReferences` and `getPidErrors`

https://github.com/ami-iit/bipedal-locomotion-framework/blob/892164bcebbfd56fa9410ef0516a4cb8d7ce9678/src/RobotInterface/YarpImplementation/include/BipedalLocomotion/RobotInterface/YarpSensorBridgeImpl.h#L1677-L1684

Accordingly, to `yarp` documentation, there is no way to specify a timeout to be sure that the function return within a device period.

S-Dafarra, 2022-01-27:
What about having a separate thread just for the "slow" calls? In this separate thread, you could call those methods and fill some buffer, which is then used in the main thread to be saved :thinking:

traversaro, 2022-01-27:
Just to clarify, the problem is just the thread that is being blocked. If you call at high rate those functions, you create non-deterministic traffic on the internal Ethernet bus of robot, and this could create unexpeted delays sending or receiving regular periodic packages.

S-Dafarra, 2022-01-27:
> Just to clarify, the problem is just the thread that is being blocked. If you call at high rate those functions, you create non-deterministic traffic on the internal Ethernet bus of robot, and this could create unexpeted delays sending or receiving regular periodic packages.

Good point

isorrentino, 2024-01-17:
Hi @GiulioRomualdi can we close this issue?

GiulioRomualdi, 2024-01-17:
For the time being we decided to switch off the steaming of the PID values

closed 2024-01-17 by GiulioRomualdi