gbionics/bipedal-locomotion-framework #510: Add VectorsCollectionWrapper to collect arbitrary vector port data into a single VectorsCollection port

prashanthr05, 2022-02-09:
With the addition of [`VectorsCollection`](https://github.com/ami-iit/bipedal-locomotion-framework/blob/master/src/YarpUtilities/thrifts/BipedalLocomotion/YarpUtilities/VectorsCollection.thrift), it is now possible to stream a collection of data through a single YARP port.
This is being successfully used in the [walking_controllers](https://github.com/robotology/walking-controllers) to stream relevant sets of data in a single port which is then passed on to the [YarpRobotLoggerDevice](https://github.com/ami-iit/bipedal-locomotion-framework/tree/master/devices/YarpRobotLoggerDevice) to log relevant data.

However, this implementation currently requires the client to prepare the VectorsCollection structure internally and stream it. This usage is not useful for recording arbitrary vector ports available on the network.

**Proposition:** A Yarp Device called `VectorsCollectionWrapper` that connects and reads from arbitrary **vector** ports on the network and populates the VectorsCollection structure and streams the relevant data in a single port.
For example, this can be useful in recording two streams of data coming from different sources, say Gazebo base state and estimator base state, collect them in a VectorsCollection port and stream which in turn can be sent as input to the 'YarpRobotLoggerDevice' to log the base state from two streams.

The configuration of the device will be as simple as,
``` xml
<?xml version="1.0" encoding="UTF-8" ?>
<device  xmlns:xi="http://www.w3.org/2001/XInclude" name="vectors-collection-wrapper" type="VectorsCollectionWrapper">
  <param name="sampling_period_in_s">0.01</param>
  <param name="port_prefix">/vcWrapper</param>
  <param name="remote_port_names">("/icubSim/floating_base/state:o", "/icubSim/head/state:o")</param>
  <param name="remote_var_names">("sim_base_state", "sim_head_state")</param>
  <param name="output_port_name">"/vcWrapper/base_state"</param>
</device>
```

The input parameters include,
-  the device's periodic thread period
- port prefix 
- remote port names which include the ports data that we want to collect
- remote var names are the identifiers for the port and must be in the same order and size as remote port names
- output vectors collection port

The output of the device,
[image]

**Note:** The scalability of this solution needs to be taken with a grain of salt. Nevertheless, for small use-case scenarios, this is useful.

cc @GiulioRomualdi @S-Dafarra @traversaro @isorrentino @paolo-viceconte  

As you can see, I have already implemented this device. If you agree, I can open a PR for the device.

GiulioRomualdi, 2022-02-09:
Just a small comment, why a device? A simple application should be enough right?

prashanthr05, 2022-02-10:
By application, do you mean a RFModule?
If that's the case, in essence both a device and RFModule runs a periodic thread. With the device implementation, we may have the advantage of passing it through a single yarprobotinterface file  along with the YarpRobotLoggerDevice.

Anyhow, besides that, I was also wondering if it is actually useful to maintain this logic in a separate device or just add this logic within the YarpRobotLoggerDevice. Do you think it's better to keep them separate?

S-Dafarra, 2022-02-15:
I think it is a good idea!

traversaro, 2022-02-16:
I like it!

PR #512 (2022-02-23): Add VectorsCollectionWrapper device
Closes #510
closed 2022-02-23 by GiulioRomualdi