gbionics/bipedal-locomotion-framework #530: QPInverseKinematics and joint limit constraints?

shikui08, 2022-05-07:
Has been using QPInverseKinematics for a while. The coding is similar to bipedal-locomotion-framework/src/IK/tests/QPInverseKinematicsTest.cpp.  But I found out that the result dont' take joint pos limits into account.
After checking out the source code, it seems to me that tasks with high priority are considered as constraints in QPInverseKinematics solver, however joint pos limits are not take into account.
Am I mistaken?
Is it possible for QPInverseKinematics to take into account joint limit constraint? 

Thanks.

GiulioRomualdi, 2022-05-09:
Hi @shikui08 thank you for using bipedal locomotion framework! Yes it's something we should add in our formulation. I can allocate a timeslot to implement the feature. Otherwise if you need it in very a short time, you may try to implement the new task and open a PR

shikui08, 2022-05-13:
> Hi @shikui08 thank you for using bipedal locomotion framework! Yes it's something we should add in our formulation. I can allocate a timeslot to implement the feature. Otherwise if you need it in very a short time, you may try to implement the new task and open a PR

Thanks for your reply. 
Considering whole body ik which respects pos limits I come across two codebases:
1) bio_ik. 
 
2) [https://github.com/mrsp/whole_body_ik/](url) 
It's Jacobian based, more straight forward, though it doesn't take into account task priorities.  

It turns out for me that each IK approach has it's pros and cons. And sometimes I find out that the result is more sensitive to the way the problem is defined, rather than the way the problem is solved.
Hope it helps~

lrapetti, 2022-05-13:
Hi @shikui08, if you are interested you can also have a look at the QPinverseKinematics joint limits implementation used for https://ieeexplore.ieee.org/stamp/stamp.jsp?arnumber=9035059 and https://www.mdpi.com/1999-4893/13/10/266.

The implemented strategy is described in https://github.com/robotology/human-dynamics-estimation/issues/132, while the code can be found in https://github.com/robotology/human-dynamics-estimation/tree/master/HumanDynamicsEstimationLibrary.

shikui08, 2022-05-14:
> Hi @shikui08, if you are interested you can also have a look at the QPinverseKinematics joint limits implementation used for https://ieeexplore.ieee.org/stamp/stamp.jsp?arnumber=9035059 and https://www.mdpi.com/1999-4893/13/10/266.
> 
> The implemented strategy is described in [robotology/human-dynamics-estimation#132](https://github.com/robotology/human-dynamics-estimation/issues/132), while the code can be found in https://github.com/robotology/human-dynamics-estimation/tree/master/HumanDynamicsEstimationLibrary.

Hi @lrapetti, thanks for your kind reply and great work. Glad to look further into the paper&code.

GiulioRomualdi, 2023-02-10:
https://github.com/ami-iit/bipedal-locomotion-framework/pull/603 implements the joint limit constraints

PR #603 (2023-02-28): Implement the JointLimitsTask for the IK
This closes #530 

cc @shikui08
closed 2023-02-28 by GiulioRomualdi