gbionics/bipedal-locomotion-framework #540: Understand how to log the YarpLog messages in YarpRobotLogVisualizer

GiulioRomualdi, 2022-06-22:
The idea is to log, along with the numerical data, the messages provided by YarpLog. This will further help us to analyze the experiments on the robot.

I noticed that all the messages are stored in `/log/<node-name>/<module-name>/<PID>`. So the idea is to create a periodic thread that checks for the active logs and stores it. 

@traversaro or @S-Dafarra do you know if `yarprun` uses a specific thrift to create the message? or is it just a simple bottle? This is an example of YarpLog collected from the walking 
```
$ yarp read ... /log/my_pc/WalkingModule/18553
[INFO] |yarp.os.Port|/tmp/port/1| Port /tmp/port/1 active at tcp://10.240.9.229:10096/
[INFO] |yarp.os.impl.PortCoreInputUnit|/tmp/port/1| Receiving input from /log/my_pc/WalkingModule/18553 to /tmp/port/1 using tcp
"[/log/my_pc/WalkingModule/18553]" "\n"
"[/log/my_pc/WalkingModule/18553]" "******************************************************************************\n"
"[/log/my_pc/WalkingModule/18553]" "This program contains Ipopt, a library for large-scale nonlinear optimization.\n"
"[/log/my_pc/WalkingModule/18553]" " Ipopt is released as open source code under the Eclipse Public License (EPL).\n"
"[/log/my_pc/WalkingModule/18553]" "         For more information visit https://github.com/coin-or/Ipopt\n"
"[/log/my_pc/WalkingModule/18553]" "******************************************************************************\n"
"[/log/my_pc/WalkingModule/18553]" "\n"
"[/log/my_pc/WalkingModule/18553]" "q desired IK     8.84133 -0.00886641  0.00217841    -3.60285     25.1932     10.4581      30.866    -3.60087     25.1922     10.4589      30.865     5.15202     1.33298   -0.262539     -32.407    -21.4179    -1.35273     5.17057     1.33034   -0.255178    -32.4197    -21.4119    -1.36044\n"
"[/log/my_pc/WalkingModule/18553]" "(level INFO) (systemtime 1655889012.20248603821) (message \"[WalkingModule::updateModule] The robot is prepared.\")\n"

```

S-Dafarra, 2022-06-22:
Probably the ``YARP_LOG_PROCESS_LABEL`` can be useful to filter the log ports you are interested in. See https://github.com/robotology/yarp/blob/master/doc/yarp_logging.md

Regarding the format, I am not sure. I can check.

S-Dafarra, 2022-06-22:
> @traversaro or @S-Dafarra do you know if `yarprun` uses a specific thrift to create the message? or is it just a simple bottle?

I think it is just a bottle. This is how it is parsed in the ``yarplogger``: https://github.com/robotology/yarp/blob/1986858acb14880bccf9073e3e74c082dc5caa0b/src/libYARP_logger/src/yarp/logger/YarpLogger.cpp#L248-L513

GiulioRomualdi, 2022-06-23:
The first working implementation can be found at this commit https://github.com/ami-iit/bipedal-locomotion-framework/commit/9968c04985c3665a912ef54ba89a500febc88525

Al the text logs will be stored in `robot_logger_device.log.<node-name>.<process-name>.p<pid-number>` in this specific case `robot_logger_device.log.test_giulio.WalkingModule.p245198`

[image]

S-Dafarra, 2022-06-23:
That's great!

DanielePucci, 2022-06-23:
This is super useful also for other teams, like @ami-iit/ironcub 

CC
@ami-iit/artificial-mechanical-intelligence

GiulioRomualdi, 2022-06-27:
Thanks to https://github.com/ami-iit/bipedal-locomotion-framework/pull/541 and https://github.com/ami-iit/robot-log-visualizer/releases/tag/v0.2.0. it is now possible to load the YarpLog messages. Given the time instant provided by the slider, the latest message is highlighted. 
The following colormap is chosen:
1. 🔴 errors
2. 🟠 warnings
3. 🔵 debug
4. 🟢 info 

https://user-images.githubusercontent.com/16744101/175880161-20d4277c-a9f1-46da-964c-54e29db8cba7.mp4

S-Dafarra, 2022-06-27:
This is awesome!!

traversaro, 2022-06-27:
Super!

PR #541 (2022-07-29): Add the possibility to log the YarpTextLogging in the YarpRobotLogger
This closes #540
closed 2022-07-29 by GiulioRomualdi