gbionics/bipedal-locomotion-framework #545: blf.ik.QPInverseKinematics.add_task stops working when the binsings are compiled using conda

GiulioRomualdi, 2022-06-28:
If I compile the bindings with the dependencies installed by conda and I run the following code I got this error

```python
In [1]: import bipedal_locomotion_framework.bindings as blf

In [2]: so3_task = blf.ik.SO3Task()

In [3]: help(so3_task)

In [4]: qp_ik = blf.ik.QPInverseKinematics()

In [5]: qp_ik.add_task(so3_task, "c", 0)
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
Input In [5], in <cell line: 1>()
----> 1 qp_ik.add_task(so3_task, "c", 0)

TypeError: add_task(): incompatible function arguments. The following argument types are supported:
    1. (self: bipedal_locomotion_framework.bindings.ik.ILinearTaskSolverIK, task: bipedal_locomotion_framework.bindings.ik.IKLinearTask, task_name: str, priority: int, weight: Optional[numpy.ndarray[numpy.float64[m, 1]]] = array([], dtype=float64)) -> bool

Invoked with: ====== QPInverseKinematics class ======
The optimization problem is composed by the following tasks:
Note: The lower is the integer associated to the priority, the higher is the priority.
==========================
, Generic Linear Task Element, 'c', 0
```

On the other hand if I compile them with apt dependencies everything went smooth
```
In [1]: import bipedal_locomotion_framework.bindings as blf

In [2]: so3_task = blf.ik.SO3Task()

In [3]: qp_ik = blf.ik.QPInverseKinematics()

In [4]: qp_ik.add_task(so3_task, "c", 0)
Out[4]: True
```

this is the outcome of conda list 
```
# packages in environment at /home/gromualdi/mambaforge/envs/blf-env:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
ace                       7.0.7                h27087fc_0    conda-forge
alsa-lib                  1.2.6.1              h7f98852_0    conda-forge
ampl-mp                   3.1.0             h2cc385e_1006    conda-forge
aom                       3.3.0                h27087fc_1    conda-forge
assimp                    5.2.4                h4c92fa2_0    conda-forge
asttokens                 2.0.5              pyhd8ed1ab_0    conda-forge
attr                      2.5.1                h166bdaf_0    conda-forge
attrs                     21.4.0             pyhd8ed1ab_0    conda-forge
backcall                  0.2.0              pyh9f0ad1d_0    conda-forge
backports                 1.0                        py_2    conda-forge
backports.functools_lru_cache 1.6.4              pyhd8ed1ab_0    conda-forge
binutils                  2.36.1               hdd6e379_2    conda-forge
binutils_impl_linux-64    2.36.1               h193b22a_2    conda-forge
binutils_linux-64         2.36                hf3e587d_10    conda-forge
boost-cpp                 1.74.0               h75c5d50_8    conda-forge
bzip2                     1.0.8                h7f98852_4    conda-forge
c-ares                    1.18.1               h7f98852_0    conda-forge
c-compiler                1.4.2                h166bdaf_0    conda-forge
ca-certificates           2022.6.15            ha878542_0    conda-forge
cairo                     1.16.0            ha61ee94_1011    conda-forge
casadi                    3.5.5           py39hc796803_10    conda-forge
catch2                    2.13.9               h924138e_0    conda-forge
cmake                     3.23.2               h5432695_0    conda-forge
compilers                 1.4.2                ha770c72_0    conda-forge
[... 258 lines cut ...]
xorg-libxau               1.0.9                h7f98852_0    conda-forge
xorg-libxdamage           1.1.5                h7f98852_1    conda-forge
xorg-libxdmcp             1.1.3                h7f98852_0    conda-forge
xorg-libxext              1.3.4                h7f98852_1    conda-forge
xorg-libxfixes            5.0.3             h7f98852_1004    conda-forge
xorg-libxi                1.7.10               h7f98852_0    conda-forge
xorg-libxrandr            1.5.2                h7f98852_1    conda-forge
xorg-libxrender           0.9.10            h7f98852_1003    conda-forge
xorg-libxt                1.2.1                h7f98852_2    conda-forge
xorg-randrproto           1.5.0             h7f98852_1001    conda-forge
xorg-renderproto          0.11.1            h7f98852_1002    conda-forge
xorg-util-macros          1.19.3               h7f98852_0    conda-forge
xorg-xextproto            7.3.0             h7f98852_1002    conda-forge
xorg-xf86vidmodeproto     2.3.1             h7f98852_1002    conda-forge
xorg-xproto               7.0.31            h7f98852_1007    conda-forge
xz                        5.2.5                h516909a_1    conda-forge
yaml                      0.2.5                h7f98852_2    conda-forge
yarp                      3.7.1                ha770c72_0    conda-forge
yarp-cxx                  3.7.1                hfee3fe3_0    conda-forge
yarp-python               3.7.1            py39h5a03fae_0    conda-forge
ycm-cmake-modules         0.14.2               h27087fc_0    conda-forge
zeromq                    4.3.4                h9c3ff4c_1    conda-forge
zlib                      1.2.12               h166bdaf_1    conda-forge
zstd                      1.5.2                h8a70e8d_1    conda-forge
```

traversaro, 2022-06-28:
I can give it a shot and rebuild the manif conda package, but this is indeed quite problematic.

GiulioRomualdi, 2022-06-28:
I was also thinking that the problem was related to pybind however [here](https://github.com/ami-iit/bipedal-locomotion-framework/runs/7024582215?check_suite_focus=true#step:3:260) the CI was running correctly and the version of pybind was `2.9.2` the very same that is used now

GiulioRomualdi, 2022-06-28:
Installing numpy 1.22.4 from conda fixes the problem

GiulioRomualdi, 2022-06-28:
While for blf is already like this right? https://github.com/robotology/robotology-superbuild/blob/27e970124efff73bb9aaf0545d79dbe297a26c84/cmake/Buildbipedal-locomotion-framework.cmake#L77

traversaro, 2022-06-28:
> While for blf is already like this right? https://github.com/robotology/robotology-superbuild/blob/27e970124efff73bb9aaf0545d79dbe297a26c84/cmake/Buildbipedal-locomotion-framework.cmake#L77

Unfortunatly, not. I checked the `info/index.json` from a blf conda package, and it is:
~~~
{
  "arch": "x86_64",
  "build": "py39hddcaf54_57",
  "build_number": 57,
  "depends": [
    "casadi >=3.5.5,<3.6.0a0",
    "cppad >=20220000.4,<20220000.5.0a0",
    "eigen",
    "idyntree >=5.2.1,<6.0a0",
    "libgcc-ng >=12",
    "libstdcxx-ng >=12",
    "lie-group-controllers >=0.1.1,<0.1.2.0a0",
    "matio-cpp >=0.2.0,<0.2.1.0a0",
    "numpy >=1.19.5,<2.0a0",
    "osqp-eigen >=0.7.0,<0.7.1.0a0",
    "pybind11-abi 4",
    "python >=3.9,<3.10.0a0",
    "python_abi 3.9.* *_cp39",
    "qhull >=2020.2,<2020.3.0a0",
    "spdlog >=1.10.0,<1.11.0a0",
    "unicycle-footstep-planner >=0.4.1,<0.4.2.0a0",
    "yarp-cxx >=3.7.1,<3.7.2.0a0",
    "yarp-telemetry >=0.5.1,<0.5.2.0a0"
  ],
  "name": "bipedal-locomotion-framework",
  "platform": "linux",
  "subdir": "linux-64",
  "timestamp": 1656381946169,
  "version": "0.7.0"
}
~~~

traversaro, 2022-06-28:
> > Installing numpy 1.22.4 from conda fixes the problem
> 
> Probably the best option is to add a run dependency of manifpy on the precise version of numpy with which it is built.

Done in https://github.com/traversaro/manif-feedstock/commit/23e95e829360f92b8d9a6bbeadb47bdd46bcf742 .

traversaro, 2022-06-28:
> Unfortunatly, not. I checked the `info/index.json` from a blf conda package, and it is:
>
> ```
> {
>   "arch": "x86_64",
>   "build": "py39hddcaf54_57",
>   "build_number": 57,
>   "depends": [
>     "casadi [...]

Not sure if it worth trying to implement something like that on robotology-superbuild, or it make more sense to just move blf to conda-forge .

GiulioRomualdi, 2022-07-29:
This has been fixed by https://github.com/ami-iit/bipedal-locomotion-framework/issues/545#issuecomment-1168787064
Thank you @traversaro

closed 2022-07-29 by GiulioRomualdi