gbionics/bipedal-locomotion-framework #548: YarpRobotLoggerDevice sometimes fails to stop

S-Dafarra, 2022-07-13:
Sometimes ``YarpRobotLoggerDevice`` does not stop and we have to kill it. The problem is that we lose data.

It would be nice to make sure that at least the ``mat`` file is saved in any case. My hunch is that some camera thread fails to stop.

The desired behavior would be that when attempting to close, the ``mat`` file is saved first, then all the videos that managed to stop properly.

cc @GiulioRomualdi

GiulioRomualdi, 2022-07-13:
Yes the issue is that one. Once I had the same problem when I  forgot to run the robot eyes camera device.

In theory robometry saves the mat file before calling the callback for saving the data

S-Dafarra, 2022-07-13:
Ok, because I was noticing that the mat file is not saved either when te closure hangs :thinking:

GiulioRomualdi, 2022-07-13:
By the way I will double check

GiulioRomualdi, 2022-07-29:
I got it! The problem is due to the text logging feature. In details it is due to 

https://github.com/ami-iit/bipedal-locomotion-framework/blob/27313b5ded985aa7fda738a233830374782e07b0/devices/YarpRobotLoggerDevice/src/YarpRobotLoggerDevice.cpp#L608

Yesterday I was able to replicate the problem where some ports were  not reachable anymore but they were still registered in the server. In this case `yarp::os::Network::connect` tries to connect the logger port to a non existing port and remains blocked. We should chheck if the port [exists](https://www.yarp.it/git-master/classyarp_1_1os_1_1NetworkBase.html#a837bcebfae3a1e338021cac8ac1777d1) before establishing the connection

S-Dafarra, 2022-07-29:
Awesome!

GiulioRomualdi, 2022-07-29:
https://github.com/ami-iit/bipedal-locomotion-framework/pull/541/commits/f108da706e3e209caa53c2641ba37d2e28cdadc8 should fix it

GiulioRomualdi, 2022-07-29:
The fix has been ported in master. Closing!

closed 2022-07-29 by GiulioRomualdi