gbionics/bipedal-locomotion-framework #586: Compilation error with OpenCV 4.7

traversaro, 2023-01-23:
~~~
2023-01-23T03:24:17.0141618Z [103/215] Building CXX object src/Perception/CMakeFiles/PerceptionFeatures.dir/src/ArucoDetector.cpp.o
2023-01-23T03:24:17.0142151Z FAILED: src/Perception/CMakeFiles/PerceptionFeatures.dir/src/ArucoDetector.cpp.o 
2023-01-23T03:24:17.0147419Z /usr/share/miniconda3/envs/test/bin/x86_64-conda-linux-gnu-c++ -DFMT_SHARED -DPerceptionFeatures_EXPORTS -DSPDLOG_COMPILED_LIB -DSPDLOG_FMT_EXTERNAL -DSPDLOG_SHARED_LIB -D_USE_MATH_DEFINES -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/Perception/include -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/ParametersHandler/include -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/GenericContainer/include -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/TextLogging/include -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/Conversions/include -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/System/include -isystem /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/include -isystem /usr/share/miniconda3/envs/test/include/eigen3 -isystem /usr/share/miniconda3/envs/test/include/opencv4 -fvisibility-inlines-hidden -fmessage-length=0 -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -pipe -isystem /usr/share/miniconda3/envs/test/include -O3 -DNDEBUG -fPIC -MD -MT src/Perception/CMakeFiles/PerceptionFeatures.dir/src/ArucoDetector.cpp.o -MF src/Perception/CMakeFiles/PerceptionFeatures.dir/src/ArucoDetector.cpp.o.d -o src/Perception/CMakeFiles/PerceptionFeatures.dir/src/ArucoDetector.cpp.o -c /home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/Perception/src/ArucoDetector.cpp
2023-01-23T03:24:17.0151536Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/Perception/src/ArucoDetector.cpp:58:48: error: 'PREDEFINED_DICTIONARY_NAME' is not a member of 'cv::aruco'
2023-01-23T03:24:17.0152212Z    58 |     std::unordered_map<std::string, cv::aruco::PREDEFINED_DICTIONARY_NAME>
2023-01-23T03:24:17.0152573Z       |                                                ^~~~~~~~~~~~~~~~~~~~~~~~~~
2023-01-23T03:24:17.0153341Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/Perception/src/ArucoDetector.cpp:58:74: error: template argument 2 is invalid
2023-01-23T03:24:17.0153967Z    58 |     std::unordered_map<std::string, cv::aruco::PREDEFINED_DICTIONARY_NAME>
2023-01-23T03:24:17.0154340Z       |                                                                          ^
2023-01-23T03:24:17.0155078Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/Perception/src/ArucoDetector.cpp:58:74: error: template argument 5 is invalid
2023-01-23T03:24:17.0156254Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/Perception/src/ArucoDetector.cpp:59:45: error: 'cv::aruco::PREDEFINED_DICTIONARY_NAME' has not been declared
2023-01-23T03:24:17.0156925Z    59 |         availableDict{{"4X4_50", cv::aruco::PREDEFINED_DICTIONARY_NAME::DICT_4X4_50},
2023-01-23T03:24:17.0157373Z       |                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~
2023-01-23T03:24:17.0158287Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/Perception/src/ArucoDetector.cpp:60:46: error: 'cv::aruco::PREDEFINED_DICTIONARY_NAME' has not been declared
2023-01-23T03:24:17.0158937Z    60 |                       {"4X4_100", cv::aruco::PREDEFINED_DICTIONARY_NAME::DICT_4X4_100},
2023-01-23T03:24:17.0159404Z       |                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~
2023-01-23T03:24:17.0160333Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/Perception/src/ArucoDetector.cpp:61:46: error: 'cv::aruco::PREDEFINED_DICTIONARY_NAME' has not been declared
2023-01-23T03:24:17.0160983Z    61 |                       {"4X4_250", cv::aruco::PREDEFINED_DICTIONARY_NAME::DICT_4X4_250},
2023-01-23T03:24:17.0161323Z       |                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~
2023-01-23T03:24:17.0162241Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/Perception/src/ArucoDetector.cpp:62:47: error: 'cv::aruco::PREDEFINED_DICTIONARY_NAME' has not been declared
2023-01-23T03:24:17.0162884Z    62 |                       {"4X4_1000", cv::aruco::PREDEFINED_DICTIONARY_NAME::DICT_4X4_1000},
2023-01-23T03:24:17.0163247Z       |                                               ^~~~~~~~~~~~~~~~~~~~~~~~~~
2023-01-23T03:24:17.0164142Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/Perception/src/ArucoDetector.cpp:63:45: error: 'cv::aruco::PREDEFINED_DICTIONARY_NAME' has not been declared
2023-01-23T03:24:17.0164850Z    63 |                       {"5X5_50", cv::aruco::PREDEFINED_DICTIONARY_NAME::DICT_5X5_50},
2023-01-23T03:24:17.0165212Z       |                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~
2023-01-23T03:24:17.0166142Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/Perception/src/ArucoDetector.cpp:64:46: error: 'cv::aruco::PREDEFINED_DICTIONARY_NAME' has not been declared
2023-01-23T03:24:17.0166869Z    64 |                       {"5X5_100", cv::aruco::PREDEFINED_DICTIONARY_NAME::DICT_5X5_100},
2023-01-23T03:24:17.0167202Z       |                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~
2023-01-23T03:24:17.0168053Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/Perception/src/ArucoDetector.cpp:65:46: error: 'cv::aruco::PREDEFINED_DICTIONARY_NAME' has not been declared
2023-01-23T03:24:17.0289391Z    65 |                       {"5X5_250", cv::aruco::PREDEFINED_DICTIONARY_NAME::DICT_5X5_250},
2023-01-23T03:24:17.0289784Z       |                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~
2023-01-23T03:24:17.0290905Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/Perception/src/ArucoDetector.cpp:66:47: error: 'cv::aruco::PREDEFINED_DICTIONARY_NAME' has not been declared
2023-01-23T03:24:17.0291582Z    66 |                       {"5X5_1000", cv::aruco::PREDEFINED_DICTIONARY_NAME::DICT_5X5_1000},
2023-01-23T03:24:17.0291948Z       |                                               ^~~~~~~~~~~~~~~~~~~~~~~~~~
2023-01-23T03:24:17.0292861Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/Perception/src/ArucoDetector.cpp:67:45: error: 'cv::aruco::PREDEFINED_DICTIONARY_NAME' has not been declared
2023-01-23T03:24:17.0293513Z    67 |                       {"6X6_50", cv::aruco::PREDEFINED_DICTIONARY_NAME::DICT_6X6_50},
2023-01-23T03:24:17.0293870Z       |                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~
2023-01-23T03:24:17.0294811Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/Perception/src/ArucoDetector.cpp:68:46: error: 'cv::aruco::PREDEFINED_DICTIONARY_NAME' has not been declared
2023-01-23T03:24:17.0295742Z    68 |                       {"6X6_100", cv::aruco::PREDEFINED_DICTIONARY_NAME::DICT_6X6_100},
2023-01-23T03:24:17.0296104Z       |                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~
2023-01-23T03:24:17.0297032Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/Perception/src/ArucoDetector.cpp:69:46: error: 'cv::aruco::PREDEFINED_DICTIONARY_NAME' has not been declared
2023-01-23T03:24:17.0297674Z    69 |                       {"6X6_250", cv::aruco::PREDEFINED_DICTIONARY_NAME::DICT_6X6_250},
2023-01-23T03:24:17.0298009Z       |                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~
2023-01-23T03:24:17.0298916Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/Perception/src/ArucoDetector.cpp:70:47: error: 'cv::aruco::PREDEFINED_DICTIONARY_NAME' has not b [... cut ...]
2023-01-23T03:24:17.0323693Z [105/215] Building CXX object src/TSID/CMakeFiles/TSID.dir/src/QPTSID.cpp.o
2023-01-23T03:24:17.0324062Z ninja: build stopped: subcommand failed.
~~~

The error is probably due to https://github.com/opencv/opencv/pull/22368, see for example https://github.com/opencv/opencv/pull/22368/files#diff-511032611136a88065135e7fb48aa6b204fbb15b9e14da270aa256a33571f9c7L303 or https://github.com/opencv/opencv/pull/22368/files#diff-76029e66ed661b9eff5df27ddeb6eea0bb9a80179bfc80a07af64fa1048cfb8eL184 .

See https://github.com/robotology/robotology-superbuild/issues/1328 .

traversaro, 2023-01-23:
@GiulioRomualdi I am not sure about the fix, probably we could just pin opencv to <=4.6 until we fix it?

traversaro, 2023-01-26:
Probabily this also affects assistive-rehab, see https://github.com/robotology/assistive-rehab/blob/da865a96f8bf5f80a2f0fab8ca9e9a8bc0fcae8e/modules/lineDetector/src/main.cpp#L147 . fyi @pattacini

pattacini, 2023-01-26:
Thanks @traversaro for pointing it out.
We're using `opencv-4.6.0`:
- https://github.com/robotology/assistive-rehab/blob/master/.github/workflows/ci.yml#L127-L128.

cc @mfussi66 @ste93 @elandini84 @randaz81

closed 2023-01-31 by GiulioRomualdi