gbionics/bipedal-locomotion-framework #629: Install python bindings

CarlottaSartore, 2023-03-17:
I had installed the bipedal locomotion framework via conda : 

`conda install -c conda-forge bipedal-locomotion-framework`

But when I import as  `import bipedal_locomotion_framework.bindings as blf` I got the following error: 

```
  File "/home/carlotta/miniforge3/envs/hardware_control_env/lib/python3.8/site-packages/bipedal_locomotion_framework/__init__.py", line 1, in <module>
    from .bindings import *
ImportError: arg(): could not convert default argument 'mixed_velocity: manif::SE3Tangent<double>' in method '<class 'bipedal_locomotion_framework.bindings.ik.SE3Task'>.set_set_point' into a Python object (type not registered yet?)
```

output of `conda list | grep bipedal-locomotion`

```
bipedal-locomotion-framework 0.12.0               ha770c72_1    conda-forge
bipedal-locomotion-framework-python 0.12.0           py38he384494_1    conda-forge
libbipedal-locomotion-framework 0.12.0               h7f54790_1    conda-forge
```
output of  `conda list | grep manif`

```
manif                     0.0.4               hcb278e6_12    conda-forge
manifpy                   0.0.4                    pypi_0    pypi
```

C.C. @GiulioRomualdi @traversaro

traversaro, 2023-03-17:
That sounds like the usual problem related to manifpy and blf being compiled with two different compilers (see https://github.com/conda-forge/manif-feedstock/issues/19), as this is not trivial to solve by adding constraints in conda, we should at least add a downstream CI check that python bindings installed by conda work as expected.

traversaro, 2023-03-17:
Actually, I think this is not correct: both the latest manifpy and the latest bipedal-locomotion-framework-python are compiled with GCC 11, see https://github.com/conda-forge/bipedal-locomotion-framework-feedstock/blob/78ff359d413ec67917adb34b05e7baab44802466/.ci_support/linux_64_.yaml#L14 and https://github.com/conda-forge/manif-feedstock/blob/b5df91dbc2793a16e9aa8d078503ee8d0be88d28/.ci_support/linux_64_.yaml#L14 . Something else is happening here.

traversaro, 2023-03-17:
I think there is something fishy happening with `manifpy`. When I install it, I get that this version is installed:
~~~
  + manifpy                                     0.0.4  py311h0c9b314_12     conda-forge/linux-64       14MB
~~~

but then when doing conda list:
~~~
(test629) traversaro@IITICUBLAP257:~$ mamba list | grep manifpy
manifpy                   0.0.4                    pypi_0    pypi
(test629) traversaro@IITICUBLAP257:~$ conda list | grep manifpy
manifpy                   0.0.4                    pypi_0    pypi
~~~

The build string is completly different.

traversaro, 2023-03-17:
In my specific case, anyhow, `import bipedal_locomotion_framework.bindings as blf` works correctly. @CarlottaSartore can you share the complete output of `conda list` and `conda info` ? Thanks!

traversaro, 2023-03-17:
Ok, apparently the conda-forge manifpy has something deeply wroong: apparently we ship a full `numpy` package inside of it, and that may be the reason for the strange version, indeed if we run ` list --verbose`, there is a warning:
~~~
INFO conda.core.prefix_data:get_conda_anchor_files_and_records(363): Package conda-forge/linux-64::manifpy-0.0.4-py38h35dd0ef_12 has multiple python anchor files.
  Using lib/python3.8/site-packages/numpy-1.24.1.dist-info/RECORD
~~~

I am not sure if this is the reason of your problems, but for sure it is definitely not good.

traversaro, 2023-03-17:
Can you try to run `mamba updgrade manifpy`?

CarlottaSartore, 2023-03-17:
```
Collecting package metadata (current_repodata.json): done
Solving environment: done

==> WARNING: A newer version of conda exists. <==
  current version: 4.12.0
  latest version: 23.1.0

Please update conda by running

    $ conda update -n base conda

## Package Plan ##

  environment location: /home/carlotta/miniforge3/envs/hardware_control_env

  added / updated specs:
    - manifpy

The following packages will be downloaded:

    package                    |            build
    ---------------------------|-----------------
    irrlicht-1.8.5             |       h3749e40_3         1.8 MB  conda-forge
    jack-1.9.22                |       h11f4161_0         453 KB  conda-forge
    jasper-4.0.0               |       h32699f2_1         630 KB  conda-forge
    libjpeg-turbo-2.1.5.1      |       h0b41bf4_0         479 KB  conda-forge
    libopencv-4.7.0            |   py38h1bd34d7_2        29.1 MB  conda-forge
    libtiff-4.5.0              |       hddfeb54_5         397 KB  conda-forge
    libudev1-253               |       h0b41bf4_0         116 KB  conda-forge
    libwebp-base-1.3.0         |       h0b41bf4_0         348 KB  conda-forge
    libxml2-2.10.3             |       hca2bb57_3         698 KB  conda-forge
    manif-0.0.4                |      hcb278e6_13          53 KB  conda-forge
    manifpy-0.0.4              |  py38h35dd0ef_13         433 KB  conda-forge
    qt-main-5.15.8             |       h67dfc38_7        50.0 MB  conda-forge
    robot-testing-framework-2.0.1|      h9c3ff4c_13         207 KB  robotology
    sdl2-2.26.4                |       h949db6a_0         1.3 MB  conda-forge
    setuptools-67.6.0          |     pyhd8ed1ab_0         566 KB  conda-forge
    wheel-0.40.0               |     pyhd8ed1ab_0          54 KB  conda-forge
    xorg-libx11-1.8.4          |       h0b41bf4_0         810 KB  conda-forge
    xorg-libxext-1.3.4         |       h0b41bf4_2          49 KB  conda-forge
    xorg-xextproto-7.3.0       |    h0b41bf4_1003          30 KB  conda-forge
    xorg-xf86vidmodeproto-2.3.1|    h7f98852_1002          23 KB  conda-forge
    yarp-cxx-3.8.0             |       h5ee7595_1         9.2 MB  conda-forge
    ycm-cmake-modules-0.12.2   |      h9c3ff4c_13         136 KB  robotology
    ------------------------------------------------------------
                                           Total:        96.7 MB

The following NEW packages will be INSTALLED:

  xorg-xf86vidmodep~ conda-forge/linux-64::xorg-xf86vidmodeproto-2.3.1-h7f98852_1002

The following packages will be REMOVED:

  asttokens-2.2.1-pyhd8ed1ab_0
  backcall-0.2.0-pyh9f0ad1d_0
  backports-1.0-pyhd8ed1ab_3
  backports.functools_lru_cache-1.6.4-pyhd8ed1ab_0
  decorator-5.1.1-pyhd8ed1ab_0
  executing-1.2.0-pyhd8ed1ab_0
  ipython-8.8.0-pyh41d4057_0
  jedi-0.18.2-pyhd8ed1ab_0
  jpeg-9e-h166bdaf_2
  lcms2-2.14-hfd0df8a_1
  libsodium-1.0.18-h36c2ea0_1
  matplotlib-inline-0.1.6-pyhd8ed1ab_0
  meshcat-python-0.3.2-pyhd8ed1ab_0
  openjpeg-2.5.0-hfec8fc6_2
  parso-0.8.3-pyhd8ed1ab_0
  pexpect-4.8.0-pyh1a96a4e_2
[... 50 lines cut ...]
yarp-cxx-3.8.0       | 9.2 MB    | ############################################################################################# | 100% 
xorg-xextproto-7.3.0 | 30 KB     | ############################################################################################# | 100% 
libudev1-253         | 116 KB    | ############################################################################################# | 100% 
xorg-xf86vidmodeprot | 23 KB     | ############################################################################################# | 100% 
irrlicht-1.8.5       | 1.8 MB    | ############################################################################################# | 100% 
robot-testing-framew | 207 KB    | ############################################################################################# | 100% 
jack-1.9.22          | 453 KB    | ############################################################################################# | 100% 
qt-main-5.15.8       | 50.0 MB   | ############################################################################################# | 100% 
xorg-libx11-1.8.4    | 810 KB    | ############################################################################################# | 100% 
jasper-4.0.0         | 630 KB    | ############################################################################################# | 100% 
xorg-libxext-1.3.4   | 49 KB     | ############################################################################################# | 100% 
libjpeg-turbo-2.1.5. | 479 KB    | ############################################################################################# | 100% 
libopencv-4.7.0      | 29.1 MB   | ############################################################################################# | 100% 
setuptools-67.6.0    | 566 KB    | ############################################################################################# | 100% 
ycm-cmake-modules-0. | 136 KB    | ############################################################################################# | 100% 
sdl2-2.26.4          | 1.3 MB    | ############################################################################################# | 100% 
libxml2-2.10.3       | 698 KB    | ############################################################################################# | 100% 
libwebp-base-1.3.0   | 348 KB    | ############################################################################################# | 100% 
manifpy-0.0.4        | 433 KB    | ############################################################################################# | 100% 
wheel-0.40.0         | 54 KB     | ############################################################################################# | 100% 
Preparing transaction: done
Verifying transaction: done
Executing transaction: done

```

CarlottaSartore, 2023-03-17:
I have re-installed: 
- pillow
- six
- yaml
- pyyaml

Still not able to load the bindings. 

Anyhow I am not sure everything went correctly, even though all the packages have been disintalled (as expected ?) the version is still 
`manifpy                   0.0.4           py38h35dd0ef_13    conda-forge`

<details><summary> Conda list</summary>
<p>

```
# packages in environment at /home/carlotta/miniforge3/envs/hardware_control_env:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
ace                       7.1.0                hcb278e6_0    conda-forge
adam                      0.0.1                    pypi_0    pypi
adam-robotics             0.0.7.dev25+dirty          pypi_0    pypi
alsa-lib                  1.2.8                h166bdaf_0    conda-forge
ampl-mp                   3.1.0             h2cc385e_1006    conda-forge
aom                       3.5.0                h27087fc_0    conda-forge
assimp                    5.2.5                hf40c2ba_0    conda-forge
attr                      2.5.1                h166bdaf_1    conda-forge
binutils_impl_linux-64    2.39                 he00db2b_1    conda-forge
binutils_linux-64         2.39                h5fc0e48_11    conda-forge
bipedal-locomotion-framework 0.12.0               ha770c72_1    conda-forge
bipedal-locomotion-framework-python 0.12.0           py38he384494_1    conda-forge
bitarray                  2.7.0                    pypi_0    pypi
boost-cpp                 1.78.0               h75c5d50_1    conda-forge
bzip2                     1.0.8                h7f98852_4    conda-forge
c-ares                    1.18.1               h7f98852_0    conda-forge
ca-certificates           2023.01.10           h06a4308_0    anaconda
cairo                     1.16.0            ha61ee94_1014    conda-forge
casadi                    3.5.5           py38h063f47d_17    conda-forge
cppad                     20230000.0           hcb278e6_0    conda-forge
dbus                      1.13.6               h5008d03_3    conda-forge
eigen                     3.4.0                h4bd325d_0    conda-forge
entrypoints               0.4                      pypi_0    pypi
expat                     2.5.0                h27087fc_0    conda-forge
ffmpeg                    5.1.2           gpl_h8dda1f0_106    conda-forge
fftw                      3.3.10          nompi_hf0379b8_106    conda-forge
fmt                       9.1.0                h924138e_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.14.2               h14ed4e7_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
freeglut                  3.2.2                h9c3ff4c_1    conda-forge
freetype                  2.12.1               hca18f0e_1    conda-forge
gcc_impl_linux-64         12.2.0              hcc96c02_19    conda-forge
gcc_linux-64              12.2.0              h4798a0e_11    conda-forge
gettext                   0.21.1               h27087fc_0    conda-forge
glib                      2.74.1               h6239696_1    conda-forge
glib-tools                2.74.1               h6239696_1    conda-forge
gmp                       6.2.1                h58526e2_0    conda-forge
gnutls                    3.7.8                hf3e180e_0    conda-forge
graphite2                 1.3.13            h58526e2_1001    conda-forge
gst-plugins-base          1.22.0               h4243ec0_2    conda-forge
gstreamer                 1.22.0               h25f0c4b_2    conda-forge
gstreamer-orc             0.4.33               h166bdaf_0    conda-forge
gxx_impl_linux-64         12.2.0              hcc96c02_19    conda-forge
gxx_linux-64              12.2.0              hb41e900_11    conda-forge
harfbuzz                  6.0.0                h8e241bc_0    conda-forge
[... 154 lines cut ...]
xorg-fixesproto           5.0               h7f98852_1002    conda-forge
xorg-inputproto           2.3.2             h7f98852_1002    conda-forge
xorg-kbproto              1.0.7             h7f98852_1002    conda-forge
xorg-libice               1.0.10               h7f98852_0    conda-forge
xorg-libsm                1.2.3             hd9c2040_1000    conda-forge
xorg-libx11               1.8.4                h0b41bf4_0    conda-forge
xorg-libxau               1.0.9                h7f98852_0    conda-forge
xorg-libxdmcp             1.1.3                h7f98852_0    conda-forge
xorg-libxext              1.3.4                h0b41bf4_2    conda-forge
xorg-libxfixes            5.0.3             h7f98852_1004    conda-forge
xorg-libxi                1.7.10               h7f98852_0    conda-forge
xorg-libxrender           0.9.10            h7f98852_1003    conda-forge
xorg-renderproto          0.11.1            h7f98852_1002    conda-forge
xorg-xextproto            7.3.0             h0b41bf4_1003    conda-forge
xorg-xf86vidmodeproto     2.3.1             h7f98852_1002    conda-forge
xorg-xproto               7.0.31            h7f98852_1007    conda-forge
xz                        5.2.6                h166bdaf_0    conda-forge
yaml                      0.2.5                h7b6447c_0    anaconda
yarp-cxx                  3.8.0                h5ee7595_1    conda-forge
ycm-cmake-modules         0.12.2              h9c3ff4c_13    robotology
zlib                      1.2.13               h166bdaf_4    conda-forge
zstd                      1.5.2                h3eb15da_6    conda-forge
```
  </p>
</details>

CarlottaSartore, 2023-03-17:
The error is still 

```
ImportError: arg(): could not convert default argument 'mixed_velocity: manif::SE3Tangent<double>' in method '<class 'bipedal_locomotion_framework.bindings.ik.SE3Task'>.set_set_point' into a Python object (type not registered yet?)

```

traversaro, 2023-03-17:
This may be related to https://github.com/conda-forge/staged-recipes/pull/20432#issuecomment-1291504197 .