gbionics/bipedal-locomotion-framework #636: CI Failure in ubuntu 20.04-apt March 2023

GiulioRomualdi, 2023-03-27:
In all the latest PR the ubuntu 20.04-apt CI fails with the following error:

```
2023-03-27T08:37:04.7308490Z [ 75%] Building CXX object bindings/python/CMakeFiles/pybind11_blf.dir/bipedal_locomotion_framework.cpp.o
2023-03-27T08:37:07.2958342Z [ 76%] Building CXX object bindings/python/CMakeFiles/pybind11_blf.dir/System/src/Advanceable.cpp.o
2023-03-27T08:37:08.8024829Z In file included from /home/runner/work/bipedal-locomotion-framework/bipedal-locomotion-framework/src/ParametersHandler/include/BipedalLocomotion/ParametersHandler/IParametersHandler.h:16,
2023-03-27T08:37:08.8025924Z                  from /home/runner/work/bipedal-locomotion-framework/bipedal-locomotion-framework/src/System/include/BipedalLocomotion/System/Advanceable.h:13,
2023-03-27T08:37:08.8026941Z                  from /home/runner/work/bipedal-locomotion-framework/bipedal-locomotion-framework/bindings/python/System/include/BipedalLocomotion/bindings/System/Advanceable.h:17,
2023-03-27T08:37:08.8027878Z                  from /home/runner/work/bipedal-locomotion-framework/bipedal-locomotion-framework/bindings/python/System/src/Advanceable.cpp:13:
2023-03-27T08:37:08.8029547Z /home/runner/work/bipedal-locomotion-framework/bipedal-locomotion-framework/src/GenericContainer/include/BipedalLocomotion/GenericContainer/TemplateHelpers.h:212:43: error: expected template-name before ‘<’ token
2023-03-27T08:37:08.8046238Z   212 | struct is_string : public std::disjunction<std::is_same<char*, typename std::decay<T>::type>,
2023-03-27T08:37:08.8047296Z       |                                           ^
2023-03-27T08:37:08.8048334Z /home/runner/work/bipedal-locomotion-framework/bipedal-locomotion-framework/src/GenericContainer/include/BipedalLocomotion/GenericContainer/TemplateHelpers.h:212:43: error: expected ‘{’ before ‘<’ token
2023-03-27T08:37:08.8049958Z /home/runner/work/bipedal-locomotion-framework/bipedal-locomotion-framework/src/GenericContainer/include/BipedalLocomotion/GenericContainer/TemplateHelpers.h:224:16: error: ‘string_view’ in namespace ‘std’ does not name a type
2023-03-27T08:37:08.8053034Z   224 | constexpr std::string_view
2023-03-27T08:37:08.8054150Z       |                ^~~~~~~~~~~
```

Here is the original error: https://github.com/ami-iit/bipedal-locomotion-framework/actions/runs/4510249345/jobs/7940979417?pr=634#step:21:1013

@traversaro as far as I understood the superbuild works fine: https://github.com/robotology/robotology-superbuild/actions/runs/4527974198/jobs/7974321290

traversaro, 2023-03-27:
> @traversaro as far as I understood the superbuild works fine: https://github.com/robotology/robotology-superbuild/actions/runs/4527974198/jobs/7974321290

Not really. The Ubuntu 20.04 on Docker (so with the CMake 3.16.3) is working fine, while the Ubuntu 20.04 on the GitHub Runner image is failing due to https://github.com/robotology/robotology-superbuild/issues/1366, so I am not arriving at the point where I could be experiencing that problem.

traversaro, 2023-03-27:
Furthermore, note that this is failing only when we explicitly disable the YARP dependency (`Testing YARP as missing dependency.`), so it is possible that on the superbuild everything works fine as we do not disable the YARP dependency.

GiulioRomualdi, 2023-03-28:
Fixed by https://github.com/ami-iit/bipedal-locomotion-framework/pull/635

closed 2023-03-28 by GiulioRomualdi