gbionics/bipedal-locomotion-framework #651: Add IK align vector direction task

lrapetti, 2023-04-13:
The objective is to add a task in the IK that minimizes the distance between the direction of two vectors (we will consider them being unit vector for now).

Let's consider that we have a desired direction $^{I}u^{\*}$ expressed in the inertia, and a local direction into one of the frame of the model $^{L}u$. Given the frame orientation $^{I}R_{L}$, we can get the current direction in the inertia $^{I}u=^{I}R_{L} \cdot ^{L}u$. 

In order to follow the IK approach implemented in https://github.com/ami-iit/bipedal-locomotion-framework/tree/master/src/IK, we should define a velocity controller that align the vector. What we need to determine is indeed a frame velocity $\omega_L$ control input for the system.

I opened this issue in order to discuss how to implement this task.

I start by proposing two approaches

### Approach A: S2 controller
Note: 
- $S^2$ is the 2-sphere group to which the 3D versor belongs ($x\in \mathbb{R}^3 , \|x\|=1$)
- $S^2$ group is [not a Lie group](https://math.berkeley.edu/~seewoo5/S2notLiegrp.pdf) (so we cannot add it to the [lie-group-controllers](https://github.com/ami-iit/lie-group-controllers))

A simple way to achieve this could be to define the angular velocity as follow:
$\omega = - K \cdot (u^{\*} \times \  u )$

### Approach B: SO(3) controller
A second approach that exploits the already existing SO(3) controller consists in first finding the rotation that aligns the vectors ($^{I}R_{L}^{*}$) and then using SO(3) controller (the rotation $^{I}R_{L}^{*}$ can be easily found from $u^{\*}$ and $u$

(for both the approaches we can also have the possibility to add a feedforward term)

lrapetti, 2023-04-13:
cc @RiccardoGrieco @S-Dafarra @GiulioRomualdi

EhsanRanjbari, 2023-04-17:
Here are the photos of the math discussed in the meeting:

[image]

[image]

cc @DanielePucci @S-Dafarra @GiulioRomualdi @mebbaid @lrapetti

EhsanRanjbari, 2023-07-11:
I tried to rewrite the math above as follows:
 
We are interested to align the gravity vector (reading from IMU accelerometer data) with the gravity vector in the world inertial frame.  The reason is to assume these two vectors are not aligned. This means that the body frame should rotate in a way to align with the gravity vector and it simply means tracking the finger motion in that direction to fulfill that alignment.

Let's denote the measured gravity vector in the body frame, $^B{V_d}$,
and the gravity vector in the world inertial frame, $^W{Z}$. Since the gravity in that frame is always in the Z-axis direction.
The control objective is $^W{Z} \rightarrow W{V_d}$.
Let's also denote the rotational transformation from the body frame to the inertial frame as $^W{R_B} = (^W{x}, ^W{y}, ^W{z})$. This follows,

$$
^W{Z} = ^W{R_B} . ^B{Z}
$$

$$
^W{V_d} = ^W{R_B} . ^B{V_d}
$$

To characterize the rotational kinematics of the robot, we need to compute the time derivative of the rotational matrix, $^W{R_B}$ as,

$$
^W{\dot{R_B}} = S(^W{\omega}) ^W{R_B},
$$

where, $S(\omega)$ is the skew-symmetric operator.
We assume $\theta$ is the angle between these two vectors. See the figure below.

<div align="center">

[image]

</div>

In order to follow the BLF approach, we need to compute a velocity input (angular velocity) that aligns these two vectors or in other words, $\theta \rightarrow 0$.
Basically, we need to rotate the vector $^W{V_d}$ around an axis that is perpendicular to both $^W{V_d}$ and $^W{Z}$. We know that the cross product of two vectors is such a vector that in fact,

$$
^W{Z} \times ^W{V_d} = |^W{Z}||^W{V_d}|\sin{\theta}.
$$

We choose a Lyapunov function as below,

$$
V = 1 - \cos{\theta}.
$$

For computing the time-derivative of it we can either use the dot product equation,

$$
^W{Z} . ^W{V_d} = |^W{Z}||^W{V_d}|\cos{\theta},
$$

and substitute $\cos{\theta}$ and compute the derivate or using the cross product as follows,

$$
\dot{V} = \dot{\theta}\sin{\theta} = ^W{\omega}\sin{\theta}.
$$

where, $^W{\omega}$, is the angular velocity of the body frame in the inertial frame,

$$
^W{\omega} = ^W{R_B}^B{\omega}.
$$

From the cross product above equation, we substitute the $\sin{\theta}$, we have,

$$
\dot{V} = (\frac{|^W{Z} \times ^W{V_d}|}{|^W{Z}||^W{V_d}|})^W{\omega}.
$$

To simplify this equation, we use the fact that

$$
|^W{Z} \times ^W{V_d}| = S(^W{Z})^W{V_d},
$$

so, the equation becomes,

$$
\dot{V} = S(Z)\hat{V_d}{\omega} = -(S(Z)\hat{V_d})^T{\omega}\leq0.
$$

Now, by choosing the control signal, $^W{\omega}$, as, $^W{\omega} = k (S(Z)\hat{V_d}) + \lambda ^W{Z}$ with $\lambda \in \mathbb{R}$.

At this point, we need to know the part of vector $^W{\omega}$ is projected onto the $x$ and $y$ axes. To compute simply we just need to compute the dot product of the unit vector of the axis with the vectors $^W{\omega}$. We have,

$$
^W{x}^T  ^W{\omega} = k ^W{x}^T  S(Z)\hat{V_d} + \lambda ^W{x^T}^W{Z},
$$

and on the $y$ axis,

$$
^W{y}^T  ^W{\omega} = k ^W{y}^T  S(Z)\:\hat{V_d} + \lambda ^W{y^T}^W{Z}
$$

we know, $^W{x}^T  S(^W{Z}) = -^W{y}^T$ and $^W{y}^T  S(^W{Z}) = ^W{x}^T$. Also, $^W{x^T}^W{Z} = 0$, and $^W{y^T}^W{Z} = 0$.
So we can rewrite it as,

$$
\left[\begin{array}.
1 & 0 & 0\\
0 & 0 & 1
\end{array}\right] \quad^W{\omega} = -k \left[\begin{array}.
0 & 1 & 0\\
1 & 0 & 0
\end{array}\right]\quad^W{\hat{V_d}}.
$$

We know $\omega = J_w  \dot{q}$ so we have,

$$
\left[\begin{array}.
1 & 0 & 0\\
0 & 0 & 1
\end{array}\right] J_w \dot{q} = -k \left[\begin{array}.
											0 & 1 & 0\\
											1 & 0 & 0
										\end{array}\right]\quad^W{\hat{V_d}}.
$$

So, in compliance with QP, $Ax = b$,

$$
\begin{array}.
A = \left[\begin{array}.
1 & 0 & 0 \\
0 & 0 & 1
\end{array}\right] J_w  \dot{q},\\
\\
b = \left[\begin{array}.
		0 & 1 & 0\\
		1 & 0 & 0\\
		\end{array}\right]\quad^W{\hat{V_d}}.
\end{array}
$$

where $J_w$ is the part associated with the angular velocities of the body frame (finger-tip),

$$
J =\left[\begin{array}.
J_v\\
J_w 
\end{array}\right].
$$

EhsanRanjbari, 2023-07-11:
Here is the preliminary implementation of the above math:

- Gravity Task from BLF: https://github.com/ami-iit/bipedal-locomotion-framework/commit/ebb7db32ff84f11062219b1fd176b0e3e2b900c8
- WeArtGloveRawData from the device: https://github.com/ami-iit/element_haptic-gloves/commit/4ca243140e2cb7c5f0a3e7a476cd6cf8059b7443
cc @S-Dafarra

S-Dafarra, 2023-08-03:
Today I gave another round of thoughts on this issue, and I realized that what we need is a bit different from what we wrote on the whiteboard in https://github.com/ami-iit/bipedal-locomotion-framework/issues/651#issuecomment-1511220042

In particular, rather than aligning the Z-axis of a frame to a given direction, we want a given direction expressed in body coordinates to be aligned with the world Z-axis, i.e. the opposite. 

In fact, suppose to have an accelerometer on a given body. This sensor measures the gravity in the body frame. Hence, we aim to apply an angular velocity to a random frame, such that the accelerometer measurement, when expressed in this frame, is aligned to the world Z-axis. In this way, we estimate the orientation of the accelerometer sensor (minus the yaw).

In symbols, defines ${}^wR_b$ the orientation of the frame $b$ we want to move, and ${}^b\hat{V}$ the unitary vector fixed in the frame $b$ that we want to align to ${}^wz = e_3 = [ 0 ~0 ~1]^\top$.

We define the following Lyapunov function

$$
V = 1 - \cos(\theta) \geq 0.
$$

Since $a^\top b = ||a|| ~||b||\cos(\theta)$ we have

$$
V = 1 - \left({}^wR_b{}^b\hat{V}\right)^\top e_3,
$$

given that both vectors are unitary.

The time derivative of the Lyapunov function is as follows

$$
\dot{V} = - \left({}^w\dot{R}_b{}^b\hat{V}\right)^\top e_3,
$$

where ${}^w\dot{R}_b = S({}^w\omega){}^w{R}_b$, with $S(\cdot)$ a skew symmetric matrix, and ${}^w\omega \in \mathbb{R}^3$ the right trivialized angular velocity. Hence,

$$
\dot{V} = - \left(S({}^w\omega){}^w{R}_b{}^b\hat{V}\right)^\top e_3.
$$

Exploiting $S(\cdot)^\top = -S(\cdot)$, we have

$$
\dot{V} = {}^b\hat{V}^\top{}^w{R}_b^\top  S({}^w\omega) e_3,
$$

and $S(a)b = -S(b)a$,

$$
\dot{V} = -{}^b\hat{V}^\top{}^w{R}_b^\top  S(e_3){}^w\omega.
$$

Then, if we choose

$$
{}^w\omega =k \left({}^b\hat{V}^\top{}^w{R}_b^\top  S(e_3)\right)^\top + \lambda e_3
$$

with $k, \lambda \in \mathbb{R}^3$, we have that $\dot{V} \leq 0$. Then,

$$
{}^w\omega = -k S(e_3) {}^w{R}_b{}^b\hat{V} + \lambda e_3.
$$

:warning: Note the sign difference!

We can introduce the right-trivialized Jacobian ${}^wJ$ such that ${}^wJ\dot{\nu} = {}^w\omega$,

$$
{}^wJ\dot{\nu} =  -k S(e_3) {}^w{R}_b{}^b\hat{V} + \lambda e_3.
$$

We can filter out the rotation around the world Z-axis, thus obtaining

$$
\begin{bmatrix}
e_1 & e_2
\end{bmatrix}^\top {}^wJ\dot{\nu} = -k \begin{bmatrix}
                                                                                   0 & -1 & 0\\
                                                                                   1 & 0 & 0 \end{bmatrix} {}^w{R}_b{}^b\hat{V}
$$

##

My understanding is that the sign difference is because we are not trying to align the Z axis of a frame to a given direction in world coordinates (case A), but rather we are aligning a vector in body coordinates to the world Z axis (case B).

##

@EhsanRanjbari reading again the math of https://github.com/ami-iit/bipedal-locomotion-framework/issues/651#issuecomment-1630812830 it seems coherent with case A, but the sentence

> We are interested to align the gravity vector (reading from IMU accelerometer data) with the gravity vector in the world inertial frame. The reason is to assume these two vectors are not aligned. [...]

seems to describe case B.
 In the past days I have modified a bunch of signs cause they were not coherent with the passages above, but maybe I introduced mistakes :blush:

S-Dafarra, 2023-08-03:
I also tried to analyze how the math is implemented in ``HumanDynamicsEstimation``. @lrapetti please correct me if I did some mistake

The desired angular velocity (right trivialized, since it is using the default representation of ``KinDynComputations``) is computed [here](https://github.com/robotology/human-dynamics-estimation/blob/d89cc08899f174d550ddbfe04ea13621235c8748/HumanDynamicsEstimationLibrary/algorithms/src/DynamicalInverseKinematics.cpp#L993-L1005) in the following form (forgetting about the feedforward)

$$
\omega = -k ~ e
$$ 

where $e$ is the error. This is computed [here](https://github.com/robotology/human-dynamics-estimation/blob/d89cc08899f174d550ddbfe04ea13621235c8748/HumanDynamicsEstimationLibrary/algorithms/src/DynamicalInverseKinematics.cpp#L629-L637) in the following form:

$$
e = {}^wR_b \left({}^wR_b^\top e_3\right) \times \left(\hat{R}^\top e_3\right)
$$

where ${}^wR_b^\top e_3$ selects the third row of the frame orientation, input [here](https://github.com/robotology/human-dynamics-estimation/blob/d89cc08899f174d550ddbfe04ea13621235c8748/HumanDynamicsEstimationLibrary/algorithms/src/DynamicalInverseKinematics.cpp#L993). Similarly, $\hat{R}^\top e_3$ selects the third row of the sensor measured rotation, filled [here](https://github.com/robotology/human-dynamics-estimation/blob/d89cc08899f174d550ddbfe04ea13621235c8748/devices/HumanStateProvider/HumanStateProvider.cpp#L2300-L2306) (actually, in this line, also a calibration matrix is added, the actual raw data arrives [here](https://github.com/robotology/human-dynamics-estimation/blob/d89cc08899f174d550ddbfe04ea13621235c8748/devices/HumanStateProvider/HumanStateProvider.cpp#L1827-L1832) from the actual sensors).
It is worth noting that $\hat{R}^\top e_3 = {}^b\hat{V}$. In fact, the third row of the sensor rotation indicates the world Z-axis measured in the sensor frame, i.e. the gravity direction, that is exactly what an accelerometer would measure. Hence,

$$
\omega = -k {}^wR_b S\left({}^wR_b^\top e_3\right) {}^b\hat{V}.
$$ 

Now, we can apply the *rotational equivalence of cross products*:

$$
S\left({}^CR_D {}^D\alpha\right) = {}^CR_D S({}^D\alpha) {}^CR_D^\top
$$

which results in 

$$
\omega = -k {}^wR_b {}^wR_b^\top S\left(e_3\right) {}^wR_b{}^b\hat{V} = -k S\left(e_3\right) {}^wR_b{}^b\hat{V}
$$ 

that is corresponding to what I wrote in https://github.com/ami-iit/bipedal-locomotion-framework/issues/651#issuecomment-1663790036.

GiulioRomualdi, 2023-11-02:
Closing thanks to https://github.com/ami-iit/bipedal-locomotion-framework/pull/717

closed 2023-11-02 by GiulioRomualdi