gbionics/bipedal-locomotion-framework #674: conda-forge CI failing on Linux May 2023

traversaro, 2023-05-17:
See https://github.com/ami-iit/bipedal-locomotion-framework/actions/workflows/conda-forge-ci.yml?query=event%3Aschedule , it has been falling for ~3 weeks as of the day I open the issue (I have to figure out how to get notifications for CI failing in blf).

traversaro, 2023-05-17:
> (I have to figure out how to get notifications for CI failing in blf)

From https://docs.github.com/en/actions/monitoring-and-troubleshooting-workflows/notifications-for-workflow-runs :

> Notifications for scheduled workflows are sent to the user who initially created the workflow. If a different user updates the cron syntax in the workflow file, subsequent notifications will be sent to that user instead. If a scheduled workflow is disabled and then re-enabled, notifications will be sent to the user who re-enabled the workflow rather than the user who last modified the cron syntax.

That is strange, can't we have two users getting the notification?

traversaro, 2023-05-17:
Indeed, the CI failed when manifpy was updated from _13 to _14 .

traversaro, 2023-05-17:
Ahh, now I remember, it was supposed to be fixed by https://github.com/conda-forge/compilers-feedstock/pull/57, but that PR got stuck.

traversaro, 2023-05-17:
For reference the pybind11 problem is https://github.com/conda-forge/pybind11-feedstock/issues/77 .

GiulioRomualdi, 2023-05-18:
Since https://github.com/ami-iit/bipedal-locomotion-framework/pull/675 fixed the issue, I think we can close it right?

traversaro, 2023-05-18:
Yes!

closed 2023-05-18 by traversaro

traversaro, 2023-07-11:
> Ahh, now I remember, it was supposed to be fixed by [conda-forge/compilers-feedstock#57](https://github.com/conda-forge/compilers-feedstock/pull/57), but that PR got stuck.

The PR was finally merged, so probably we can remove the workaround added in https://github.com/ami-iit/bipedal-locomotion-framework/pull/675 .