gbionics/bipedal-locomotion-framework #696: Support ProxQP  OSQP to solve QPIK and TSID QP problems

traversaro, 2023-06-30:
It would be interesting to support the ProxQP solver (https://github.com/Simple-Robotics/proxsuite) when solving QP problems created by QPIK and TSID. 

This would probably require:
* Abstract existing calls to osqp-eigen under an abstraction layer (let's call it BLFQP)
* Add support in BLFQP for ProxQP, either by directly writing the integration with ProxQP or by interfacing with an existing abstraction layer for QP programs, such as https://github.com/isri-aist/QpSolverCollection or CasADi .

GiulioRomualdi, 2023-07-01:
I think having an abstraction layer for the solvers is doable.

traversaro, 2023-07-01:
For reference, the abstraction layer used in `stack-of-tasks/tsid` is https://github.com/stack-of-tasks/tsid/blob/v1.7.0/include/tsid/solvers/solver-HQP-base.hpp .

traversaro, 2023-08-22:
Other recent QP solvers that report intersting performances on benchmarks:
* https://github.com/oxfordcontrol/Clarabel.cpp
* https://github.com/PREDICT-EPFL/piqp