gbionics/bipedal-locomotion-framework #710: building from an earlier fork/commit broken:  `qh_version2' needs to compile with fPIC

mebbaid, 2023-07-20:
I have tried to build the `bipedal` framework against a branch in my fork in the `ergocub-torso`. When i do `make install`, the following error is thrown

```
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libqhullstatic_r.a(global_r.c.o): warning: relocation against `qh_version2' in read-only section `.text'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libqhullstatic_r.a(global_r.c.o): relocation R_X86_64_PC32 against symbol `qh_version2' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: final link failed: bad value
```

Things to note:

- This issue appeared after the recent superbuild update on the torso
- On my machine, this error is not reproducible. 
- this error is triggered by either TSID or the Planners 

The status of the robotologysuperbuild repos in  machine is as follows 

<details>
  <summary>Click me</summary>
  
  ### My machine status
  
```
-------------------------------------------
/home/mebbaid/robot-code/robotology-superbuild: master
2 months ago|Merge pull request #1406 from robotology/traversaro-patch-3|3590ac25f8928d19fb50a0df2c1a0002b6051819
## master...origin/master
--------------------------------------------
/home/mebbaid/robot-code/robotology-superbuild/src/bipedal-locomotion-framework/: HEAD
4 months ago|fix CMakeLists.txt copy /paste mistake|c50812d94f5ffbb5672af7d37883fec75583ee39
## HEAD (no branch)
?? .vscode/
?? src/ReducedModelControllers/include/BipedalLocomotion/ReducedModelControllers/AdaptiveCentroidalMPC.h
?? src/ReducedModelControllers/src/AdaptiveCentroidalMPC.cpp
?? src/ReducedModelControllers/tests/AdaptiveCentroidalMPCTest.cpp
--------------------------------------------
/home/mebbaid/robot-code/robotology-superbuild/src/BlockFactory/: HEAD
1 year, 5 months ago|Add new Matlab version to MATLAB_VERSIONS_MAPPING in FindMatlab (#66)|11b9a6ed18ca5b67fec690e2de1c48bfcc719dd8
## HEAD (no branch)
--------------------------------------------
/home/mebbaid/robot-code/robotology-superbuild/src/casadi/: HEAD
2 years, 3 months ago|Merge tag '3.5.5' into support_osqp_0.6.2|07e196b4f35cfaec28e2fd42c29dbda459b0a56c
## HEAD (no branch)
--------------------------------------------
/home/mebbaid/robot-code/robotology-superbuild/src/casadi-matlab-bindings/: HEAD
1 year, 5 months ago|Bump version to 3.5.5.2 (#8)|f43e8062bec91c90fa0cbd2c98ad830db47857b8
## HEAD (no branch)
--------------------------------------------
/home/mebbaid/robot-code/robotology-superbuild/src/CppAD/: HEAD
1 year, 2 months ago|stable/20220000: Advance to version 20220000.4.|9aeef77084458a87b3c672b028cd2868b55ee790
## HEAD (no branch)
--------------------------------------------
/home/mebbaid/robot-code/robotology-superbuild/src/GazeboYARPPlugins/: honda_proto3_hand
3 months ago|Merge branch 'robotology:master' into honda_proto3_hand|7b625736a20aaaf1e5f4df7a9be3d1bc71fe56ba
## honda_proto3_hand
--------------------------------------------
/home/mebbaid/robot-code/robotology-superbuild/src/HumanDynamicsEstimation/: HEAD
11 months ago|Bump human-dynamics-estimation version to 2.6.0|e0bb2cdb906571532f5dbebf53dd5280e29aee4d
## HEAD (no branch)
--------------------------------------------
/home/mebbaid/robot-code/robotology-superbuild/src/human-gazebo/: HEAD
11 months ago|Bump version to v1.1.0|021f91a31bd30df3c2ade8b223e6135120162d4e
## HEAD (no branch)
--------------------------------------------
/home/mebbaid/robot-code/robotology-superbuild/src/ICUB/: HEAD
11 months ago|Prepare for the release|bba71571f5d6c1a05e76dfc4d643836715f163ae
## HEAD (no branch)
--------------------------------------------
/home/mebbaid/robot-code/robotology-superbuild/src/ICUBcontrib/: HEAD
3 years, 1 month ago|added official maintainer|cd6352688c64aadc0a044925532d08fbcb688886
## HEAD (no branch)
[... 102 lines cut ...]
1 year, 2 months ago|Merge pull request #18 from ami-iit/trackers|e51f925b0998e043ac9a20032923de20130acf93
## HEAD (no branch)
--------------------------------------------
/home/mebbaid/robot-code/robotology-superbuild/src/yarp-devices-forcetorque/: HEAD
1 year, 2 months ago|Fix compatibility with YARP 3.7 (#30)|13a07b25944528e6595b0b6dc8fccae75c18cb40
## HEAD (no branch)
--------------------------------------------
/home/mebbaid/robot-code/robotology-superbuild/src/yarp-matlab-bindings/: HEAD
1 year ago|Merge pull request #91 from robotology/onlylinkrequiredmatlablibraries|4e797d36ef0e1d64b6ac7beed02a749407cefb05
## HEAD (no branch)
--------------------------------------------
/home/mebbaid/robot-code/robotology-superbuild/src/yarp-openvr-trackers/: HEAD
1 year, 2 months ago|Merge pull request #9 from HosameldinMohamed/updateDoc|e6694adeb30d18f279fabd06fd867d76b44b3415
## HEAD (no branch)
--------------------------------------------
/home/mebbaid/robot-code/robotology-superbuild/src/YCM/: HEAD
1 year, 1 month ago|Merge pull request #414 from robotology/fixgraphviz3|2583228ed11f0fe5ba32497943b112099cd1af81
## HEAD (no branch)
--------------------------------------------

```

</details>

Any help would be appreciated @GiulioRomualdi @S-Dafarra @traversaro

S-Dafarra, 2023-07-20:
Have you tried with a clean build?

mebbaid, 2023-07-20:
> Have you tried with a clean build?

yeah I have tried with a clean build. 

In the end, to run the experiment I have to use my machine, but I would rather be able to test on the torso.

traversaro, 2023-07-20:
If you do not need it, I think you can disable FRAMEWORK_USES_PCL or a similar option, the problem may be related to the qhull conflict problem we have on the camera bridge PR.

mebbaid, 2023-07-20:
The Ubuntu version is 20 as discussed f2f. I will disable that option tomorrow since I just left the lab and check how things work. Thanks @traversaro

mebbaid, 2023-07-21:
@traversaro it seems that the option `FRAMEWORK_USES_PCL` is actually disabled by default in the bipedal

GiulioRomualdi, 2023-07-21:
> The Ubuntu version is 20 as discussed f2f. I will disable that option tomorrow since I just left the lab and check how things work. Thanks @traversaro

In ergocub-torso we have ubuntu 22.04 no 20 🤔

mebbaid, 2023-07-21:
> In ergocub-torso we have ubuntu 22.04 no 20 

Yeah, I was just answering Stefano who was wondering about my own machine

GiulioRomualdi, 2023-07-24:
One thing that changed is the superbuild. If I'm not mistaken in the previous version qhull was compiled by src while now it is installed by app (I had to rerun the apt installation script)

GiulioRomualdi, 2023-07-24:
@isorrentino cleaned the build and everything is compiling, she also noticed that the current date of the torso is old 

Indeed `date` returns
```
Mon Mar 20 15:49:28 UTC 2023
```
while it should be
```
lun 24 lug 2023, 11:37:41, CEST
```

traversaro, 2023-09-04:
Is that the torso of ergoCub ?

mebbaid, 2023-09-04:
> Is that the torso of ergoCub ?

yeah.

traversaro, 2023-09-04:
What is the status now? I would like to check the issue if the torso is on your branch (b.t.w., can you link the offending branch)?

mebbaid, 2023-09-04:
> What is the status now? I would like to check the issue if the torso is on your branch (b.t.w., can you link the offending branch)?

the offending branch is https://github.com/mebbaid/bipedal-locomotion-framework/tree/feature/CentroidalMPC

> What is the status now? 

I switched back everything to master and `qhull` to the one installed by apt.

traversaro, 2023-09-04:
I think to solve the issue you just need to cherry-pick (or copy manually, it is just a line) the fix in https://github.com/mebbaid/bipedal-locomotion-framework/commit/d1a905ebba5bae3620c5b593903c0a32f600d81b .

mebbaid, 2023-09-05:
> I think to solve the issue you just need to cherry-pick (or copy manually, it is just a line) the fix in [mebbaid@d1a905e](https://github.com/mebbaid/bipedal-locomotion-framework/commit/d1a905ebba5bae3620c5b593903c0a32f600d81b) , xref: #565 .

Thanks, we `make uninstall` the `qhull` compiled from source on the torso and it the fix works. I am closing this issue.