gbionics/bipedal-locomotion-framework #714: Disable homebrew CI job?

traversaro, 2023-07-31:
The homebrew CI jobs has been failing for several weeks with:
~~~
53/55 Test #52: QPInverseKinematicsUnitTests ..........................***Failed    1.97 sec
Randomness seeded to: 3280794946
[2023-07-31 15:58:29.174] [thread: 25492] [blf] [debug] [StdImplementation::getParameterPrivate] Parameter named 'mask' not found.
[2023-07-31 15:58:29.175] [thread: 25492] [blf] [info] [SE3Task::initialize]  [IK-SE3Task - Frame name:  link19] Unable to find the mask parameter. The default value is used: true true true.
[2023-07-31 15:58:29.175] [thread: 25492] [blf] [debug] [StdImplementation::getParameterPrivate] Parameter named 'use_orientation_exogenous_feedback' not found.
[2023-07-31 15:58:29.175] [thread: 25492] [blf] [info] [SE3Task::initialize]  [IK-SE3Task - Frame name:  link19] Unable to find the use_orientation_exogenous_feedback parameter. The default value is used: false.
[2023-07-31 15:58:29.175] [thread: 25492] [blf] [debug] [StdImplementation::getParameterPrivate] Parameter named 'use_position_exogenous_feedback' not found.
[2023-07-31 15:58:29.175] [thread: 25492] [blf] [info] [SE3Task::initialize]  [IK-SE3Task - Frame name:  link19] Unable to find the use_position_exogenous_feedback parameter. The default value is used: false.
[2023-07-31 15:58:29.175] [thread: 25492] [blf] [debug] [StdImplementation::getParameterPrivate] Parameter named 'mask' not found.
[2023-07-31 15:58:29.175] [thread: 25492] [blf] [info] [CoMTask::initialize] [CoMTask] Unable to find the mask parameter. The default value is used: true true true.
[2023-07-31 15:58:29.356] [thread: 25492] [blf] [debug] [StdImplementation::getParameterPrivate] Parameter named 'mask' not found.
[2023-07-31 15:58:29.356] [thread: 25492] [blf] [info] [SE3Task::initialize]  [IK-SE3Task - Frame name:  link34] Unable to find the mask parameter. The default value is used: true true true.
[2023-07-31 15:58:29.356] [thread: 25492] [blf] [debug] [StdImplementation::getParameterPrivate] Parameter named 'use_orientation_exogenous_feedback' not found.
[2023-07-31 15:58:29.356] [thread: 25492] [blf] [info] [SE3Task::initialize]  [IK-SE3Task - Frame name:  link34] Unable to find the use_orientation_exogenous_feedback parameter. The default value is used: false.
[2023-07-31 15:58:29.356] [thread: 25492] [blf] [debug] [StdImplementation::getParameterPrivate] Parameter named 'use_position_exogenous_feedback' not found.
[2023-07-31 15:58:29.356] [thread: 25492] [blf] [info] [SE3Task::initialize]  [IK-SE3Task - Frame name:  link34] Unable to find the use_position_exogenous_feedback parameter. The default value is used: false.
[2023-07-31 15:58:29.356] [thread: 25492] [blf] [debug] [StdImplementation::getParameterPrivate] Parameter named 'mask' not found.
[2023-07-31 15:58:29.356] [thread: 25492] [blf] [info] [CoMTask::initialize] [CoMTask] Unable to find the mask parameter. The default value is used: true true true.
[2023-07-31 15:58:29.658] [thread: 25492] [blf] [debug] [StdImplementation::getParameterPrivate] Parameter named 'mask' not found.
[2023-07-31 15:58:29.658] [thread: 25492] [blf] [info] [SE3Task::initialize]  [IK-SE3Task - Frame name:  link29] Unable to find the mask parameter. The default value is used: true true true.
[2023-07-31 15:58:29.658] [thread: 25492] [blf] [debug] [StdImplementation::getParameterPrivate] Parameter named 'use_orientation_exogenous_feedback' not found.
[2023-07-31 15:58:29.658] [thread: 25492] [blf] [info] [SE3Task::initialize]  [IK-SE3Task - Frame name:  link29] Unable to find the use_orientation_exogenous_feedback parameter. The default value is used: false.
[2023-07-31 15:58:29.658] [thread: 25492] [blf] [debug] [StdImplementation::getParameterPrivate] Parameter named 'use_position_exogenous_feedback' not found.
[2023-07-31 15:58:29.658] [thread: 25492] [blf] [info] [SE3Task::initialize]  [IK-SE3Task - Frame name:  link29] Unable to find the use_position_exogenous_feedback parameter. The default value is used: false.
[2023-07-31 15:58:29.658] [thread: 25492] [blf] [debug] [StdImplementation::getParameterPrivate] Parameter named 'mask' not found.
[2023-07-31 15:58:29.658] [thread: 25492] [blf] [info] [CoMTask::initialize] [CoMTask] Unable to find the mask parameter. The default value is used: true true true.
[2023-07-31 15:58:29.932] [thread: 25492] [blf] [debug] [QPInverseKinematics::advance] The solver found an inaccurate feasible solution.
[2023-07-31 15:58:29.964] [thread: 25492] [blf] [debug] [QPInverseKinematics::advance] The solver found an inaccurate feasible solution.
[2023-07-31 15:58:29.993] [thread: 25492] [blf] [debug] [QPInverseKinematics::advance] The solver found an inaccurate feasible solution.
Error: 7-31 15:58:30.027] [thread: 25492] [blf] [error] [QPInverseKinematics::advance] osqp was not able to find a feasible solution.

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
QPInverseKinematicsUnitTests is a Catch2 v3.0.1 host application.
Run with -? for options

-------------------------------------------------------------------------------
QP-IK [With strict limits]
-------------------------------------------------------------------------------
/Users/runner/work/bipedal-locomotion-framework/bipedal-locomotion-framework/src/IK/tests/QPInverseKinematicsTest.cpp:370
...............................................................................

/Users/runner/work/bipedal-locomotion-framework/bipedal-locomotion-framework/src/IK/tests/QPInverseKinematicsTest.cpp:445: FAILED:
  REQUIRE( ikAndTasks.ik->advance() )
with expansion:
  false

[2023-07-31 15:58:30.052] [thread: 25492] [blf] [debug] [StdImplementation::getParameterPrivate] Parameter named 'verbosity' not found.
[2023-07-31 15:58:30.052] [thread: 25492] [blf] [info] [QPInverseKinematics::initialize] 'verbosity' not found. The following parameter will be used 'false'.
[2023-07-31 15:58:30.052] [thread: 25492] [blf] [debug] [StdImplementation::getParameterPrivate] Parameter named 'mask' not found.
[2023-07-31 15:58:30.052] [thread: 25492] [blf] [info] [SE3Task::initialize]  [IK-SE3Task - Frame name:  link29] Unable to find the mask parameter. The default value is used: true true true.
[2023-07-31 15:58:30.052] [thread: 25492] [blf] [debug] [StdImplementation::getParameterPrivate] Parameter named 'use_orientation_exogenous_feedback' not found.
[2023-07-31 15:58:30.052] [thread: 25492] [blf] [info] [SE3Task::initialize]  [IK-SE3Task - Frame name:  link29] Unable to find the use_orientation_exogenous_feedback parameter. The default value is used: false.
[2023-07-31 15:58:30.052] [thread: 25492] [blf] [debug] [StdImplementation::getParameterPrivate] Parameter named 'use_position_exogenous_feedback' not found.
[2023-07-31 15:58:30.052] [thread: 25492] [blf] [info] [SE3Task::initialize]  [IK-SE3Task - Frame name:  link29] Unable to find the use_position_exogenous_feedback parameter. The default value is used: false.
[2023-07-31 15:58:30.053] [thread: 25492] [blf] [debug] [StdImplementation::getParameterPrivate] Parameter named 'mask' not found.
[2023-07-31 15:58:30.053] [thread: 25492] [blf] [info] [CoMTask::initialize] [CoMTask] Unable to find the mask parameter. The default value is used: true true true.
[2023-07-31 15:58:30.053] [thread: 25492] [blf] [debug] [StdImplementation::getParameterPrivate] Parameter named 'weight_provider_type' not found.
Warning: 31 15:58:30.053] [thread: 25492] [blf] [warning] [QPInverseKinematics::build] Unable to get the parameter 'weight_provider_type' for the task in the group 'REGULARIZATION_TASK'. The default one will be used. Default: 'ConstantWeightProvider'.
===============================================================================
test cases:   3 |   2 passed | 1 failed
assertions: 449 | 448 passed | 1 failed
~~~

Can we just disable it, as I guess no one in the lab is using homebrew?

GiulioRomualdi, 2023-07-31:
Yes I agree we need to clean the CI. While doing that I would also add the ci for ubuntu 22.04

GiulioRomualdi, 2024-04-05:
Removed in #822

closed 2024-04-05 by GiulioRomualdi