gbionics/bipedal-locomotion-framework #723: Support QPIK and TSID QP problems with no constraint

traversaro, 2023-09-14:
This is actually a osqp-eigen issue, but it is a good thing to have a placeholder for this here. fyi @Gio-DS @LoreMoretti @S-Dafarra

davidegorbani, 2023-12-14:
I experienced the same issue implementing a class that performs the IK for a human in [biomechanical-analysis-framework](https://github.com/ami-iit/biomechanical-analysis-framework); setting the priority of all the task to `highPriority` or `lowPriority` I get the following error:
```
[OsqpEigen::SparseMatrixHelper::eigenSparseMatrixToTriplets] The eigenSparseMatrix is empty.
[OsqpEigen::Solver::updateLinearConstraintsMatrix] Unable to evaluate triplets from the old hessian matrix.
[2023-12-14 12:24:27.020] [thread: 46797] [blf] [error] [QPInversekinematics::Impl::updateSolver] Unable to set the constraint matrix.
[2023-12-14 12:24:27.020] [thread: 46797] [blf] [error] [QPInverseKinematics::advance] Unable to update the QP solver.
```

GiulioRomualdi, 2023-12-15:
Thank you all @davidegorbani for adding the details. I'm currently working on fixing this issue. I hope I will open a PR within today 

Stay tune 🎶

GiulioRomualdi, 2023-12-15:
Hi @davidegorbani https://github.com/ami-iit/bipedal-locomotion-framework/pull/784 should fix your issue, you can give it a try: https://github.com/ami-iit/bipedal-locomotion-framework/pull/784

In the meantime, I'm writing the test in blf to avoid this happening again

davidegorbani, 2023-12-15:
I tested and now I do not get the error anymore; thank you @GiulioRomualdi!

GiulioRomualdi, 2023-12-15:
Great!

closed 2023-12-15 by GiulioRomualdi