gbionics/bipedal-locomotion-framework #735: Add minimization of joint torques to TSID

traversaro, 2023-10-05:
Similar to Equation 17 of https://ieeexplore.ieee.org/stamp/stamp.jsp?arnumber=7759126, even if in our case probably we can just add a $\lamba |\tau|^2$ term to the cost.

traversaro, 2023-10-05:
cc @ami-iit/isaac

traversaro, 2023-10-05:
SOT's TSID has the TaskActuationEquality for this: https://github.com/stack-of-tasks/tsid/blob/f938e35d880a4f8db8655c8c9af58e356631835e/src/tasks/task-actuation-equality.cpp#L26 .

akhilsathuluri, 2023-10-06:
One possibility is to use the `VariableRegularizationTask` as follows:

```
self.left_foot_regularization_task = blf.tsid.VariableRegularizationTask()
self.left_foot_regularization_task_name = "left_foot_regularization_task"
self.left_foot_regularization_task_priority = 1
self.left_foot_regularization_task_weight = 1e-1*np.ones(6)
self.left_foot_regularization_task_param_handler = blf.parameters_handler.StdParametersHandler()
self.left_foot_regularization_task_param_handler.set_parameter_string(name='variable_name', value= self.variable_name_left_contact)
self.left_foot_regularization_task_param_handler.set_parameter_int(name='variable_size', value= 6)
self.left_foot_regularization_task.initialize(param_handler = self.left_foot_regularization_task_param_handler)
```

where the `variable_name` should be replaced by the variable name that needs to regularised also within the optimisation problems variable handler, that could be, `joint_torques` in this case.

xela-95, 2023-10-06:
> One possibility is to use the `VariableRegularizationTask` as follows:
>
> ```
> self.left_foot_regularization_task = blf.tsid.VariableRegularizationTask()
> self.left_foot_regularization_task_name = [...]

Thank you @akhilsathuluri, I'll try this task! :)

xela-95, 2023-10-06:
CC @Gio-DS

traversaro, 2023-10-23:
Given the comments I guess we can close.