gbionics/bipedal-locomotion-framework #754: Unable to locate the python bindings after #752

isorrentino, 2023-11-07:
https://github.com/ami-iit/bipedal-locomotion-framework/pull/752 installs the python bindings in `/install/lib/python3.10/`. However, I got error by importing them with 
```
import bipedal_locomotion_framework.bindings as blf
```

with

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'bipedal_locomotion_framework'
```

isorrentino, 2023-11-07:
This is the output of pythonpath
```
echo $PYTHONPATH 
/home/isorrentino/dev/robotology-superbuild/build/install/lib/python3/dist-packages:/home/isorrentino/dev/element_sensorless-torque-control/code/python/
```
as you can see `build/install/lib/python3.10/` is not there

isorrentino, 2023-11-07:
cc @traversaro @GiulioRomualdi

traversaro, 2023-11-07:
https://github.com/ami-iit/bipedal-locomotion-framework/pull/752 changed the default installation location for Python bindings, see also the comment in https://github.com/ami-iit/bipedal-locomotion-framework/issues/751#issuecomment-1794518123 . However, if you use an updated version of the robotology-superbuild (after https://github.com/robotology/robotology-superbuild/pull/1508), everything should work fine.

isorrentino, 2023-11-08:
Thank you, I'm going to update the superbuild.

isorrentino, 2023-11-08:
> Updating the `robotology-superbuild` to the latest `master` I'm able to locate the python bindings. @GiulioRomualdi feel free to close this issue if you think no actions are required.

Actually, I see the bindings only if I run 

```shell
python3
import bipedal_locomotion_framework.bindings as blf
blf.robot_dynamics_estimator
```

from the `build` folder of `bipedal-locomotion-framework`. From all the other locations, if I run the same, I cannot locate the bindings again.

isorrentino, 2023-11-08:
Last update: doing `make` from the build of the `robotology-superbuild` now I can see the bindings from everywhere.

closed 2023-11-08 by GiulioRomualdi