gbionics/bipedal-locomotion-framework #789: ImportError: arg(): could not convert default argument into a Python object when using BLF

Zweisteine96, 2024-01-04:
Hi @traversaro @S-Dafarra @GiulioRomualdi ,

at very beginning when importing BLF python module, I had some other issues which are reported here: https://github.com/ami-iit/element_pi-learning-human-estimation/issues/36#issuecomment-1876906558, this morning I thought I finally fixed the problem, but then I found I can't use `TomlParametersHandler()` from `bipedal_locomotion_framework.bindings.parameters_handler`. So I did some uninstall-cleaning-install things, then I found myself stuck in the following bug if I try to import the BLF module:
```
Traceback (most recent call last):
  File "IKSolver.py", line 4, in <module>
    import bipedal_locomotion_framework.bindings.parameters_handler as blfh
  File "/home/cheng/robotology-superbuild/build/install/lib/python3/dist-packages/bipedal_locomotion_framework/__init__.py", line 1, in <module>
    from .bindings import *
ImportError: arg(): could not convert default argument into a Python object (type not registered yet?). Compile in debug mode for more information.
```

I'm using a virtual environment with python version 3.8.10. Before importing BLF python module, I did following things:
- In `robotology/src/manif` switch to `devel` branch and update to the newest, then in `robotology/build/src/manif` I did `cmake`+`make install`.
- In `robotology/src/blf` switch to `v0.17.0`, then in `robotology/build/src/blf` I did `cmake`+`make install` things.
- Then I did `pip install git+https://github.com/artivis/manif@devel`.
In the end I have `bipedal_locomotion_framework   0.17.0` and `manifpy  0.1.dev756`. I can import `manifpy` without any problem. I tried to find the reason of above mentioned issue with BLF importing, and only got this: https://github.com/patrikhuber/eos/issues/256 and this: https://github.com/neumond/pybrood/issues/4, which are most related to my case.

Do you have any idea what could be the reason and how could I tackle it? Many thanks :)

GiulioRomualdi, 2024-01-04:
Hi @Zweisteine96, if you installed BLF with the superbuild and enabled Python bindings, there's no need to install manif with pip since it's included in the superbuild. Keep in mind that the superbuild doesn't install toml, so if needed, manually install toml++. If you're not modifying BLF's source code, consider installing it via conda in a conda environment as outlined in the readme.

Zweisteine96, 2024-01-04:
> there's no need to install manif with pip since it's included in the superbuild

I think this `pip install git+https://github.com/artivis/manif@devel` is for installing the manifpy?

> Keep in mind that the superbuild doesn't install toml

Back in time I tried to enable the option of using toml++ when ccmake, but got the same results...

GiulioRomualdi, 2024-01-04:
> > there's no need to install manif with pip since it's included in the superbuild
> 
> I think this `pip install git+https://github.com/artivis/manif@devel` is for installing the manifpy?

Manif is compiled and installed by the superbuild if you installed from sources

> 
> > Keep in mind that the superbuild doesn't install toml
> 
> Back in time I tried to enable the option of using toml++ when ccmake, but got the same results...

Yes, indeed, you need it to install it manually from https://github.com/marzer/tomlplusplus if you decide to compile the superbuild from sources 

> 
> > consider installing it via conda in a conda environment
> 
> but the BLF is already in robotology-superbuild, can't I just it from there?

If you don't need to modify blf and all the other packages in the superbuild, you can avoid compiling it and just install the package in a conda env with conda commands.

Zweisteine96, 2024-01-04:
> Yes, indeed, you need it to install it manually from https://github.com/marzer/tomlplusplus if you decide to compile the superbuild from sources

so this means before I enable the using toml++ option in ccmake I need to manually install toml from the above link?

Zweisteine96, 2024-01-04:
Follow-up: I' ve cloned the toml++ repo and compiled it. Then when I tried to enable the "using toml++" option when compiling blf, even I've given the path where it can find the tomlPlusPlus.cmake file, it still reported not finding required cmake file.

traversaro, 2024-01-04:
> Follow-up: I' ve cloned the toml++ repo and compiled it. Then when I tried to enable the "using toml++" option when compiling blf, even I've given the path where it can find the tomlPlusPlus.cmake file, it still reported not finding required cmake file. 

Can you report the exact error you are experiencing? Where and with which command did you installed tomlplusplus?

Zweisteine96, 2024-01-05:
Hi @traversaro, sure! So I first cloned the toml++ repo to my `home` folder. Then I `mkdir build` and `ccmake ..`+`make` inside `build` folder. 

Then inside `robotology/build/src/blf` I did `ccmake .` and as shown below I put the path of this file `tomlplusplusConfig.cmake`:
```
tomlplusplus_DIR                 /home/cheng/tomlplusplus/build/install/lib/cmake/    
```
Then after configure the following error occured:
```
CMake Error at cmake/BipedalLocomotionFrameworkFindDependencies.cmake:58 (find_package):
   By not providing "Findtomlplusplus.cmake" in CMAKE_MODULE_PATH this project
   has asked CMake to find a package configuration file provided by
   "tomlplusplus", but CMake did not find one.

   Could not find a package configuration file provided by "tomlplusplus"
   (requested version 3.0.1) with any of the following names:

     tomlplusplusConfig.cmake
     tomlplusplus-config.cmake

   Add the installation prefix of "tomlplusplus" to CMAKE_PREFIX_PATH or set
   "tomlplusplus_DIR" to a directory containing one of the above files.  If
   "tomlplusplus" provides a separate development package or SDK, be sure it
   has been installed.
 Call Stack (most recent call first):
   cmake/BipedalLocomotionFrameworkDependencies.cmake:93 (checkandset_dependency)
   CMakeLists.txt:21 (include)
```

traversaro, 2024-01-05:
> Hi @traversaro, sure! So I first cloned the toml++ repo to my `home` folder. Then I `mkdir build` and `ccmake ..`+`make` inside `build` folder.
>
> Then inside `robotology/build/src/blf` I did `ccmake [...]

Are you sure that `/home/cheng/tomlplusplus/build/install/lib/cmake/` contains `tomlplusplusConfig.cmake` ? According to the build steps you just described, you did not pass `-DCMAKE_INSTALL_PREFIX=/home/cheng/tomlplusplus/build/install` during tomlplusplus CMake configuration, and then you did not executed `make install`,  so I am not sure how `tomlplusplusConfig.cmake` could end in `/home/cheng/tomlplusplus/build/install/lib/cmake/`. Furthermore, I would expect  `tomlplusplusConfig.cmake` to be in `/home/cheng/tomlplusplus/build/install/lib/cmake/tomlplusplus`, based on https://github.com/marzer/tomlplusplus/blob/d8fa9a1fddc90254cac2366dde23f0b613bc1280/cmake/install-rules.cmake#L29 .

traversaro, 2024-01-05:
> > Furthermore, I would expect tomlplusplusConfig.cmake to be in /home/cheng/tomlplusplus/build/install/lib/cmake/tomlplusplus,
> 
> You are right. I missed a folder `tomlplusplus`...

I think you also missed to report a few steps in https://github.com/ami-iit/bipedal-locomotion-framework/issues/789#issuecomment-1878505724 on how you configured and installed `tomlplusplus`, otherwise it is impossible that just by following the commands you reported there, you handed up with a tomlplusplus installation in `/home/cheng/tomlplusplus/build/install`.

Zweisteine96, 2024-01-05:
maybe these are the missing part?
```
BUILD_EXAMPLES                   OFF                                                                                                                                                                     
 BUILD_FUZZER                     OFF                                                                                                                                                                     
 CMAKE_BUILD_TYPE                                                                                                                                                                                         
 CMAKE_INSTALL_PREFIX             /home/cheng/tomlplusplus/build/instal
```

traversaro, 2024-01-05:
It is a bit difficult to debug the problem you report. Can you try to reproduce it in a clean venv+robotology-superbuild, and if you are successful report the steps here. In this way, it is easier to everyone to reproduce you error and debug it. Ideally, if you could capture that in a Dockerfile, it would be even easier. Furthermore, do you have any specific requirement to compile the superbuild from source with apt dependencies, or you are just interested in the running Python code? If you just require to run Python code, as @GiulioRomualdi already suggested perhaps preparing a clean conda environment and installing the blf package there could be the easiest way forward.

traversaro, 2024-01-05:
> maybe these are the missing part?
> 
> ```
> BUILD_EXAMPLES                   OFF                                                                                                                                                                     
>  BUILD_FUZZER                     OFF                                                                                                                                                                     
>  CMAKE_BUILD_TYPE                                                                                                                                                                                         
>  CMAKE_INSTALL_PREFIX             /home/cheng/tomlplusplus/build/install
> ```

To be honest, I can't know which commands did you use to install tomlplusplus on your environment. If you do not remember how you reached the state you are in, I strongly suggest you to start from a clean slate, otherwise it is quite difficult to be able to correctly debug any problem.

traversaro, 2024-01-05:
If you are not sure how to start from scratch, I can help you next week, just schedule a meeting on my calendar.

Zweisteine96, 2024-01-05:
> If you are not sure how to start from scratch, I can help you next week, just schedule a meeting on my calendar.

That's exactly what I was about to ask. I'd love to do that!

Zweisteine96, 2024-01-08:
In the end thanks to @traversaro, the problem is solved by creating a clean conda env instead of mix-using multiple package managers. I think we can close this issue now.

traversaro, 2024-01-08:
> In the end thanks to @traversaro, the problem is solved by creating a clean conda env instead of mix-using multiple package managers. I think we can close this issue now.

Feel free to close!

Zweisteine96, 2024-01-08:
Closing as the issue is tackled.

closed 2024-01-08 by Zweisteine96