gbionics/bipedal-locomotion-framework #799: Gravity task and distance task QPIK test fails on windows

GiulioRomualdi, 2024-01-19:
In #730 I noticed that the #790 I noticed that Gravity task and distance task QPIK test fails on windows  @EhsanRanjbari and @S-Dafarra.
As @s-dafarra mentioned this is weird since the tests started failing recentely with a huge error. This issue aims at investigating what is happening

_Originally posted by @GiulioRomualdi in https://github.com/ami-iit/bipedal-locomotion-framework/issues/790#issuecomment-1900629439_

GiulioRomualdi, 2024-01-19:
To simplify the debug:
- working configuration: https://github.com/ami-iit/bipedal-locomotion-framework/actions/runs/7523465014
- first not working job: https://github.com/ami-iit/bipedal-locomotion-framework/actions/runs/7536153930

GiulioRomualdi, 2024-01-19:
That's the diff between the two conda envs. The main difference I see is iDynTree @traversaro

```patch
5c5
< aom                       3.5.0                h63175ca_0    conda-forge
---
> aom                       3.6.1                h63175ca_0    conda-forge
7c7
< assimp                    5.2.5                h4dcb625_0    conda-forge
---
> assimp                    5.3.1                h4dcb625_1    conda-forge
13c13
< casadi                    3.6.2           py310ha06830a_0    conda-forge
---
> casadi                    3.6.3           py310hc65f482_3    conda-forge
21c21
< dav1d                     1.2.0                hcfcfb64_0    conda-forge
---
> dav1d                     1.2.1                hcfcfb64_0    conda-forge
27c27
< ffmpeg                    5.1.2           gpl_h5037a79_109    conda-forge
---
> ffmpeg                    5.1.2           gpl_h8bb4bc8_112    conda-forge
42a43
> glfw                      3.3.9                hcfcfb64_0    conda-forge
53c54
< idyntree                  8.1.0           py310h481cdde_1    conda-forge
---
> idyntree                  10.2.0          py310h042274b_0    conda-forge
58c59
< ipopt                     3.14.11              ha9547d1_0    conda-forge
---
> ipopt                     3.14.13              ha9547d1_0    conda-forge
69,70c70,71
< libclang                  15.0.7          default_h77d9078_3    conda-forge
< libclang13                15.0.7          default_h77d9078_3    conda-forge
---
> libclang                  15.0.7          default_hde6756a_4    conda-forge
> libclang13                15.0.7          default_h85b4d89_4    conda-forge
77c78
< libhwloc                  2.9.1                h51c2c0f_0    conda-forge
---
> libhwloc                  2.9.3           default_haede6df_1009    conda-forge
87c88
< libosqp                   0.6.2                h63175ca_4    conda-forge
---
> libosqp                   0.6.3                h63175ca_0    conda-forge
95c96
< libunicycle-footstep-planner 0.6.0                h63175ca_0    conda-forge
---
> libunicycle-footstep-planner 0.7.0                h63175ca_0    conda-forge
98c99
< libxml2                   2.10.4               hc3477c8_0    conda-forge
---
> libxml2                   2.12.4               hc3477c8_1    conda-forge
109c110
< metis                     5.1.0             h63175ca_1007    conda-forge
---
> metis                     5.1.1                h63175ca_2    conda-forge
121c122
< osqp-eigen                0.8.0                h63175ca_0    conda-forge
---
> osqp-eigen                0.8.1                h6d7489e_0    conda-forge
129c130
< proxsuite                 0.3.7           py310h5588dad_0    conda-forge
---
> proxsuite                 0.5.1           py310h5588dad_1    conda-forge
229c230
< tbb                       2021.9.0             h91493d7_0    conda-forge
---
> tbb                       2021.11.0            h91493d7_0    conda-forge
275c276
< USERDOMAIN_ROAMINGPROFILE=fv-az741-700
---
> USERDOMAIN_ROAMINGPROFILE=fv-az1112-453
278c279
< HOSTNAME=fv-az741-700
---
> HOSTNAME=fv-az1112-453
280c281
< GITHUB_PATH=D:\a\_temp\_runner_file_commands\add_path_13b8f5dc-a1bb-4553-ae5d-846d4ce8c280
---
> GITHUB_PATH=D:\a\_temp\_runner_file_commands\add_path_cdab868f-d3f1-45e2-95a3-55d06301913d
288,289c289,290
< GITHUB_RUN_NUMBER=4071
< RUNNER_NAME=GitHub Actions 6
---
> GITHUB_RUN_NUMBER=4072
> RUNNER_NAME=GitHub Actions 5
306c307
< USERDOMAIN=fv-az741-700
---
> USERDOMAIN=fv-az1112-453
343c344
< RUNNER_TRACKING_ID=github_687aa1cb-3cde-4d1e-aba6-9032b5f8e5bf
---
> RUNNER_TRACKING_ID=github_670ff035-73ae-4224-890a-b2457d691e6b
350c351
< GITHUB_STATE=D:\a\_temp\_runner_file_commands\save_state_13b8f5dc-a1bb-4553-ae5d-846d4ce8c280
---
> GITHUB_STATE=D:\a\_temp\_runner_file_commands\save_state_cdab868f-d3f1-45e2-95a3-55d06301913d
355c356
< GITHUB_ENV=D:\a\_temp\_runner_file_commands\set_env_13b8f5dc-a1bb-4553-ae5d-846d4ce8c280
---
> GITHUB_ENV=D:\a\_temp\_runner_file_commands\set_env_cdab868f-d3f1-45e2-95a3-55d06301913d
361c362
< GITHUB_RUN_ID=7523465014
---
> GITHUB_RUN_ID=7536153930
365c366
< GITHUB_STEP_SUMMARY=D:\a\_temp\_runner_file_commands\step_summary_13b8f5dc-a1bb-4553-ae5d-846d4ce8c280
---
> GITHUB_STEP_SUMMARY=D:\a\_temp\_runner_file_commands\step_summary_cdab868f-d3f1-45e2-95a3-55d06301913d
367c368
< COMPUTERNAME=fv-az741-700
---
> COMPUTERNAME=fv-az1112-453
376c377
< LOGONSERVER=\\fv-az741-700
---
> LOGONSERVER=\\fv-az1112-453
449c450
< GITHUB_OUTPUT=D:\a\_temp\_runner_file_commands\set_output_13b8f5dc-a1bb-4553-ae5d-846d4ce8c280
---
> GITHUB_OUTPUT=D:\a\_temp\_runner_file_commands\set_output_cdab868f-d3f1-45e2-95a3-55d06301913d
```

traversaro, 2024-01-19:
> That's the diff between the two conda envs. The main difference I see is iDynTree @traversaro

I think the problem is metis 5.1.1 . It created problems in other packages (see https://github.com/conda-forge/mumps-feedstock/issues/106) and this is why in conda-forge we reverted back to pin metis to 5.1.0 right this afternoon: https://github.com/conda-forge/conda-forge-pinning-feedstock/pull/5396 . I tought we were safe from that as we did not migrated ipopt to metis 5.1.1 (see https://github.com/conda-forge/ipopt-feedstock/pull/97), but I guess that ipopt Windows packages do not depend directly on metis, and so once a new mumps compiled against metis 5.1.1 version was released, we ended up with ipopt installed with metis 5.1.1 . 

The good news is that we are now safe, we just need to compile ipopt with the latest mumps (happening soon, the migration to 5.6.2 was merged in https://github.com/conda-forge/conda-forge-pinning-feedstock/pull/5274 and the bot will start to open PRs soon) and the CI will be fixed.

traversaro, 2024-01-19:
The nice thing is that I did metis 5.1.1 related comments and fixes the last few days, but I was not aware that it already was affecting blf! :D

traversaro, 2024-01-19:
See https://conda-forge.org/status/#mumps_mpi562 for the migration that should as a side effect fix this.

S-Dafarra, 2024-01-19:
I was thinking more a problem of osqp since it is the solver we use for the IK. I don't know if osqp is affected by metis

S-Dafarra, 2024-01-19:
Actually, also osqp changed from 0.6.2 to 0.6.3

GiulioRomualdi, 2024-01-19:
Yes, exactly, the IK uses osqp 🤔

traversaro, 2024-01-19:
Ah, you are right, I was confusing this with iDynTree's IK.

traversaro, 2024-01-19:
I reproduced the failure locally with a minimal env create as:
~~~
mamba create -n blfmindev cmake pkg-config ninja make spdlog eigen idyntree liblie-group-controllers nlohmann_json osqp-eigen vs2022_win-64
~~~

detail env:
~~~
assimp                    5.3.1                h81f0834_2    conda-forge
bzip2                     1.0.8                hcfcfb64_5    conda-forge
c-compiler                1.7.0                hcfcfb64_0    conda-forge
ca-certificates           2023.11.17           h56e8100_0    conda-forge
catch2                    3.5.2                h181d51b_0    conda-forge
clangdev                  5.0.0                   flang_3    conda-forge
cmake                     3.28.1               hf0feee3_0    conda-forge
compilers                 1.7.0                h57928b3_0    conda-forge
cxx-compiler              1.7.0                h91493d7_0    conda-forge
eigen                     3.4.0                h91493d7_0    conda-forge
flang                     5.0.0           he025d50_20180525    conda-forge
flang_win-64              5.0.0           h13ae965_20180526    conda-forge
fmt                       10.2.1               h181d51b_0    conda-forge
fortran-compiler          1.7.0                h9655429_0    conda-forge
gettext                   0.21.1               h5728263_0    conda-forge
glfw                      3.3.9                hcfcfb64_0    conda-forge
gtest                     1.14.0               h91493d7_1    conda-forge
idyntree                  10.2.0          py312h5ac85b2_1    conda-forge
intel-openmp              2023.2.0         h57928b3_50497    conda-forge
ipopt                     3.14.13              ha9547d1_0    conda-forge
irrlicht                  1.8.5                h65f4d7e_4    conda-forge
krb5                      1.21.2               heb0366b_0    conda-forge
libblas                   3.9.0              20_win64_mkl    conda-forge
libboost                  1.82.0               h65993cd_6    conda-forge
libcblas                  3.9.0              20_win64_mkl    conda-forge
libcurl                   8.5.0                hd5e4a3a_0    conda-forge
libexpat                  2.5.0                h63175ca_1    conda-forge
libffi                    3.4.2                h8ffe710_5    conda-forge
libflang                  5.0.0           h6538335_20180525    conda-forge
libglib                   2.78.3               h16e383f_0    conda-forge
libhwloc                  2.9.3           default_haede6df_1009    conda-forge
libiconv                  1.17                 hcfcfb64_2    conda-forge
libjpeg-turbo             3.0.0                hcfcfb64_1    conda-forge
liblapack                 3.9.0              20_win64_mkl    conda-forge
liblie-group-controllers  0.2.0                h63175ca_0    conda-forge
libosqp                   0.6.3                h63175ca_0    conda-forge
libpng                    1.6.39               h19919ed_0    conda-forge
libqdldl                  0.1.5                h63175ca_1    conda-forge
libsqlite                 3.44.2               hcfcfb64_0    conda-forge
libssh2                   1.11.0               h7dfc565_0    conda-forge
libuv                     1.44.2               hcfcfb64_1    conda-forge
libxml2                   2.12.4               hc3477c8_1    conda-forge
libzlib                   1.2.13               hcfcfb64_5    conda-forge
llvm-meta                 5.0.0                         0    conda-forge
m2w64-gcc-libgfortran     5.3.0                         6    conda-forge
m2w64-gcc-libs            5.3.0                         7    conda-forge
m2w64-gcc-libs-core       5.3.0                         7    conda-forge
m2w64-gmp                 6.1.0                         2    conda-forge
m2w64-libwinpthread-git   5.0.0.4634.697f757               2    conda-forge
make                      4.3                  h3d2af85_1    conda-forge
manif                     0.0.4               h63175ca_16    conda-forge
metis                     5.1.1                h63175ca_2    conda-forge
mkl                       2023.2.0         h6a75c08_50497    conda-forge
msys2-conda-epoch         20160418                      1    conda-forge
mumps-seq                 5.2.1               h1f49738_14    conda-forge
ninja                     1.11.1               h91493d7_0    conda-forge
nlohmann_json             3.11.2               h39d44d4_0    conda-forge
numpy                     1.26.3          py312h8753938_0    conda-forge
openmp                    5.0.0                    vc14_1    conda-forge
openssl                   3.2.0                hcfcfb64_1    conda-forge
osqp-eigen                0.8.1                h6d7489e_0    conda-forge
pcre2                     10.42                h17e33f8_0    conda-forge
pip                       23.3.2             pyhd8ed1ab_0    conda-forge
pkg-config                0.29.2            h2bf4dc2_1008    conda-forge
pthreads-win32            2.9.1                hfa6e2cd_3    conda-forge
python                    3.12.1          h2628c8c_1_cpython    conda-forge
python_abi                3.12                    4_cp312    conda-forge
sdl                       1.2.68               h21dd15a_0    conda-forge
sdl2                      2.28.5               h63175ca_0    conda-forge
setuptools                69.0.3             pyhd8ed1ab_0    conda-forge
spdlog                    1.12.0               h64d2f7d_2    conda-forge
[... cut ...]
vs2019_win-64             19.29.30139         he1865b1_18    conda-forge
vs2022_win-64             19.38.33130         h0bfb142_18    conda-forge
vswhere                   3.1.4                h57928b3_0    conda-forge
wheel                     0.42.0             pyhd8ed1ab_0    conda-forge
xz                        5.2.6                h8d14728_0    conda-forge
zlib                      1.2.13               hcfcfb64_5    conda-forge
zstd                      1.5.5                h12be248_0    conda-forge
~~~

traversaro, 2024-01-19:
Also forcing libosqp to 0.6.2 result in the same error being generated:
~~~
# packages in environment at C:\Users\straversaro\AppData\Local\miniforge3\envs\blfmin2:
#
# Name                    Version                   Build  Channel
assimp                    5.2.5                h4dcb625_0    conda-forge
boost-cpp                 1.78.0               h9f4b32c_4    conda-forge
bzip2                     1.0.8                hcfcfb64_5    conda-forge
ca-certificates           2023.11.17           h56e8100_0    conda-forge
cmake                     3.28.1               hf0feee3_0    conda-forge
eigen                     3.4.0                h91493d7_0    conda-forge
fmt                       10.2.1               h181d51b_0    conda-forge
gettext                   0.21.1               h5728263_0    conda-forge
glfw                      3.3.9                hcfcfb64_0    conda-forge
gtest                     1.14.0               h91493d7_1    conda-forge
idyntree                  9.1.0           py311h42847bd_0    conda-forge
intel-openmp              2023.2.0         h57928b3_50497    conda-forge
ipopt                     3.14.12              ha9547d1_1    conda-forge
irrlicht                  1.8.5                h65f4d7e_4    conda-forge
krb5                      1.21.2               heb0366b_0    conda-forge
libblas                   3.9.0              20_win64_mkl    conda-forge
libcblas                  3.9.0              20_win64_mkl    conda-forge
libcurl                   8.5.0                hd5e4a3a_0    conda-forge
libexpat                  2.5.0                h63175ca_1    conda-forge
libffi                    3.4.2                h8ffe710_5    conda-forge
libflang                  5.0.0           h6538335_20180525    conda-forge
libglib                   2.78.3               h16e383f_0    conda-forge
libhwloc                  2.9.3           default_haede6df_1009    conda-forge
libiconv                  1.17                 hcfcfb64_2    conda-forge
libjpeg-turbo             3.0.0                hcfcfb64_1    conda-forge
liblapack                 3.9.0              20_win64_mkl    conda-forge
liblie-group-controllers  0.2.0                h63175ca_0    conda-forge
libosqp                   0.6.2                h63175ca_4    conda-forge
libpng                    1.6.39               h19919ed_0    conda-forge
libqdldl                  0.1.5                h63175ca_1    conda-forge
libsqlite                 3.44.2               hcfcfb64_0    conda-forge
libssh2                   1.11.0               h7dfc565_0    conda-forge
libuv                     1.44.2               hcfcfb64_1    conda-forge
libxml2                   2.12.4               hc3477c8_1    conda-forge
libzlib                   1.2.13               hcfcfb64_5    conda-forge
llvm-meta                 5.0.0                         0    conda-forge
m2w64-gcc-libgfortran     5.3.0                         6    conda-forge
m2w64-gcc-libs            5.3.0                         7    conda-forge
m2w64-gcc-libs-core       5.3.0                         7    conda-forge
m2w64-gmp                 6.1.0                         2    conda-forge
m2w64-libwinpthread-git   5.0.0.4634.697f757               2    conda-forge
make                      4.3                  h3d2af85_1    conda-forge
manif                     0.0.4               h63175ca_16    conda-forge
metis                     5.1.0             h63175ca_1007    conda-forge
mkl                       2023.2.0         h6a75c08_50497    conda-forge
msys2-conda-epoch         20160418                      1    conda-forge
mumps-seq                 5.2.1               h1f49738_14    conda-forge
ninja                     1.11.1               h91493d7_0    conda-forge
nlohmann_json             3.11.2               h39d44d4_0    conda-forge
numpy                     1.26.3          py311h0b4df5a_0    conda-forge
openmp                    5.0.0                    vc14_1    conda-forge
openssl                   3.2.0                hcfcfb64_1    conda-forge
osqp-eigen                0.8.0                h63175ca_0    conda-forge
pcre2                     10.42                h17e33f8_0    conda-forge
pip                       23.3.2             pyhd8ed1ab_0    conda-forge
pkg-config                0.29.2            h2bf4dc2_1008    conda-forge
pthreads-win32            2.9.1                hfa6e2cd_3    conda-forge
python                    3.11.7          h2628c8c_1_cpython    conda-forge
python_abi                3.11                    4_cp311    conda-forge
sdl                       1.2.68               h21dd15a_0    conda-forge
sdl2                      2.28.5               h63175ca_0    conda-forge
setuptools                69.0.3             pyhd8ed1ab_0    conda-forge
spdlog                    1.12.0               h64d2f7d_2    conda-forge
tbb                       2021.11.0            h91493d7_0    conda-forge
tk                        8.6.13               h5226925_1    conda-forge
tl-optional               1.1.0                h91493d7_1    conda-forge
tzdata                    2023d                h0c530f3_0    conda-forge
ucrt                      10.0.22621.0         h57928b3_0    conda-forge
vc                        14.3                hcf57466_18    conda-forge
vc14_runtime              14.38.33130         h82b7239_18    conda-forge
vs2015_runtime            14.38.33130         hcb4865c_18    conda-forge
vs2022_win-64             19.38.33130         h0bfb142_18    conda-forge
vswhere                   3.1.4                h57928b3_0    conda-forge
wheel                     0.42.0             pyhd8ed1ab_0    conda-forge
xz                        5.2.6                h8d14728_0    conda-forge
zlib                      1.2.13               hcfcfb64_5    conda-forge
zstd                      1.5.5                h12be248_0    conda-forge
~~~

traversaro, 2024-01-19:
An env created with:
~~~
mamba create -n blfmindev4 cmake pkg-config ninja make spdlog eigen idyntree liblie-group-controllers nlohmann_json osqp-eigen vs2022_win-64 tbb==2021.9.0 proxsuite==0.3.7 libxml2==2.12.4 aom==3.5.0 assimp==5.2.5 casadi==3.6.2 dav1d==1.2.0 ffmpeg=5.1.2 idyntree==8.1.0 ipopt==3.14.11 libhwloc==2.9.1 libosqp==0.6.2 libunicycle-footstep-planner==0.6.0 metis==5.1.0 osqp-eigen==0.8.0 proxsuite==0.3.7
~~~

works fine instead.

traversaro, 2024-01-20:
Ok, after a few iterations I found two environments quite similar (with same libosqp and osqp-eigen version, but one that fails, and one not.

One that fails:
~~~
# Not working
aom                       3.5.0                h63175ca_0    conda-forge
assimp                    5.2.5                h4dcb625_0    conda-forge
boost-cpp                 1.78.0               h9f4b32c_4    conda-forge
bzip2                     1.0.8                hcfcfb64_5    conda-forge
ca-certificates           2023.11.17           h56e8100_0    conda-forge
casadi                    3.6.2           py311h4ef4915_1    conda-forge
cmake                     3.28.1               hf0feee3_0    conda-forge
dav1d                     1.2.1                hcfcfb64_0    conda-forge
eigen                     3.4.0                h91493d7_0    conda-forge
expat                     2.5.0                h63175ca_1    conda-forge
ffmpeg                    6.0.0           gpl_h1f67d4f_104    conda-forge
fmt                       10.2.1               h181d51b_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 h77eed37_1    conda-forge
fontconfig                2.14.2               hbde0cde_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
freetype                  2.12.1               hdaf720e_2    conda-forge
gettext                   0.21.1               h5728263_0    conda-forge
glfw                      3.3.9                hcfcfb64_0    conda-forge
gtest                     1.14.0               h91493d7_1    conda-forge
idyntree                  9.1.0           py311h42847bd_0    conda-forge
intel-openmp              2023.2.0         h57928b3_50497    conda-forge
ipopt                     3.14.12              ha9547d1_1    conda-forge
irrlicht                  1.8.5                h65f4d7e_4    conda-forge
krb5                      1.21.2               heb0366b_0    conda-forge
libblas                   3.9.0              20_win64_mkl    conda-forge
libcblas                  3.9.0              20_win64_mkl    conda-forge
libcurl                   8.5.0                hd5e4a3a_0    conda-forge
libexpat                  2.5.0                h63175ca_1    conda-forge
libffi                    3.4.2                h8ffe710_5    conda-forge
libflang                  5.0.0           h6538335_20180525    conda-forge
libglib                   2.78.3               h16e383f_0    conda-forge
libhwloc                  2.9.1           nocuda_h15da153_6    conda-forge
libiconv                  1.17                 hcfcfb64_2    conda-forge
libjpeg-turbo             3.0.0                hcfcfb64_1    conda-forge
liblapack                 3.9.0              20_win64_mkl    conda-forge
liblie-group-controllers  0.2.0                h63175ca_0    conda-forge
libopus                   1.3.1                h8ffe710_1    conda-forge
libosqp                   0.6.2                h63175ca_4    conda-forge
libpng                    1.6.39               h19919ed_0    conda-forge
libqdldl                  0.1.5                h63175ca_1    conda-forge
libsqlite                 3.44.2               hcfcfb64_0    conda-forge
libssh2                   1.11.0               h7dfc565_0    conda-forge
libunicycle-footstep-planner 0.6.0                h63175ca_1    conda-forge
libuv                     1.44.2               hcfcfb64_1    conda-forge
libxml2                   2.12.4               hc3477c8_1    conda-forge
libzlib                   1.2.13               hcfcfb64_5    conda-forge
llvm-meta                 5.0.0                         0    conda-forge
m2w64-gcc-libgfortran     5.3.0                         6    conda-forge
m2w64-gcc-libs            5.3.0                         7    conda-forge
m2w64-gcc-libs-core       5.3.0                         7    conda-forge
m2w64-gmp                 6.1.0                         2    conda-forge
m2w64-libwinpthread-git   5.0.0.4634.697f757               2    conda-forge
make                      4.3                  h3d2af85_1    conda-forge
manif                     0.0.4               h63175ca_16    conda-forge
metis                     5.1.0             h63175ca_1007    conda-forge
mkl                       2023.2.0         h6a75c08_50497    conda-forge
msys2-conda-epoch         20160418                      1    conda-forge
mumps-seq                 5.2.1               h1f49738_14    conda-forge
ninja                     1.11.1               h91493d7_0    conda-forge
nlohmann_json             3.11.2               h39d44d4_0    conda-forge
numpy                     1.26.3          py311h0b4df5a_0    conda-forge
[... 158 lines cut ...]
@@ -20,11 +20,10 @@
 fonts-conda-forge         1                             0    conda-forge
 freetype                  2.12.1               hdaf720e_2    conda-forge
 gettext                   0.21.1               h5728263_0    conda-forge
-glfw                      3.3.9                hcfcfb64_0    conda-forge
 gtest                     1.14.0               h91493d7_1    conda-forge
-idyntree                  9.1.0           py311h42847bd_0    conda-forge
+idyntree                  8.1.0           py311he5a1352_1    conda-forge
[... cut ...]
 libsqlite                 3.44.2               hcfcfb64_0    conda-forge
 libssh2                   1.11.0               h7dfc565_0    conda-forge
-libunicycle-footstep-planner 0.6.0                h63175ca_1    conda-forge
+libunicycle-footstep-planner 0.6.0                h63175ca_0    conda-forge
 libuv                     1.44.2               hcfcfb64_1    conda-forge
 libxml2                   2.12.4               hc3477c8_1    conda-forge
 libzlib                   1.2.13               hcfcfb64_5    conda-forge

~~~

GiulioRomualdi, 2024-01-20:
So, at this point, I was wondering if there is something in idyntree. Notice that the IK uses idyntree to compute jacobians and transforms

traversaro, 2024-01-20:
I also suspect some subtle ABI incompatibility doing something here, even if the perfectly repeatable error seems to indicate otherwise. There is a bunch of rebuilds being done due to new ipopt, new mumps, old metis, let's see what happens after all of that.

traversaro, 2024-01-21:
Ok, I created an env with the latest mumps and ipopt and idyntree, and it still fails:
~~~
mamba create -n blfmindev cmake pkg-config ninja make spdlog eigen idyntree liblie-group-controllermamba create -n blfmindev cmake pkg-config ninja make spdlog eigen idyntree liblie-group-controllers nlohmann_json osqp-eigen vs2022_wiQPInverseKinematicsUnitTestsn-64
mamba activate blfmindev 
git clone https://github.com/ami-iit/bipedal-locomotion-framework
cd bipedal-locomotion-framework
cmake -GNinja -DCMAKE_INSTALL_PREFIX=$CONDA_PREFIX -DBUILD_TESTING:BOOL=ON ..
ninja
ctest -VV -R QPInverseKinematicsUnitTests
mamba list
~~~

output:

~~~
(blfmindev) C:\src\bipedal-locomotion-framework\buildnew>ctest -VV -R QPInverseKinematicsUnitTests
UpdateCTestConfiguration  from :C:/src/bipedal-locomotion-framework/buildnew/DartConfiguration.tcl
Parse Config file:C:/src/bipedal-locomotion-framework/buildnew/DartConfiguration.tcl
UpdateCTestConfiguration  from :C:/src/bipedal-locomotion-framework/buildnew/DartConfiguration.tcl
Parse Config file:C:/src/bipedal-locomotion-framework/buildnew/DartConfiguration.tcl
Test project C:/src/bipedal-locomotion-framework/buildnew
Constructing a list of tests
Done constructing a list of tests
Updating test list for fixtures
Added 0 tests to meet fixture requirements
Checking test dependency graph...
Checking test dependency graph end
test 42
    Start 42: QPInverseKinematicsUnitTests

42: Test command: C:\src\bipedal-locomotion-framework\buildnew\bin\QPInverseKinematicsUnitTests.exe
42: Working Directory: C:/src/bipedal-locomotion-framework/buildnew/src/IK/tests
42: Test timeout computed to be: 1500
42: [2024-01-21 15:04:19.203] [thread: 1548] [blf] [info] [SE3Task::initialize]  [IK-SE3Task - Frame name:  link19] Unable to find the mask parameter. The default value is used: true true true.
42: [2024-01-21 15:04:19.203] [thread: 1548] [blf] [info] [SE3Task::initialize]  [IK-SE3Task - Frame name:  link19] Unable to find the use_orientation_exogenous_feedback parameter. The default value is used: false.
42: [2024-01-21 15:04:19.203] [thread: 1548] [blf] [info] [SE3Task::initialize]  [IK-SE3Task - Frame name:  link19] Unable to find the use_position_exogenous_feedback parameter. The default value is used: false.
42: [2024-01-21 15:04:19.203] [thread: 1548] [blf] [info] [CoMTask::initialize] [CoMTask] Unable to find the mask parameter. The default value is used: true true true.
42: [2024-01-21 15:04:19.205] [thread: 1548] [blf] [info] [SE3Task::initialize]  [IK-SE3Task - Frame name:  link34] Unable to find the mask parameter. The default value is used: true true true.
42: [2024-01-21 15:04:19.205] [thread: 1548] [blf] [info] [SE3Task::initialize]  [IK-SE3Task - Frame name:  link34] Unable to find the use_orientation_exogenous_feedback parameter. The default value is used: false.
42: [2024-01-21 15:04:19.205] [thread: 1548] [blf] [info] [SE3Task::initialize]  [IK-SE3Task - Frame name:  link34] Unable to find the use_position_exogenous_feedback parameter. The default value is used: false.
42: [2024-01-21 15:04:19.205] [thread: 1548] [blf] [info] [CoMTask::initialize] [CoMTask] Unable to find the mask parameter. The default value is used: true true true.
42: [2024-01-21 15:04:19.207] [thread: 1548] [blf] [info] [SE3Task::initialize]  [IK-SE3Task - Frame name:  link29] Unable to find the mask parameter. The default value is used: true true true.
42: [2024-01-21 15:04:19.207] [thread: 1548] [blf] [info] [SE3Task::initialize]  [IK-SE3Task - Frame name:  link29] Unable to find the use_orientation_exogenous_feedback parameter. The default value is used: false.
42: [2024-01-21 15:04:19.207] [thread: 1548] [blf] [info] [SE3Task::initialize]  [IK-SE3Task - Frame name:  link29] Unable to find the use_position_exogenous_feedback parameter. The default value is used: false.
42: [2024-01-21 15:04:19.207] [thread: 1548] [blf] [info] [CoMTask::initialize] [CoMTask] Unable to find the mask parameter. The default value is used: true true true.
42: [2024-01-21 15:04:19.209] [thread: 1548] [blf] [info] [QPInverseKinematics::initialize] 'verbosity' not found. The following parameter will be used 'false'.
42: [2024-01-21 15:04:19.209] [thread: 1548] [blf] [info] [SE3Task::initialize]  [IK-SE3Task - Frame name:  link29] Unable to find the mask parameter. The default value is used: true true true.
42: [2024-01-21 15:04:19.209] [thread: 1548] [blf] [info] [SE3Task::initialize]  [IK-SE3Task - Frame name:  link29] Unable to find the use_orientation_exogenous_feedback parameter. The default value is used: false.
42: [2024-01-21 15:04:19.209] [thread: 1548] [blf] [info] [SE3Task::initialize]  [IK-SE3Task - Frame name:  link29] Unable to find the use_position_exogenous_feedback parameter. The default value is used: false.
42: [2024-01-21 15:04:19.209] [thread: 1548] [blf] [info] [CoMTask::initialize] [CoMTask] Unable to find the mask parameter. The default value is used: true true true.
42: [2024-01-21 15:04:19.209] [thread: 1548] [blf] [warning] [QPInverseKinematics::build] Unable to get the parameter 'weight_provider_type' for the task in the group 'REGULARIZATION_TASK'. The default one will be used. Default: 'ConstantWeightProvider'.
42: [2024-01-21 15:04:19.209] [thread: 1548] [blf] [warning] [QPInverseKinematics::build] Unable to get the parameter 'weight_provider_type' for the task in the group 'DISTANCE_TASK'. The default one will be used. Default: 'ConstantWeightProvider'.
[... cut ...]
vs2015_runtime            14.38.33130         hcb4865c_18    conda-forge
vs2022_win-64             19.38.33130         h0bfb142_18    conda-forge
vswhere                   3.1.4                h57928b3_0    conda-forge
wheel                     0.42.0             pyhd8ed1ab_0    conda-forge
xz                        5.2.6                h8d14728_0    conda-forge
zlib                      1.2.13               hcfcfb64_5    conda-forge
zstd                      1.5.5                h12be248_0    conda-forge
~~~

traversaro, 2024-01-21:
I wonder if this failure is related to the change in `iDynTree::getRandomModel`, that started to export also prismatic joints, see https://github.com/robotology/idyntree/pull/1057 and https://github.com/robotology/idyntree/commit/9c7f2b36a2505f132700744bfffb142728639a57

traversaro, 2024-01-21:
> I wonder if this failure is related to the change in `iDynTree::getRandomModel`, that started to export also prismatic joints, see [robotology/idyntree#1057](https://github.com/robotology/idyntree/pull/1057) and [robotology/idyntree@9c7f2b3](https://github.com/robotology/idyntree/commit/9c7f2b36a2505f132700744bfffb142728639a57)

Ok, indeed running that test just on a model with revolute joints solve the problem. I guess the problem is indeed in some kind of bug with Jacobian on models with prismatic joints, that have been in iDynTree since prismatic joints were added, but was hidden by the fact that prismatic joints were not actually tested in getRandomModel and in KinDynComputations tests. As the problem is actually in iDynTree, I think it make sense just to test revolute joints in BLF. I guess there is nothing specifically Windows-related in the error, we were just kind of "lucky" or get this on Windows. I guess that if we run multiple times `getRandomModel` on Linux, we probably would eventually reach this problem.

traversaro, 2024-01-21:
(unrelated, diff for debug, posting it not to lose it):

~~~diff
@@ -967,6 +1003,12 @@ TEST_CASE("QP-IK [Distance and Gravity tasks Unconstrained]")
         baseVelocity = ik->getOutput().baseVelocity.coeffs();
         jointVelocity = ik->getOutput().jointVelocity;

+        std::cerr << "--- iteration: "  << iteration << std::endl;
+        std::cerr << "--- baseVelocity:\n" << baseVelocity << std::endl;
+        std::cerr << "--- jointVelocity:\n" << jointVelocity << std::endl;
+        std::cerr << "--- world_H_target current translation norm: " << toManifPose(kinDyn->getWorldTransform(desiredSetPoints.targetFrameDistance)).translation().norm() << std::endl;
+        std::cerr << "--- world_H_target desired translation norm: " << desiredSetPoints.targetDistance << std::endl;
+
         // propagate the dynamical system
         system.dynamics->setControlInput({baseVelocity, jointVelocity});
         system.integrator->integrate(0s, dT);
~~~

traversaro, 2024-01-21:
https://github.com/ami-iit/bipedal-locomotion-framework/pull/800 should provide a workaround for the problem in blf, the actual issue will be tracked in https://github.com/robotology/idyntree/issues/1149 .

GiulioRomualdi, 2024-01-22:
Fixed by #800!

closed 2024-01-22 by GiulioRomualdi