gbionics/bipedal-locomotion-framework #807: YarpSensorBridge fails in attaching FT IMU in simulation

GiulioRomualdi, 2024-02-13:
I'm trying to test https://github.com/icub-tech-iit/ergocub-software/pull/217 with `YarpRobotLoggerDevice`. To do so I modified the configuration file of the logger as 90894633d38ff366636528fe4b7b944fffc6f3c5

However when I run the device I  got
```
[ERROR] [2024-02-13 09:39:06.481] [thread: 438810] [blf] [YarpSensorBridge::Impl::checkAttachedMASSensors] Expecting the same number of MAS sensors attached to the Bridge as many mentioned in the initialization step. Number of MAS sensor in the interface: 0. Number of sensor in initialization: 4.
[ERROR] [2024-02-13 09:39:06.481] [thread: 438810] [blf] [YarpSensorBridge::Impl::attachAndCheckMASSensors] Could not find at least one of the required sensors.
[ERROR] [2024-02-13 09:39:06.481] [thread: 438810] [blf] [YarpSensorBridge::Impl::attachAllInertials] Unable to attach the accelerometer as MAS.
[ERROR] [2024-02-13 09:39:06.481] [thread: 438810] [blf] [YarpSensorBridge::setDriversList] Failed to attach to one or more device drivers.
[ERROR] [2024-02-13 09:39:06.481] [thread: 438810] [blf] [YarpRobotLoggerDevice::attachAll] Could not attach drivers list to sensor bridge.
[ERROR] Device yarp-robot-logger cannot execute attach
[ERROR] Cannot run attach action on device yarp-robot-logger
```

@G-Cervettini did you ever expirienced the same issue?

GiulioRomualdi, 2024-02-13:
I tried to replicate @G-Cervettini setup but the issue is still there

G-Cervettini, 2024-02-13:
> I'm trying to test [icub-tech-iit/ergocub-software#217](https://github.com/icub-tech-iit/ergocub-software/pull/217) with `YarpRobotLoggerDevice`. To do so I modified the configuration file of the logger as [9089463](https://github.com/ami-iit/bipedal-locomotion-framework/commit/90894633d38ff366636528fe4b7b944fffc6f3c5)
> 
> However when I run the device I got
> 
> ```
> [ERROR] [2024-02-13 09:39:06.481] [thread: 438810] [blf] [YarpSensorBridge::Impl::checkAttachedMASSensors] Expecting the same number of MAS sensors attached to the Bridge as many mentioned in the initialization step. Number of MAS sensor in the interface: 0. Number of sensor in initialization: 4.
> [ERROR] [2024-02-13 09:39:06.481] [thread: 438810] [blf] [YarpSensorBridge::Impl::attachAndCheckMASSensors] Could not find at least one of the required sensors.
> [ERROR] [2024-02-13 09:39:06.481] [thread: 438810] [blf] [YarpSensorBridge::Impl::attachAllInertials] Unable to attach the accelerometer as MAS.
> [ERROR] [2024-02-13 09:39:06.481] [thread: 438810] [blf] [YarpSensorBridge::setDriversList] Failed to attach to one or more device drivers.
> [ERROR] [2024-02-13 09:39:06.481] [thread: 438810] [blf] [YarpRobotLoggerDevice::attachAll] Could not attach drivers list to sensor bridge.
> [ERROR] Device yarp-robot-logger cannot execute attach
> [ERROR] Cannot run attach action on device yarp-robot-logger
> ```
> 
> @G-Cervettini did you ever expirienced the same issue?

Looks like the error I had back in time [here](https://github.com/ami-iit/element_floatingbase-exteroceptive-estimation/issues/18#issuecomment-1640252791).

closed 2024-02-13 by GiulioRomualdi

S-Dafarra, 2024-02-15:
cc @rob-mau I don't remember if you were experiencing the same issue the other day

rob-mau, 2024-02-15:
Yes, I was experiencing some similar issue (there was some problem with "l_leg_ft", "r_leg_ft" in the `yarp-robot-logger.xml` configuration file). I solved the problem by updating the `ergocub-software` repo.