gbionics/bipedal-locomotion-framework #812: YARPRobotLoggerDevice: excessively long time horizon for signals logged with YARP_CLOCK

rob-mau, 2024-02-19:
I would like to report a peculiar behavior of the `YARPRobotLoggerDevice`. If I run it with `YARP_CLOCK`:
```
YARP_CLOCK=/clock YARP_ROBOT_NAME=ergoCubGazeboV1_1 yarprobotinterface --config launch-yarp-robot-logger.xml
```

and then I launch the `robot-log-visualizer`, the signal plot displays an excessively long time horizon (on the order of 1e9 seconds!).

[image]

Conversely, when I launch the logger without `YARP_CLOCK`, the amplitude of the time window is consistent with the logger's execution time.

S-Dafarra, 2024-02-22:
Can you add the mat file here?

cc @GiulioRomualdi

GiulioRomualdi, 2024-02-22:
I'm afraid it is something related to the clock when we destroy the device @traversaro

S-Dafarra, 2024-02-23:
When pressing CTRL+C, the ``close`` method should be called (see [here](https://github.com/robotology/yarp/blob/6ca59397df63374d7112b977b9bd425208b6f5c6/src/libYARP_dev/src/yarp/dev/Drivers.cpp#L344-L347), then [here](https://github.com/robotology/yarp/blob/6ca59397df63374d7112b977b9bd425208b6f5c6/src/libYARP_os/src/yarp/os/Terminator.cpp#L48) and finally [here](https://github.com/robotology/yarp/blob/6ca59397df63374d7112b977b9bd425208b6f5c6/src/libYARP_dev/src/yarp/dev/Drivers.cpp#L558)). I noticed that [in the ``close`` method](https://github.com/ami-iit/bipedal-locomotion-framework/blob/eb4b82d12620ba7a3151bb5ea6febb665db01a45/devices/YarpRobotLoggerDevice/src/YarpRobotLoggerDevice.cpp#L1689-L1723) we do not stop the acquisition thread. We probably should add a [``this->askToStop();``](https://www.yarp.it/latest/classyarp_1_1os_1_1PeriodicThread.html#aa92156efefdf172da41077d5962747a2) at the beginning of the ``close`` method

rob-mau, 2024-02-23:
> Can you add the mat file here?

[robot_logger_device_2024_02_16_09_31_31.zip](https://github.com/ami-iit/bipedal-locomotion-framework/files/14385691/robot_logger_device_2024_02_16_09_31_31.zip)

GiulioRomualdi, 2024-02-23:
> When pressing CTRL+C, the `close` method should be called (see [...]

Nice catch! @rob-mau could you try to implement what @S-Dafarra suggests and open a PR for that :) ?

DanielePucci, 2024-02-25:
CC @nicktrem

S-Dafarra, 2024-02-27:
> When pressing CTRL+C, the `close` method should be called (see [...]

Probably we should also substitute the default destructor adding a ``this->stop()`` to make sure that the thread is stopped before starting to destroy stuff.

rob-mau, 2024-03-03:
> rob-mau could you try to implement what S-Dafarra suggests and open a PR for that :) ?

I have implemented the suggested changes and opened a PR. The problem still persists.

GiulioRomualdi, 2024-04-13:
We may double-check the attachments levels as explained in  https://github.com/robotology/robots-configuration/issues/377

GiulioRomualdi, 2024-04-15:
> We may double-check the attachments levels as explained in [robotology/robots-configuration#377](https://github.com/robotology/robots-configuration/issues/377)

I checked them, I fixed one of the detach level still the problem is there.

I noticed that when I press ctrl-c we got

```
^C[WARNING] |yarp.os.NetworkClock| Destroying network clock
[ERROR] [2024-04-15 11:45:01.780] [thread: 65590] [blf] [YarpSensorBridge::Impl::readAllJointSensors] Unable to read from IEncodersTimed interface, use previous measurement.
[ERROR] [2024-04-15 11:45:01.780] [thread: 65590] [blf] [YarpSensorBridge::Impl::readAllJointSensors] Unable to read from ITorqueControl interface, use previous measurement.
[ERROR] |yarp.device.multipleanalogsensorsclient| No data received in the last 1713170648.783669 seconds, timeout enabled.
[ERROR] |yarp.device.multipleanalogsensorsclient| Sensor of type ThreeAxisLinearAccelerometers with index 1 has non-MAS_OK status.
[ERROR] [2024-04-15 11:45:01.780] [thread: 65590] [blf] [YarpSensorBridge::Impl::readMASSensor] Unable to read from r_foot_rear_ft_imu, use previous measurement.
```

So the clock is destroyed while the  thread is still running. As you can notice 
```
[WARNING] |yarp.os.NetworkClock| Destroying network clock
```

is before
```
[ERROR] [2024-04-15 11:45:01.780] [thread: 65590] [blf] [YarpSensorBridge::Impl::readAllJointSensors] Unable to read from IEncodersTimed interface, use previous measurement.
```

GiulioRomualdi, 2024-04-15:
I added a print at the beginning of the `close` and `detachAll` methods.

This is the trace. The clock is destroyed before the `detachAll` is called. That's could be the reason why @rob-mau didn't solve the problem https://github.com/ami-iit/bipedal-locomotion-framework/issues/812#issuecomment-1975200369
```
C[WARNING] |yarp.os.NetworkClock| Destroying network clock
[INFO] |yarp.os.Network| Success: port-to-port persistent connection added.
[INFO] |yarp.os.impl.PortCoreInputUnit|/iiticublap261/yarprobotinterface/69520/clock:i| Removing input from /clock to /iiticublap261/yarprobotinterface/69520/clock:i
[INFO] |yarp.os.RFModule| [try 1 of 3] Trying to shut down.
[WARNING] Interrupt # 1 # received.
[INFO] Interrupt received. Stopping all running threads.
[INFO] All actions for action level 15 of startup phase started. Waiting for unfinished actions.
[INFO] All actions for action level 15 of startup phase finished.
[INFO] startup phase finished.
[INFO] run phase starting...
[INFO] |yarp.os.RFModule| RFModule closing.
[INFO] interrupt1 phase starting...
[INFO] interrupt1 phase finished.
[INFO] shutdown phase starting...
[INFO] Entering action level 2 of phase shutdown
[INFO] Executing detach action, level 2 on device yarp-robot-logger with parameters []
-----> detachAll
[INFO] All actions for action level 2 of shutdown phase started. Waiting for unfinished actions.
[INFO] All actions for action level 2 of shutdown phase finished.
[INFO] Entering action level 20 of phase shutdown
[INFO] Executing detach action, level 20 on device mas-remapper with parameters []
[INFO] All actions for action level 20 of shutdown phase started. Waiting for unfinished actions.
[INFO] All actions for action level 20 of shutdown phase finished.
[INFO] Closing device yarp-robot-logger
-----> close
```

GiulioRomualdi, 2024-04-15:
Perhaps @traversaro and/or @randaz81 have some hints

S-Dafarra, 2024-04-15:
I have noticed that the clock gets destroyed when calling [``removeClock``](https://github.com/robotology/yarp/blob/e4762b0b4a7946bf9379b183379bd32353d54a26/src/libYARP_os/src/yarp/os/Time.cpp#L85), which is done in [``NetworkBase::finiMinimum``](https://github.com/robotology/yarp/blob/e4762b0b4a7946bf9379b183379bd32353d54a26/src/libYARP_os/src/yarp/os/Network.cpp#L948)

GiulioRomualdi, 2024-04-15:
> I have noticed that the clock gets destroyed when calling [`removeClock`](https://github.com/robotology/yarp/blob/e4762b0b4a7946bf9379b183379bd32353d54a26/src/libYARP_os/src/yarp/os/Time.cpp#L85), [...]

That is called by the [destructor of the RFModule](https://github.com/robotology/yarp/blob/e4762b0b4a7946bf9379b183379bd32353d54a26/src/libYARP_os/src/yarp/os/RFModule.cpp#L310-L313)

S-Dafarra, 2024-04-15:
I think I might have an idea. ``yarprobotinterface`` is itself a ``RFModule``. When pressing CTRL-C, it sets the use of ``SystemClock`` that in turns delete the Network clock https://github.com/robotology/yarp/blob/e4762b0b4a7946bf9379b183379bd32353d54a26/src/libYARP_os/src/yarp/os/RFModule.cpp#L246

S-Dafarra, 2024-04-15:
I guess that the problem is not the switch to the ``SystemClock`` per-se. I guess that the big number arises because the network clock and the System clock might use different a different epoch.

GiulioRomualdi, 2024-04-15:
> I guess that the problem is not the switch to the `SystemClock` per-se. I guess that the big number arises because the network clock and the System clock might use a different epoch.

From a theoretical perspective, I don't see anything preventing the two clocks from starting at different epochs 🤔 

In our case, the network clock is generated by https://github.com/robotology/gazebo-yarp-plugins/tree/master/plugins/clock.

The content of the `/clock` port is as follows:
```
0 1000000
0 2000000
0 3000000
0 4000000
0 5000000
0 6000000
0 7000000
0 8000000
0 9000000
0 10000000
```
when I start Gazebo. 

As explained [here](https://github.com/robotology/gazebo-yarp-plugins/blob/29d474fbc674ef0f402cc9b42056bab88f99eb88/plugins/clock/src/Clock.cc#L163C8-L170), the first value represents seconds and the second represents nanoseconds. It starts from 0.0000000 s. However, starting the network clock from the system clock might not be the correct solution, as the network clock, i.e., the Gazebo simulation, could quickly diverge from the system clock (realtime factor different from 1)

GiulioRomualdi, 2024-04-15:
Talking we @S-Dafarra we thought that we could avoid saving the data if the difference between the current time instant and the previous one is greater than a threshold

GiulioRomualdi, 2024-04-18:
Fixed by #839

PR #839 (2024-04-18): Add maximum admissible time step parameter to YarpRobotLoggerDevice
This PR implements what is discussed in https://github.com/ami-iit/bipedal-locomotion-framework/issues/812#issuecomment-2057032884. I tested it and it fixes #812 (@rob-mau)
closed 2024-04-18 by GiulioRomualdi