gbionics/bipedal-locomotion-framework #830: Unable to compile blf if roshumble is installed

GiulioRomualdi, 2024-04-03:
I was trying to compile blf in a pc with roshumbe installed with apt (ergocub001-lap) and I got the following error 
```
[ 70%] Performing configure step for 'bipedal-locomotion-framework'
Not searching for unused variables given on the command line.
loading initial cache file /usr/local/src/robot/robotology-superbuild/build/src/bipedal-locomotion-framework/CMakeFiles/YCMTmp/bipedal-locomotion-framework-cache-Release.cmake
-- Found rosidl_generator_c: 3.1.5 (/opt/ros/humble/share/rosidl_generator_c/cmake)
-- Found rosidl_adapter: 3.1.5 (/opt/ros/humble/share/rosidl_adapter/cmake)
-- Found rosidl_generator_cpp: 3.1.5 (/opt/ros/humble/share/rosidl_generator_cpp/cmake)
CMake Error at /opt/ros/humble/share/rosidl_typesupport_c/cmake/get_used_typesupports.cmake:35 (message):
  No 'rosidl_typesupport_c' found
Call Stack (most recent call first):
  /opt/ros/humble/share/rosidl_typesupport_c/cmake/rosidl_typesupport_c-extras.cmake:8 (get_used_typesupports)
  /opt/ros/humble/share/rosidl_typesupport_c/cmake/rosidl_typesupport_cConfig.cmake:41 (include)
  /opt/ros/humble/share/builtin_interfaces/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /opt/ros/humble/share/builtin_interfaces/cmake/builtin_interfacesConfig.cmake:41 (include)
  /opt/ros/humble/share/libstatistics_collector/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /opt/ros/humble/share/libstatistics_collector/cmake/libstatistics_collectorConfig.cmake:41 (include)
  /opt/ros/humble/share/rclcpp/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /opt/ros/humble/share/rclcpp/cmake/rclcppConfig.cmake:41 (include)
  cmake/BipedalLocomotionFrameworkDependencies.cmake:31 (find_package)
  CMakeLists.txt:21 (include)
```

GiulioRomualdi, 2024-04-03:
Probably something similar to https://github.com/ros2/ros2/issues/1144

traversaro, 2024-04-03:
Probably could be fixed by `ros-humble-rosidl-typesupport-c` . I guess it is something strictly related to the specific set of dependencies installed on the machine. Can you report the `apt list --installed` output?

fyi @SimoneMic

GiulioRomualdi, 2024-04-03:
Here they are [deps.txt](https://github.com/ami-iit/bipedal-locomotion-framework/files/14854476/deps.txt)
`ros-humble-rosidl-typesupport-c` seems to be installed
```
ros-humble-rosidl-typesupport-c/now 2.0.1-1jammy.20231117.180656 amd64 [installed,upgradable to: 2.0.1-1jammy.20240217.042649]
ros-humble-rosidl-typesupport-cpp/now 2.0.1-1jammy.20231117.181033 amd64 [installed,upgradable to: 2.0.1-1jammy.20240217.043731]
```

traversaro, 2024-04-03:
The related cmake code is https://github.com/ros2/rosidl_typesupport/blob/f512ead3afa646c9d0e7e0ee26fe03efdc48aaaa/rosidl_typesupport_c/cmake/get_used_typesupports.cmake#L33C29-L33C52 . Somehow some resources are missing.

GiulioRomualdi, 2024-04-12:
Yesteday I've tried to recompile blf on the same PC I was having issues. I started from a clean build. I didn't have any issue. 🤔  

For the time being I'm closing the issue. In case of problem I will reopen it