gbionics/bipedal-locomotion-framework #859: Unable to run the CentroidalMPC with MA97 on a conda installation

GiulioRomualdi, 2024-06-25:
```
 ** On entry to DGEMM  parameter number  4 had an illegal value
 ** On entry to DGEMM  parameter number  4 had an illegal value
 ** On entry to DGEMM  parameter number  4 had an illegal value
 ** On entry to DGEMM  parameter number  4 had an illegal value
 ** On entry to DGEMM  parameter number  4 had an illegal value
 ** On entry to DGEMM  parameter number  4 had an illegal value
 ** On entry to DGEMM  parameter number  4 had an illegal value
 ** On entry to DGEMM  parameter number  4 had an illegal value
 ** On entry to DGEMM  parameter number  4 had an illegal value
 ** On entry to DGEMM  parameter number  4 had an illegal value
Function solver (0x5a57c1808000)

```

This is the conda env
```
name: robotologyenv
channels:
  - <unknown>
  - conda-forge
dependencies:
  - _libgcc_mutex=0.1=conda_forge
  - _openmp_mutex=4.5=2_kmp_llvm
  - _sysroot_linux-64_curr_repodata_hack=3=h69a702a_14
  - ace=7.1.4=h59595ed_0
  - aiohttp=3.9.3=py312h98912ed_1
  - aiosignal=1.3.1=pyhd8ed1ab_0
  - alsa-lib=1.2.11=hd590300_1
  - ampl-mp=3.1.0=h2cc385e_1006
  - aom=3.7.1=h59595ed_0
  - asio=1.29.0=h59595ed_0
  - assimp=5.4.0=h8343317_0
  - asttokens=2.4.1=pyhd8ed1ab_0
  - atk-1.0=2.38.0=hd4edc92_1
  - attr=2.5.1=h166bdaf_1
  - attrs=23.2.0=pyh71513ae_0
  - bash-completion=2.11=ha770c72_1
  - binutils=2.40=hdd6e379_0
  - binutils_impl_linux-64=2.40=hf600244_0
  - binutils_linux-64=2.40=hdade7a5_3
  - blosc=1.21.5=h0f2a231_0
  - boost=1.84.0=h8da182e_2
  - brotli=1.1.0=hd590300_1
  - brotli-bin=1.1.0=hd590300_1
  - brotli-python=1.1.0=py312h30efb56_1
  - bzip2=1.0.8=hd590300_5
  - c-ares=1.28.1=hd590300_0
  - c-compiler=1.7.0=hd590300_0
  - ca-certificates=2024.2.2=hbcca054_0
  - cached-property=1.5.2=hd8ed1ab_1
  - cached_property=1.5.2=pyha770c72_1
  - cairo=1.18.0=h3faef2a_0
  - certifi=2024.2.2=pyhd8ed1ab_0
  - charset-normalizer=3.3.2=pyhd8ed1ab_0
  - cmake=3.28.3=hcfe8598_0
  - coinhsl=2023.11.17.0=hf90ab62_0
  - colorama=0.4.6=pyhd8ed1ab_0
  - compilers=1.7.0=ha770c72_0
  - contourpy=1.2.1=py312h8572e83_0
  - cuda-version=11.8=h70ddcb2_3
  - cudatoolkit=11.8.0=h4ba93d1_13
  - cudnn=8.9.7.29=hbc23b4c_3
  - cxx-compiler=1.7.0=h00ab1b0_0
  - cycler=0.12.1=pyhd8ed1ab_0
  - dav1d=1.2.1=hd590300_0
  - dbus=1.13.6=h5008d03_3
  - decorator=5.1.1=pyhd8ed1ab_0
  - double-conversion=3.3.0=h59595ed_0
  - eigen=3.4.0=h00ab1b0_0
[... 347 lines cut ...]
  - xorg-libxdmcp=1.1.3=h7f98852_0
  - xorg-libxext=1.3.4=h0b41bf4_2
  - xorg-libxfixes=5.0.3=h7f98852_1004
  - xorg-libxi=1.7.10=h7f98852_0
  - xorg-libxinerama=1.1.5=h27087fc_0
  - xorg-libxrandr=1.5.2=h7f98852_1
  - xorg-libxrender=0.9.11=hd590300_0
  - xorg-libxt=1.3.0=hd590300_1
  - xorg-libxtst=1.2.3=h7f98852_1002
  - xorg-libxxf86vm=1.1.5=h0b41bf4_0
  - xorg-randrproto=1.5.0=h7f98852_1001
  - xorg-recordproto=1.14.2=h7f98852_1002
  - xorg-renderproto=0.11.1=h7f98852_1002
  - xorg-util-macros=1.19.3=h7f98852_0
  - xorg-xextproto=7.3.0=h0b41bf4_1003
  - xorg-xf86vidmodeproto=2.3.1=h7f98852_1002
  - xorg-xproto=7.0.31=h7f98852_1007
  - xz=5.2.6=h166bdaf_0
  - yaml=0.2.5=h7f98852_2
  - yarl=1.9.4=py312h98912ed_0
  - zeromq=4.3.5=h59595ed_1
  - zlib=1.2.13=hd590300_5
  - zstd=1.5.5=hfc55251_0
prefix: /home/ergocub/mambaforge/envs/robotologyenv
```

GiulioRomualdi, 2024-06-25:
cc @traversaro

GiulioRomualdi, 2024-06-25:
While this one is a working configuration 
```
name: dnn-mpc-env
channels:
  - <unknown>
  - conda-forge
dependencies:
  - _libgcc_mutex=0.1=conda_forge
  - _openmp_mutex=4.5=2_gnu
  - _sysroot_linux-64_curr_repodata_hack=3=h69a702a_14
  - ace=8.0.0=hac33072_1
  - aiohttp=3.9.5=py312h98912ed_0
  - aiosignal=1.3.1=pyhd8ed1ab_0
  - alsa-lib=1.2.12=h4ab18f5_0
  - ampl-mp=3.1.0=h2cc385e_1006
  - aom=3.7.1=h59595ed_0
  - asio=1.29.0=h59595ed_0
  - assimp=5.4.1=h8343317_0
  - asttokens=2.4.1=pyhd8ed1ab_0
  - atk-1.0=2.38.0=hd4edc92_1
  - attr=2.5.1=h166bdaf_1
  - attrs=23.2.0=pyh71513ae_0
  - bash-completion=2.11=ha770c72_1
  - binutils=2.40=h4852527_7
  - binutils_impl_linux-64=2.40=ha1999f0_7
  - binutils_linux-64=2.40=hb3c18ed_9
  - blosc=1.21.5=h0f2a231_0
  - boost=1.84.0=h9cebb41_3
  - brotli=1.1.0=hd590300_1
  - brotli-bin=1.1.0=hd590300_1
  - bzip2=1.0.8=hd590300_5
  - c-ares=1.28.1=hd590300_0
  - c-compiler=1.7.0=hd590300_1
  - ca-certificates=2024.6.2=hbcca054_0
  - cached-property=1.5.2=hd8ed1ab_1
  - cached_property=1.5.2=pyha770c72_1
  - cairo=1.18.0=h3faef2a_0
  - certifi=2024.6.2=pyhd8ed1ab_0
  - cmake=3.28.3=hcfe8598_0
  - coinhsl=2023.11.17.0=hf90ab62_0
  - compilers=1.7.0=ha770c72_1
  - contourpy=1.2.1=py312h8572e83_0
  - cudatoolkit=11.8.0=h4ba93d1_13
  - cxx-compiler=1.7.0=h00ab1b0_1
  - cycler=0.12.1=pyhd8ed1ab_0
  - dav1d=1.2.1=hd590300_0
  - dbus=1.13.6=h5008d03_3
  - decorator=5.1.1=pyhd8ed1ab_0
  - double-conversion=3.3.0=h59595ed_0
  - eigen=3.4.0=h00ab1b0_0
  - exceptiongroup=1.2.0=pyhd8ed1ab_2
  - executing=2.0.1=pyhd8ed1ab_0
  - expat=2.5.0=hcb278e6_1
  - ffmpeg=6.1.1=gpl_h186bccc_100
  - flann=1.9.2=h2b5ea80_0
  - fmt=10.2.1=h00ab1b0_0
  - font-ttf-dejavu-sans-mono=2.37=hab24e00_0
  - font-ttf-inconsolata=3.000=h77eed37_0
  - font-ttf-source-code-pro=2.038=h77eed37_0
  - font-ttf-ubuntu=0.83=h77eed37_2
  - fontconfig=2.14.2=h14ed4e7_0
  - fonts-conda-ecosystem=1=0
  - fonts-conda-forge=1=0
  - fonttools=4.53.0=py312h9a8786e_0
  - fortran-compiler=1.7.0=heb67821_1
  - freeglut=3.2.2=hac7e632_2
  - freetype=2.12.1=h267a509_2
  - fribidi=1.0.10=h36c2ea0_0
  - frozenlist=1.4.1=py312h98912ed_0
  - gcc=12.3.0=h915e2ae_13
[... 294 lines cut ...]
  - xorg-libxdmcp=1.1.3=h7f98852_0
  - xorg-libxext=1.3.4=h0b41bf4_2
  - xorg-libxfixes=5.0.3=h7f98852_1004
  - xorg-libxi=1.7.10=h7f98852_0
  - xorg-libxinerama=1.1.5=h27087fc_0
  - xorg-libxrandr=1.5.2=h7f98852_1
  - xorg-libxrender=0.9.11=hd590300_0
  - xorg-libxt=1.3.0=hd590300_1
  - xorg-libxtst=1.2.3=h7f98852_1002
  - xorg-libxxf86vm=1.1.5=h0b41bf4_0
  - xorg-randrproto=1.5.0=h7f98852_1001
  - xorg-recordproto=1.14.2=h7f98852_1002
  - xorg-renderproto=0.11.1=h7f98852_1002
  - xorg-util-macros=1.19.3=h7f98852_0
  - xorg-xextproto=7.3.0=h0b41bf4_1003
  - xorg-xf86vidmodeproto=2.3.1=h7f98852_1002
  - xorg-xproto=7.0.31=h7f98852_1007
  - xz=5.2.6=h166bdaf_0
  - yaml=0.2.5=h7f98852_2
  - yarl=1.9.4=py312h98912ed_0
  - zeromq=4.3.5=h75354e8_4
  - zlib=1.2.13=h4ab18f5_6
  - zstd=1.5.6=ha6fb4c9_0
prefix: /home/ergocub/mambaforge/envs/dnn-mpc-env
```

traversaro, 2024-06-26:
diff:
~~~
--- <unnamed>
+++ <unnamed>
@@ -1,46 +1,51 @@
-this is the working environment
-name: dnn-mpc-env
+this is the non working environment
+name: robotologyenv
 channels:
   - <unknown>
   - conda-forge
 dependencies:
   - _libgcc_mutex=0.1=conda_forge
-  - _openmp_mutex=4.5=2_gnu
+  - _openmp_mutex=4.5=2_kmp_llvm
   - _sysroot_linux-64_curr_repodata_hack=3=h69a702a_14
-  - ace=8.0.0=hac33072_1
-  - aiohttp=3.9.5=py312h98912ed_0
+  - ace=7.1.4=h59595ed_0
+  - aiohttp=3.9.3=py312h98912ed_1
   - aiosignal=1.3.1=pyhd8ed1ab_0
-  - alsa-lib=1.2.12=h4ab18f5_0
+  - alsa-lib=1.2.11=hd590300_1
   - ampl-mp=3.1.0=h2cc385e_1006
   - aom=3.7.1=h59595ed_0
   - asio=1.29.0=h59595ed_0
-  - assimp=5.4.1=h8343317_0
+  - assimp=5.4.0=h8343317_0
   - asttokens=2.4.1=pyhd8ed1ab_0
   - atk-1.0=2.38.0=hd4edc92_1
   - attr=2.5.1=h166bdaf_1
   - attrs=23.2.0=pyh71513ae_0
   - bash-completion=2.11=ha770c72_1
-  - binutils=2.40=h4852527_7
-  - binutils_impl_linux-64=2.40=ha1999f0_7
-  - binutils_linux-64=2.40=hb3c18ed_9
+  - binutils=2.40=hdd6e379_0
+  - binutils_impl_linux-64=2.40=hf600244_0
+  - binutils_linux-64=2.40=hdade7a5_3
   - blosc=1.21.5=h0f2a231_0
-  - boost=1.84.0=h9cebb41_3
+  - boost=1.84.0=h8da182e_2
   - brotli=1.1.0=hd590300_1
   - brotli-bin=1.1.0=hd590300_1
+  - brotli-python=1.1.0=py312h30efb56_1
   - bzip2=1.0.8=hd590300_5
   - c-ares=1.28.1=hd590300_0
-  - c-compiler=1.7.0=hd590300_1
-  - ca-certificates=2024.6.2=hbcca054_0
+  - c-compiler=1.7.0=hd590300_0
+  - ca-certificates=2024.2.2=hbcca054_0
   - cached-property=1.5.2=hd8ed1ab_1
   - cached_property=1.5.2=pyha770c72_1
   - cairo=1.18.0=h3faef2a_0
-  - certifi=2024.6.2=pyhd8ed1ab_0
+  - certifi=2024.2.2=pyhd8ed1ab_0
+  - charset-normalizer=3.3.2=pyhd8ed1ab_0
   - cmake=3.28.3=hcfe8598_0
   - coinhsl=2023.11.17.0=hf90ab62_0
-  - compilers=1.7.0=ha770c72_1
+  - colorama=0.4.6=pyhd8ed1ab_0
+  - compilers=1.7.0=ha770c72_0
   - contourpy=1.2.1=py312h8572e83_0
+  - cuda-version=11.8=h70ddcb2_3
   - cudatoolkit=11.8.0=h4ba93d1_13
-  - cxx-compiler=1.7.0=h00ab1b0_1
+  - cudnn=8.9.7.29=hbc23b4c_3
+  - cxx-compiler=1.7.0=h00ab1b0_0
   - cycler=0.12.1=pyhd8ed1ab_0
[... 422 lines cut ...]
   - xcb-util=0.4.0=hd590300_1
@@ -349,7 +387,7 @@
   - xcb-util-keysyms=0.4.0=h8ee46fc_1
   - xcb-util-renderutil=0.3.9=hd590300_1
   - xcb-util-wm=0.4.1=h8ee46fc_1
-  - xkeyboard-config=2.42=h4ab18f5_0
+  - xkeyboard-config=2.41=hd590300_0
   - xorg-compositeproto=0.4.2=h7f98852_1001
   - xorg-damageproto=1.2.1=h7f98852_1002
   - xorg-fixesproto=5.0=h7f98852_1002
@@ -381,7 +419,7 @@
   - xz=5.2.6=h166bdaf_0
   - yaml=0.2.5=h7f98852_2
   - yarl=1.9.4=py312h98912ed_0
-  - zeromq=4.3.5=h75354e8_4
-  - zlib=1.2.13=h4ab18f5_6
-  - zstd=1.5.6=ha6fb4c9_0
-prefix: /home/ergocub/mambaforge/envs/dnn-mpc-env
+  - zeromq=4.3.5=h59595ed_1
+  - zlib=1.2.13=hd590300_5
+  - zstd=1.5.5=hfc55251_0
+prefix: /home/ergocub/mambaforge/envs/robotologyenv
~~~

The most relevant change seems the ipopt version change. In theory I can run the `CentroidalMPC` to try to reproduce the problem?

S-Dafarra, 2024-07-01:
So actually the working configuration has a more recent ``ipopt`` version?

traversaro, 2024-07-01:
> So actually the working configuration has a more recent `ipopt` version?

Yes.

GiulioRomualdi, 2024-10-11:
Given https://github.com/ami-iit/bipedal-locomotion-framework/issues/859#issuecomment-2189658510 I would close the issue

closed 2024-10-11 by GiulioRomualdi