gbionics/bipedal-locomotion-framework #875: Make encoders acceleration measurements optional in Yarp Sensors Bridge

traversaro, 2024-08-13:
The YARP sensorbridge at the moment requires the joint acceleration reading to be available (see https://github.com/ami-iit/bipedal-locomotion-framework/blob/4ea37800355531fe4c78f12a4ed0aa426a748ed3/src/RobotInterface/YarpImplementation/include/BipedalLocomotion/RobotInterface/YarpSensorBridgeImpl.h#L1537). However, it is quite common for robots not to provide acceleration readings/measurements/estimates, so I think it could make sense to permit to disable acceleration readings.

fyi @LoreMoretti

LoreMoretti, 2024-08-13:
Maybe the following could work.

Change the [YarpSensorBridgeImpl::readAllJointSensors](https://github.com/ami-iit/bipedal-locomotion-framework/blob/4ea37800355531fe4c78f12a4ed0aa426a748ed3/src/RobotInterface/YarpImplementation/include/BipedalLocomotion/RobotInterface/YarpSensorBridgeImpl.h#L1522) signature to something like:

```
bool readAllJointSensors(bool checkForNan = false, bool readEncoderAccelerations = true)
```

and change the subsequent logic so that the accelerations get read only if `readEncoderAccelerations == true`.

Then add a member function to [YarpSensorBridge](https://github.com/ami-iit/bipedal-locomotion-framework/blob/4ea37800355531fe4c78f12a4ed0aa426a748ed3/src/RobotInterface/YarpImplementation/src/YarpSensorBridge.cpp), like:

```
void YarpSensorBridge::disableEncoderAccelerationReading()
{
     m_readEncoderAcceleration = false;
}
```

Which can be called by the user at the initialization stage based on a custom and user-defined logic. This member function sets a new boolean data member `m_readEncoderAcceleration` to false.

Afterwards change all the calls to  `YarpSensorBridgeImpl::readAllJointSensors` so that they use `m_readEncoderAcceleration` as the second input argument.

traversaro, 2024-08-13:
Yes, I think that could make sense!

GiulioRomualdi, 2024-08-19:
Solved by https://github.com/ami-iit/bipedal-locomotion-framework/pull/876

closed 2024-08-19 by GiulioRomualdi