gbionics/bipedal-locomotion-framework #877: YarpLoggerDevice cannot log data at 1kHz

LoreMoretti, 2024-08-22:
We have run some data collection campaign using the **YarpLoggerDevice** to record and save data at different sample rates.

But, we have realized that **YarpLoggerDevice** was not able to log data at `1 kHz`.

Therefore, we have tested **YarpLoggerDevice** with 4 different sample rates, and we got:

1. at 1[kHz],   `max(dt) = 0.0025 [s]`    (`dt` should be `0.001 [s]` to respect real time)
2. at 500[Hz], `max(dt) = 0.0025 [s]`  (`dt` should be `0.002 [s]` to respect real time)
3. at 200[Hz], `max(dt) = 0.0056 [s]`  (`dt` should be `0.005 [s]` to respect real time)
4. at 100[Hz], `max(dt) = 0.0105 [s]`  (`dt` should be `0.010 [s]` to respect real time)

with `dt` being the time between one sample and the following one.

It follows also the `dt` plots for the 4 different datasets:

[image]

Also, we checked that, within each dataset, the logged vectors (such as joint position, joint velocity, joint acceleration) have the same size (i.e. number of samples).

Lastly, the **blf** version we have been using is https://github.com/isorrentino/bipedal-locomotion-framework/commit/3d371885c22a1f896c07a0843466a4e34fb9009d.

cc @GiulioRomualdi @isorrentino

GiulioRomualdi, 2024-08-22:
Hi @LoreMoretti, thank you for posting this :) You are right. According to the plots you show, the fact that when you collect at 1kHz, you get a maximum delta t of 0.0025s, is related to an outlier. I believe this outlier can be removed. Regardless, it seems that when we set a specific frequency, we consistently get a maximum error of 0.0006s (0.6ms). This error is negligible if we collect data at 100Hz, but it becomes significant when collecting data at 1kHz.

A possibility is:
1. remove the outliner
2. then Resample the data at the desired frequency

GiulioRomualdi, 2024-08-22:
cc also @traversaro @S-Dafarra

traversaro, 2024-08-22:
By the way, more in general if you want to reduce the jitter in the logging time, you need to reduce the scheduling jitter of thread. This can be done by the classical methods used for real time computing, that is:
* Increase the performance of the thread
* Change the OS scheduler used
* Change the linux kernel used and switch to use one that uses the so-called RT patches

An introduction to this topics I like a lot is the "Real-time programming with Linux" series of blog posts by Shuhao Wu: https://shuhaowu.com/blogseries.html#rt-linux-programming . The https://github.com/cactusdynamics/cactus-rt repo from the same author is good example of those techniques, even if to use them in our codebase I would prefer to have some kind of abstraction that is portable also to macOS on Windows (even using non-realtime features in those cases).

LoreMoretti, 2024-08-26:
Also, after reading the resources shared by @traversaro and others (like https://www.youtube.com/watch?v=NrjXEaTSyrw), we could also investigate the benefits brought by changing the current logic of [`Yarp:os:PeriodicThread`](https://github.com/robotology/yarp/blob/2fd8e6993755b1d6e2a3b61d05313ea28255b638/src/libYARP_os/src/yarp/os/PeriodicThread.cpp#L256-L266).

Now, if I got it right, the thread it is set to sleep for an amount of time given by the difference `next_awaken_time - elapsed_time`. But the awakening process from the scheduler takes some time, this means that the tread could wake up a bit later than expected.

We could change it to something like sleep for `next_awaken_time - (elapsed_time + awakening_time)` (where `awakening_time ` has to be determined empirically). The idea is to awake the thread a bit before, and then insert in the `run` function some `while` loop that keeps checking if it is time to start the operations of the thread.

The drawback could be that the CPU is kept busied with more not useful operations (i.e. the `while` loop), but the thread jitter might be less.

Without testing it, I am not sure if this could lead to improvements and also the entity of the improvements, and also how this code change would affect all other software we use.

LoreMoretti, 2024-08-26:
Another idea that came to my mind (reading for example https://unix.stackexchange.com/questions/326579/how-to-ensure-exclusive-cpu-availability-for-a-running-process) is to try to run the logger on a separate laptop reserving one CPU core just for the YarpLoggerDevice application. 

Also here, I guess that it is difficult to determine upfront the entity of the improvements (also compared to the alternative solution of just [increasing](https://manpages.ubuntu.com/manpages/focal/en/man2/nice.2.html#:~:text=nice()%20adds%20inc%20to,are%20clamped%20to%20the%20range.) the priority of the `YarpLoggerDevice` thread)

LoreMoretti, 2024-08-27:
I have tried to implement what I explained in [this comment](https://github.com/ami-iit/bipedal-locomotion-framework/issues/877#issuecomment-2309634852) through this [commit](url). I have hard-coded some numeric values in the code, but this is just inteded for testing at this stage.

Then, I have run the `YarpLoggerDevice` to record some data at `1kHz` on `ergocubSN001` from my laptop with ethernet:

[image]

With mean values being:

- `0.001069588421232` for **yarp3.9**
- `0.001047208306053` for **yarp3.9 with modification**

So the modification seems to improve a bit the `YarpLoggerDevice` performances.

I could run more tests to check if the statistics are confirmed. Still, perhaps these tiny improvements aren't worth changing that piece of YARP.

S-Dafarra, 2024-08-27:
The variance seems well reduced anyhow!

FYI the link to the commit appears to be broken

LoreMoretti, 2024-08-27:
> FYI the link to the commit appears to be broken

Thanks for the catch! I should have fixed it

S-Dafarra, 2024-08-27:
> Still, perhaps these tiny improvements aren't worth changing that piece of YARP.

In any way, it is something that can be done inside the ``run`` method (subtracting from the user-specified period the awakening time). Also, if the problem is the awakening, then you should avoid the delay in https://github.com/LoreMoretti/yarp/commit/8a44daa286594eda6b7f0df9c5ef6b927d33e14c#diff-d271dd18aa050a91366a7acdb6a223823dcb1771603035cace056e6fc5f337d6R243 because otherwise you end up having the same issue.

LoreMoretti, 2024-08-31:
> In any way, it is something that can be done inside the run method (subtracting from the user-specified period the awakening time)

I have modified the `YarpLoggerDevice`. Some preliminary code can be found at this [commit](https://github.com/LoreMoretti/bipedal-locomotion-framework/commit/8cdc1c5235db5ea73f22f967deef7a42cd685085).

Then to test it, I have run the `YarpLoggerDevice` to record some data at `1kHz` on `ergocubSN001` from my laptop with ethernet connection. It follows the `dt` of the data:

[image]

With mean values being:

- `0.001059455351969` for **original** `YarpLoggerDevice`
- `0.001000461184516` for **modified** `YarpLoggerDevice`

In both cases, I have being using **yarp3.9**, the only difference is within the `YarpLoggerDevice` itself.

More tests should be run to confirm the statistics, but the preliminary results seem good.

traversaro, 2024-09-02:
Can we have an option to disable busy waiting? There are many context in which you do not want to do busy waiting at all.

> Increase the performance of the thread

Ahh, my bad, this sentence was meant to write "increse the priority of the thread". If you did not tried that, I would try, it is a simple linux-specific code you can wrap in a #ifdef block, and I guess it can be really helpful even without a realtime kernel (that we may want to explore the use anyhow if we are interested in this, the early wakeup is the last of the strategies listed in https://shuhaowu.com/blogseries.html#rt-linux-programming, not the first :D ).

traversaro, 2024-09-02:
Even without changing the priority, at least changing the scheduler used to use one of the realtime schedulers `SCHED_FIFO` ?

traversaro, 2024-09-02:
> Even without changing the priority, at least changing the scheduler used to use one of the realtime schedulers `SCHED_FIFO` ?

See https://stackoverflow.com/questions/51100861/stdthread-with-pthread-setschedparam-with-c-11 and the last suggestion of https://chatgpt.com/share/9e47ab2c-1302-433f-aa5a-5caa25b296ef .

LoreMoretti, 2024-09-02:
> See https://stackoverflow.com/questions/51100861/stdthread-with-pthread-setschedparam-with-c-11 and the last suggestion of https://chatgpt.com/share/9e47ab2c-1302-433f-aa5a-5caa25b296ef .

I have implemented it at this [commit](https://github.com/LoreMoretti/bipedal-locomotion-framework/commit/77f58bafbf71f267b4620c87cba6200e9a2a37cf).

Then to test it, I have run the `YarpLoggerDevice` to record some data at `1kHz` on `ergocubSN001` from my laptop with ethernet connection. It follows the `dt` of the data:

[image]

Here the thread priority used:

```
[INFO] [2024-09-02 18:03:10.482] [thread: 70526] [blf] [YarpRobotLoggerDevice::threadInit] Scheduling policy set to SCHED_FIFO with priority 80

```

A part from something happening at the beginning, it seems that increasing the thread priority and set a FIFO policy, helped to reduce the number of spikes. It could be interesting to combine it with the `awakening before` strategy of the green curve.

The `original` and `modified` are the same curves as the one of this [comment](https://github.com/ami-iit/bipedal-locomotion-framework/issues/877#issuecomment-2322945485).

Note that, the first time I run the YarpLoggerDevice, I was getting this [error](https://gist.github.com/kaiwan/cd4985f3dbfeb8dfc44d1f0e4ee67dec#file-errno_codes-txt-L12). Therefore, with the help of @S-Dafarra, we had to `sudo` a bunch of things (`yarp namespace`, `yarp conf`, `YARP_ROBOT_NAME`, `source ~/robotology/robotology-superbuild/build/install/share/robotology-superbuild/setup.sh`) and run `yarprobotinterface --config launch-yarp-robot-logger.xml` with root privileges.

LoreMoretti, 2024-09-03:
> It could be interesting to combine it with the awakening before strategy of the green curve.

I have implemented the needed code at this [commit](https://github.com/LoreMoretti/bipedal-locomotion-framework/commit/d63a499339fe6cef1add8c96a76a1410652eb8ff). 
Then I did a successful smoke test running the `YarpLoggerDevice` with a Gazebo simulation. I will test in on the robot to determine the performances.

Regarding:

> Note that, the first time I run the YarpLoggerDevice, I was getting this [error](https://gist.github.com/kaiwan/cd4985f3dbfeb8dfc44d1f0e4ee67dec#file-errno_codes-txt-L12). Therefore, with the help [...]

I tried different strategies to avoid to `sudo` all the things.

1. [set CAP_SYS_NICE capability](https://stackoverflow.com/questions/7635515/how-to-set-cap-sys-nice-capability-to-a-linux-user), thanks @traversaro for the hint.

But, I could not make it work. This is because after I did `sudo setcap 'cap_sys_nice=eip' </home/lomoretti/robotology/robotology-superbuild/build/install/bin/yarprobotinterface>`, when I run the `yarprobotinterface` (to start the logger device), I get:

`yarprobotinterface: error while loading shared libraries: libYARP_robotinterface.so.3: cannot open shared object file: No such file or directory
`

But, checking the application `rpath`, with `readelf -d /home/lomoretti/robotology/robotology-superbuild/build/install/bin/yarprobotinterface | grep RPATH`, I get:

```
 0x000000000000000f (RPATH)              Library rpath: [/home/lomoretti/mambaforge/envs/robsub_gzsim/lib:$ORIGIN/:$ORIGIN/../lib:$ORIGIN/../]
```

and `libYARP_robotinterface.so.3` should be included in `$ORIGIN/../lib`, being `$ORIGIN` the directory where the application (i.e. `yarprobotinterface`) is located.

Though, if I trace the application with `strace yarprobotinterface 2>&1` it seems that `$ORIGIN/../lib` is not used as search path, as shown in the following terminal log.

<details>
  <summary>strace log</summary>
  
  ```
  execve("/home/lomoretti/robotology/robotology-superbuild/build/install/bin/yarprobotinterface", ["yarprobotinterface"], 0x7ffded454340 /* 150 vars */) = 0
access("/etc/suid-debug", F_OK)         = -1 ENOENT (No such file or directory)
brk(NULL)                               = 0x55b482f94000
arch_prctl(0x3001 /* ARCH_??? */, 0x7ffda9db0f60) = -1 EINVAL (Invalid argument)
fcntl(0, F_GETFD)                       = 0
fcntl(1, F_GETFD)                       = 0
fcntl(2, F_GETFD)                       = 0
access("/etc/suid-debug", F_OK)         = -1 ENOENT (No such file or directory)
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7faf7c693000
readlink("/proc/self/exe", "/home/lomoretti/robotology/robot"..., 4096) = 85
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/lomoretti/mambaforge/envs/robsub_gzsim/lib/glibc-hwcaps/x86-64-v4/libYARP_robotinterface.so.3", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/home/lomoretti/mambaforge/envs/robsub_gzsim/lib/glibc-hwcaps/x86-64-v4", 0x7ffda9db0180, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/lomoretti/mambaforge/envs/robsub_gzsim/lib/glibc-hwcaps/x86-64-v3/libYARP_robotinterface.so.3", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/home/lomoretti/mambaforge/envs/robsub_gzsim/lib/glibc-hwcaps/x86-64-v3", 0x7ffda9db0180, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/lomoretti/mambaforge/envs/robsub_gzsim/lib/glibc-hwcaps/x86-64-v2/libYARP_robotinterface.so.3", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/home/lomoretti/mambaforge/envs/robsub_gzsim/lib/glibc-hwcaps/x86-64-v2", 0x7ffda9db0180, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/lomoretti/mambaforge/envs/robsub_gzsim/lib/tls/haswell/avx512_1/x86_64/libYARP_robotinterface.so.3", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/home/lomoretti/mambaforge/envs/robsub_gzsim/lib/tls/haswell/avx512_1/x86_64", 0x7ffda9db0180, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/lomoretti/mambaforge/envs/robsub_gzsim/lib/tls/haswell/avx512_1/libYARP_robotinterface.so.3", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/home/lomoretti/mambaforge/envs/robsub_gzsim/lib/tls/haswell/avx512_1", 0x7ffda9db0180, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/lomoretti/mambaforge/envs/robsub_gzsim/lib/tls/haswell/x86_64/libYARP_robotinterface.so.3", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/home/lomoretti/mambaforge/envs/robsub_gzsim/lib/tls/haswell/x86_64", 0x7ffda9db0180, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/lomoretti/mambaforge/envs/robsub_gzsim/lib/tls/haswell/libYARP_robotinterface.so.3", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/home/lomoretti/mambaforge/envs/robsub_gzsim/lib/tls/haswell", 0x7ffda9db0180, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/lomoretti/mambaforge/envs/robsub_gzsim/lib/tls/avx512_1/x86_64/libYARP_robotinterface.so.3", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/home/lomoretti/mambaforge/envs/robsub_gzsim/lib/tls/avx512_1/x86_64", 0x7ffda9db0180, 0) = -1 ENOENT (No such file or directory)
[... cut ...] ry
) = 145
exit_group(127)                         = ?
+++ exited with 127 +++

  ```
</details>

I am not sure why. I have built the robotology-superbuild from source (with conda dependencies), maybe it works if one installs directly the conda binary packages of the superbuild.
 
 
 2. ` sudo -E /home/lomoretti/robotology/robotology-superbuild/build/install/bin/yarprobotinterface --config launch-yarp-robot-logger.xml`. This seems to work fine on my laptop (wihtout **"sudo-ing"** anything else).

GiulioRomualdi, 2024-09-03:
We should understand if the created matfile has the right reading access

LoreMoretti, 2024-09-04:
> I will test in on the robot to determine the performances.

I have run the `YarpLoggerDevice` to record some data at `1kHz` on `ergocubSN001` from my laptop with ethernet connection. It follows the `dt` of the data:

[image]

The `blue curve` refers to the YarpRobotLoggerDevice with the **"awake before feature"** (as introduced [here](https://github.com/ami-iit/bipedal-locomotion-framework/issues/877#issuecomment-2322945485)). The `blue curve` refers to the YarpRobotLoggerDevice with the same **"awake before feature"** but also with FIFO policy and increased priority of the thread.

Qualitatively, it seems that the orange curve exhibits less spikes (therefore improving the performances). Of course, I should run more tests to confirm the statics.

Also, at this [commit](https://github.com/LoreMoretti/bipedal-locomotion-framework/commit/d63a499339fe6cef1add8c96a76a1410652eb8ff) 
I have added a config parameter called `enable_real_time_scheduling` which enables (when set to `true`) both the **"awake before feature"** as well as the FIFO policy and higher priority. In this way the user can choose to use it or not. If not specified, `enable_real_time_scheduling` is by default set to `false`.

I can open a PR to add the changes of the YarpRobotLoggerDevice that allowed to reach these perfomances.

Instead, regarding:

> We should understand if the created matfile has the right reading access

I was able to load the created matfile on `robot-log-visualizer` or on `matlab`. Checking the file permission, I get:

```
(robsub_gzsim) lomoretti@IITICUBLAP226:~/Desktop/ExperimentsLorenzo/debug/YarpLoggerTime/modifiedBLF_withFIFOscheduler$ ls -l robot_logger_device_2024_09_04_14_49_01.mat 
-rw-r--r-- 1 root root 361014952 Sep  4 14:49 robot_logger_device_2024_09_04_14_49_01.mat
```

So it seems that also **others from root** can read the file.

LoreMoretti, 2024-09-10:
I am preparing a PR: #883