gbionics/bipedal-locomotion-framework #882: Python tests timeout in conda-forge Ubuntu CI

traversaro, 2024-09-10:


traversaro, 2024-09-10:
Last successful build: https://github.com/ami-iit/bipedal-locomotion-framework/actions/runs/10736770024/job/29776870295 

First failing build: https://github.com/ami-iit/bipedal-locomotion-framework/actions/runs/10740675945/job/29789331628

traversaro, 2024-09-10:
Diff of working vs non-working CI:

~~~
--- <unnamed>
+++ <unnamed>
@@ -81,7 +81,7 @@
 hdf4                      4.2.15               h2a13503_7    conda-forge
 hdf5                      1.14.3          nompi_hdf9ad27_105    conda-forge
 icu                       73.2                 h59595ed_0    conda-forge
-icub-models               2.6.0           py311h4332511_0    conda-forge
+icub-models               2.7.0           py311hfdbb021_1    conda-forge
 idna                      3.8                pyhd8ed1ab_0    conda-forge
 idyntree                  12.3.0          py311h7a77ce7_0    conda-forge
 imath                     3.1.11               hfc55251_0    conda-forge
@@ -222,8 +222,8 @@
 mpg123                    1.32.6               h59595ed_0    conda-forge
 msgpack-python            1.0.8           py311hd18a35c_1    conda-forge
 multidict                 6.0.5           py311h9ecbd09_1    conda-forge
-mumps-include             5.7.3                ha770c72_2    conda-forge
-mumps-seq                 5.7.3                h96124f5_2    conda-forge
+mumps-include             5.7.3                ha770c72_3    conda-forge
+mumps-seq                 5.7.3                h96124f5_3    conda-forge
 mysql-common              8.0.33               hf1915f5_6    conda-forge
 mysql-libs                8.0.33               hca2cd23_6    conda-forge
 ncurses                   6.5                  he02047a_1    conda-forge
@@ -412,7 +412,7 @@
 yarl                      1.9.11          py311h9ecbd09_0    conda-forge
 yarp                      3.9.0                ha770c72_4    conda-forge
 yarp-python               3.9.0           py311hb03c08e_4    conda-forge
-ycm-cmake-modules         0.16.8               h5888daf_0    conda-forge
+ycm-cmake-modules         0.16.9               h5888daf_0    conda-forge
 zipp                      3.20.1             pyhd8ed1ab_0    conda-forge
 zlib                      1.3.1                h4ab18f5_1    conda-forge
 zstd                      1.5.6                ha6fb4c9_0    conda-forge
~~~

However, something else was noticeable before and after:

~~~
manif                     0.0.4               he02047a_17    conda-forge
manifpy                   0.0.5           py311hd0835d6_0    conda-forge
~~~

this is important because:
* using different manif and manifpy versions will not work in theory. Probably this is something that we may want to solve at the manif package level in conda-forge
* Why the latest manifpy is not used? That means we are not using https://github.com/conda-forge/manif-feedstock/pull/48, and so still using the old GCC with all the related pybind11-problems described in https://github.com/conda-forge/pybind11-feedstock/issues/77 . 

However, I do not know why this results in the test going in timeout instead of the exception we are used to when using different compilers.

traversaro, 2024-09-10:
Trying to force manif 0.0.4 results in this error:

~~~
(blfdev) traversaro@IITBMP014LW012:~/bipedal-locomotion-framework$ conda install cmake compilers make ninja pkg-config                       "idyntree>=12.2.1" "yarp>=3.5.0" libmatio libmatio-cpp librobometry                       liblie-group-controllers eigen qhull "casadi>=3.5.5" cppad spdlog                       nlohmann_json manif==0.0.5 manifpy==0.0.5 pybind11 numpy pytest scipy opencv pcl                       tomlplusplus libunicycle-footstep-planner "icub-models>=1.
23.4"                       ros-humble-rclcpp onnxruntime-cpp libbayes-filters-lib cmake-package-check
Channels:
 - robostack-staging
 - conda-forge
Platform: linux-64
Collecting package metadata (repodata.json): done
Solving environment: failed

LibMambaUnsatisfiableError: Encountered problems while solving:
  - nothing provides libopencv 4.2.0 py36_5 needed by opencv-4.2.0-py36_5

Could not solve for environment specs
The following packages are incompatible
├─ casadi >=3.5.5  is installable with the potential options
│  ├─ casadi [3.5.5|3.6.0|3.6.2] would require
│  │  └─ libosqp >=0.6.2,<0.6.3.0a0 , which can be installed;
│  ├─ casadi 3.5.5 would require
│  │  └─ python_abi 3.6.* *_cp36m, which can be installed;
│  ├─ casadi 3.5.5 would require
│  │  └─ python_abi 3.7.* *_cp37m, which can be installed;
│  ├─ casadi [3.5.5|3.6.2|3.6.3] would require
│  │  └─ python_abi 3.8.* *_cp38, which can be installed;
│  ├─ casadi [3.5.5|3.6.2|3.6.3] would require
│  │  └─ python_abi 3.9.* *_cp39, which can be installed;
│  ├─ casadi [3.6.2|3.6.3] would require
│  │  └─ python_abi 3.10.* *_cp310, which can be installed;
│  ├─ casadi [3.6.2|3.6.3] would require
│  │  └─ python_abi 3.11.* *_cp311, which can be installed;
│  ├─ casadi 3.6.3 would require
│  │  └─ ipopt >=3.14.13,<3.14.14.0a0 , which can be installed;
│  ├─ casadi [3.6.3|3.6.4|3.6.5|3.6.6] would require
│  │  └─ tinyxml2 >=10.0.0,<11.0a0 , which can be installed;
│  └─ casadi 3.6.3 would require
│     └─ python >=3.12,<3.13.0a0 , which requires
│        └─ python_abi 3.12.* *_cp312, which conflicts with any installable versions previously reported;
├─ idyntree >=12.2.1  is installable with the potential options
│  ├─ idyntree [12.2.1|12.3.0] would require
│  │  └─ libxml2 [>=2.12.6,<3.0a0 |>=2.12.7,<3.0a0 ] with the potential options
│  │     ├─ libxml2 2.12.7 would require
│  │     │  ├─ icu >=73.2,<74.0a0 , which can be installed;
│  │     │  └─ libzlib [>=1.3.1,<1.4.0a0 |>=1.3.1,<2.0a0 ] with the potential options
│  │     │     ├─ libzlib 1.3.1 would require
│  │     │     │  └─ zlib 1.3.1 *_1, which can be installed;
│  │     │     └─ libzlib 1.3.1 would require
│  │     │        └─ zlib 1.3.1 *_0, which can be installed;
│  │     ├─ libxml2 [2.12.6|2.12.7] would require
│  │     │  └─ icu >=73.2,<74.0a0 , which can be installed;
│  │     └─ libxml2 2.12.7 would require
│  │        ├─ icu >=75.1,<76.0a0 , which conflicts with any installable versions previously reported;
│  │        └─ libzlib >=1.3.1,<2.0a0 , which can be installed (as previously explained);
│  ├─ idyntree [12.3.0|12.3.1|12.3.2|12.3.3|12.4.0] would require
│  │  ├─ assimp >=5.4.1,<5.4.2.0a0 , which requires
│  │  │  ├─ libboost >=1.84.0,<1.85.0a0  with the potential options
│  │  │  │  ├─ libboost 1.84.0 would require
│  │  │  │  │  ├─ boost-cpp 1.84.0* , which can be installed;
│  │  │  │  │  ├─ icu >=75.1,<76.0a0 , which conflicts with any installable versions previously reported;
│  │  │  │  │  └─ libzlib >=1.3.1,<2.0a0 , which can be installed (as previously explained);
│  │  │  │  └─ libboost 1.84.0 would require
│  │  │  │     └─ boost-cpp 1.84.0* , which can be installed;
│  │  │  └─ libzlib >=1.3.1,<2.0a0 , which can be installed (as previously explained);
│  │  └─ libxml2 >=2.12.7,<3.0a0  with the potential options
│  │     ├─ libxml2 2.12.7, which can be installed (as previously explained);
│  │     ├─ libxml2 [2.12.6|2.12.7], which can be installed (as previously explained);
│  │     └─ libxml2 2.12.7, which cannot be installed (as previously explained);
│  ├─ idyntree 12.3.0 would require
│  │  ├─ libxml2 >=2.12.7,<3.0a0  with the potential options
[... 528 lines cut ...]
   │     ├─ yarp-python 3.7.2 would require
   │     │  ├─ python_abi 3.8 *_pypy38_pp73, which can be installed;
   │     │  └─ yarp-cxx 3.7.2 h87c413d_9, which cannot be installed (as previously explained);
   │     ├─ yarp-python 3.7.2 would require
   │     │  ├─ python_abi 3.9.* *_cp39, which can be installed;
   │     │  └─ yarp-cxx 3.7.2 h87c413d_9, which cannot be installed (as previously explained);
   │     └─ yarp-python 3.7.2 would require
   │        ├─ python_abi 3.9 *_pypy39_pp73, which can be installed;
   │        └─ yarp-cxx 3.7.2 h87c413d_9, which cannot be installed (as previously explained);
   ├─ yarp 3.8.0 would require
   │  └─ yarp-cxx 3.8.0 h87c413d_0, which cannot be installed (as previously explained);
   ├─ yarp 3.8.0 would require
   │  └─ yarp-cxx [3.8.0 h5ee7595_1|3.8.0 h5ee7595_2], which can be installed (as previously explained);
   ├─ yarp [3.8.0|3.8.1] would require
   │  └─ yarp-cxx [3.8.0 h245b40a_3|3.8.1 h245b40a_0|3.8.1 h245b40a_1|3.8.1 h245b40a_2|3.8.1 h245b40a_3], which can be installed (as previously explained);
   ├─ yarp 3.8.1 would require
   │  └─ yarp-cxx 3.8.1 h264dbdb_4, which can be installed (as previously explained);
   ├─ yarp [3.8.1|3.9.0] would require
[... cut ...]
   │  └─ yarp-cxx [3.8.1 h57bc1a5_6|3.8.1 h57bc1a5_7|3.8.1 hc95d4b6_5], which can be installed (as previously explained);
   └─ yarp 3.9.0 would require
      └─ libyarp [3.9.0 h9fe3f26_4|3.9.0 ha614a09_1|3.9.0 ha614a09_2|3.9.0 he3b8073_3|3.9.0 hf094a1e_5], which can be installed (as previously explained).
~~~

There is something fishy here. Unfortunately there are a few issue intersecting (pybind11 2.11.0 bugs, numpy 2.0 migration happening with pybind11 2.12.0 required, robostack not migrated). I need to look this this evening.

traversaro, 2024-09-10:
Argh:

~~~
├─ manif 0.0.5  is installable and it requires
│  └─ gtest >=1.15.2,<1.15.3.0a0 , which can be installed;
~~~

traversaro, 2024-09-10:
The test hangs on mumps, see the stack:

~~~
0x00007ffff7d79adf in __GI___clock_nanosleep (clock_id=clock_id@entry=0, flags=flags@entry=0,
    req=req@entry=0x7fffffff0000, rem=rem@entry=0x0) at ../sysdeps/unix/sysv/linux/clock_nanosleep.c:78
#1  0x00007ffff7d86a27 in __GI___nanosleep (req=req@entry=0x7fffffff0000, rem=rem@entry=0x0)
    at ../sysdeps/unix/sysv/linux/nanosleep.c:25
#2  0x00007ffff7db672c in usleep (useconds=<optimized out>) at ../sysdeps/posix/usleep.c:31
#3  0x00007fff44fdde65 in __dmumps_fac_omp_m_MOD_dmumps_perform_copies ()
   from /home/traversaro/miniforge3/envs/conda-forge/lib/python3.11/site-packages/idyntree/../../../././libdmumps_seq.so
#4  0x00007fff44fe292f in __dmumps_fac_omp_m_MOD_dmumps_fac_l0_omp._omp_fn.0 ()
   from /home/traversaro/miniforge3/envs/conda-forge/lib/python3.11/site-packages/idyntree/../../../././libdmumps_seq.so
#5  0x00007fffef384cac in GOMP_parallel (
    fn=0x7fff44fe2060 <__dmumps_fac_omp_m_MOD_dmumps_fac_l0_omp._omp_fn.0>, data=0x7fffffff0a40,
    num_threads=18, flags=0) at ../../../libgomp/parallel.c:178
#6  0x00007fff44fdff9d in __dmumps_fac_omp_m_MOD_dmumps_fac_l0_omp ()
   from /home/traversaro/miniforge3/envs/conda-forge/lib/python3.11/site-packages/idyntree/../../../././libdmumps_seq.so
#7  0x00007fff450e52fc in dmumps_fac_b_ ()
   from /home/traversaro/miniforge3/envs/conda-forge/lib/python3.11/site-packages/idyntree/../../../././libdmumps_seq.so
#8  0x00007fff450a46e1 in dmumps_fac_driver_ ()
   from /home/traversaro/miniforge3/envs/conda-forge/lib/python3.11/site-packages/idyntree/../../../././libdmumps_seq.so
#9  0x00007fff45136111 in dmumps_ ()
   from /home/traversaro/miniforge3/envs/conda-forge/lib/python3.11/site-packages/idyntree/../../../././libdmumps_seq.so
#10 0x00007fff4513c809 in dmumps_f77_ ()
   from /home/traversaro/miniforge3/envs/conda-forge/lib/python3.11/site-packages/idyntree/../../../././libdmumps_seq.so
~~~

the update from 5.7.3=*_2 to 5.7.3=*_3 broke something.

traversaro, 2024-09-10:
Cool, setting `OMP_NUM_THREADS=1` fix the test hanging.

GiulioRomualdi, 2024-09-23:
Fixed by https://github.com/ami-iit/bipedal-locomotion-framework/pull/884. Thank you @traversaro

closed 2024-09-23 by GiulioRomualdi