gbionics/bipedal-locomotion-framework #885: Error with python bindings

xela-95, 2024-09-12:
I have an issue quite similar to the one had by @CarlottaSartore in #629 .

I have a pixi environment created from the following `pixi.toml`:

```toml
[project]
authors = ["Alessandro Croci <alessandro.croci@iit.it>"]
channels = ["conda-forge"]
description = "Add a short description here"
name = "component_parallel-differentiable-simulations"
platforms = ["linux-64"]
version = "0.1.0"

[tasks]

[dependencies]
bipedal-locomotion-framework = "*"
jaxsim = "*"
casadi = "==3.6.5"
numpy = "*"
mujoco = "*"
idyntree = "*"
adam-robotics = "*"
liecasadi = "*"
loop-rate-limiters = "*"
ipython = "*"
resolve-robotics-uri-py = "*"
ergocub-software = "*"
python = "*"
```

When I try to import the blf python bindings via
```python
import bipedal_locomotion_framework.bindings as blf
```

I get the following error:
```
Traceback (most recent call last):
  File "/home/acroci/repos/component_parallel-differentiable-simulations/src/debug.py", line 1, in <module>
    import bipedal_locomotion_framework.bindings as blf
  File "/home/acroci/repos/component_parallel-differentiable-simulations/.pixi/envs/default/lib/python3.12/site-packages/bipedal_locomotion_framework/__init__.py", line 1, in <module>
    from .bindings import *
ImportError: arg(): could not convert default argument 'mixed_velocity: manif::SE3Tangent<double>' in method '<class 'bipedal_locomotion_framework.bindings.ik.SE3Task'>.set_set_point' into a Python object (type not registered yet?)
```

The versions of `blf` and and `manif` are:
```
bipedal-locomotion-framework          0.18.0        ha770c72_15                  9.1 KiB     conda  bipedal-locomotion-framework-0.18.0-ha770c72_15.conda
bipedal-locomotion-framework-python   0.18.0        py312h6834624_15             595.6 KiB   conda  bipedal-locomotion-framework-python-0.18.0-py312h6834624_15.conda
libbipedal-locomotion-framework       0.18.0        h7f52459_15                  1.9 MiB     conda  libbipedal-locomotion-framework-0.18.0-h7f52459_15.conda
```

```
manif                                 0.0.5         h5888daf_2                   63.3 KiB    conda  manif-0.0.5-h5888daf_2.conda
manifpy                               0.0.5         py312h39a7861_2              580.5 KiB   conda  manifpy-0.0.5-py312h39a7861_2.conda
```

CC @traversaro

xela-95, 2024-09-12:
Here's the `pixi.lock` file:

<details><summary>Details</summary>
<p>

```
version: 5
environments:
  default:
    channels:
    - url: https://conda.anaconda.org/conda-forge/
    packages:
      linux-64:
      - conda: https://conda.anaconda.org/conda-forge/linux-64/_libgcc_mutex-0.1-conda_forge.tar.bz2
      - conda: https://conda.anaconda.org/conda-forge/linux-64/_openmp_mutex-4.5-2_gnu.tar.bz2
      - conda: https://conda.anaconda.org/conda-forge/noarch/absl-py-2.1.0-pyhd8ed1ab_0.conda
      - conda: https://conda.anaconda.org/conda-forge/linux-64/ace-8.0.0-hac33072_1.conda
      - conda: https://conda.anaconda.org/conda-forge/noarch/adam-robotics-0.3.0-pyh780aa2d_1.conda
      - conda: https://conda.anaconda.org/conda-forge/noarch/aiohappyeyeballs-2.4.0-pyhd8ed1ab_0.conda
      - conda: https://conda.anaconda.org/conda-forge/linux-64/aiohttp-3.10.5-py312h66e93f0_1.conda
      - conda: https://conda.anaconda.org/conda-forge/noarch/aiosignal-1.3.1-pyhd8ed1ab_0.tar.bz2
      - conda: https://conda.anaconda.org/conda-forge/linux-64/alsa-lib-1.2.12-h4ab18f5_0.conda
      - conda: https://conda.anaconda.org/conda-forge/linux-64/ampl-mp-3.1.0-h2cc385e_1006.tar.bz2
      - conda: https://conda.anaconda.org/conda-forge/linux-64/aom-3.9.1-hac33072_0.conda
      - conda: https://conda.anaconda.org/conda-forge/linux-64/assimp-5.4.2-h2334b4d_0.conda
      - conda: https://conda.anaconda.org/conda-forge/noarch/asttokens-2.4.1-pyhd8ed1ab_0.conda
      - conda: https://conda.anaconda.org/conda-forge/linux-64/attr-2.5.1-h166bdaf_1.tar.bz2
      - conda: https://conda.anaconda.org/conda-forge/noarch/attrs-24.2.0-pyh71513ae_0.conda
      - conda: https://conda.anaconda.org/conda-forge/linux-64/bipedal-locomotion-framework-0.18.0-ha770c72_15.conda
      - conda: https://conda.anaconda.org/conda-forge/linux-64/bipedal-locomotion-framework-python-0.18.0-py312h6834624_15.conda
      - conda: https://conda.anaconda.org/conda-forge/linux-64/blosc-1.21.6-hef167b5_0.conda
      - conda: https://conda.anaconda.org/conda-forge/linux-64/brotli-1.1.0-hb9d3cd8_2.conda
      - conda: https://conda.anaconda.org/conda-forge/linux-64/brotli-bin-1.1.0-hb9d3cd8_2.conda
      - conda: https://conda.anaconda.org/conda-forge/linux-64/bzip2-1.0.8-h4bc722e_7.conda
      - conda: https://conda.anaconda.org/conda-forge/linux-64/c-ares-1.33.1-heb4867d_0.conda
      - conda: https://conda.anaconda.org/conda-forge/linux-64/ca-certificates-2024.8.30-hbcca054_0.conda
      - conda: https://conda.anaconda.org/conda-forge/linux-64/cairo-1.18.0-hbb29018_2.conda
      - conda: https://conda.anaconda.org/conda-forge/linux-64/casadi-3.6.5-py312h1a19fd5_4.conda
      - conda: https://conda.anaconda.org/conda-forge/noarch/certifi-2024.8.30-pyhd8ed1ab_0.conda
      - conda: https://conda.anaconda.org/conda-forge/linux-64/cli11-2.4.1-h59595ed_0.conda
      - conda: https://conda.anaconda.org/conda-forge/noarch/colorama-0.4.6-pyhd8ed1ab_0.tar.bz2
      - conda: https://conda.anaconda.org/conda-forge/noarch/coloredlogs-15.0.1-pyhd8ed1ab_3.tar.bz2
      - conda: https://conda.anaconda.org/conda-forge/linux-64/console_bridge-1.0.2-h924138e_1.tar.bz2
      - conda: https://conda.anaconda.org/conda-forge/linux-64/contourpy-1.3.0-py312h68727a3_1.conda
      - conda: https://conda.anaconda.org/conda-forge/linux-64/cppad-20240000.6-he02047a_0.conda
      - conda: https://conda.anaconda.org/conda-forge/noarch/cycler-0.12.1-pyhd8ed1ab_0.conda
      - conda: https://conda.anaconda.org/conda-forge/linux-64/dav1d-1.2.1-hd590300_0.conda
      - conda: https://conda.anaconda.org/conda-forge/linux-64/dbus-1.13.6-h5008d03_3.tar.bz2
      - conda: https://conda.anaconda.org/conda-forge/noarch/decorator-5.1.1-pyhd8ed1ab_0.tar.bz2
      - conda: https://conda.anaconda.org/conda-forge/noarch/docstring_parser-0.16-pyhd8ed1ab_0.conda
      - conda: https://conda.anaconda.org/conda-forge/linux-64/double-conversion-3.3.0-h59595ed_0.conda
      - conda: https://conda.anaconda.org/conda-forge/linux-64/eigen-3.4.0-h00ab1b0_0.conda
      - conda: https://conda.anaconda.org/conda-forge/linux-64/elfutils-0.191-h924a536_0.conda
      - conda: https://conda.anaconda.org/conda-forge/linux-64/ergocub-software-0.7.5-ha770c72_0.conda
      - conda: https://conda.anaconda.org/conda-forge/noarch/etils-1.9.4-pyhd8ed1ab_0.conda
      - conda: https://conda.anaconda.org/conda-forge/noarch/exceptiongroup-1.2.2-pyhd8ed1ab_0.conda
      - conda: https://conda.anaconda.org/conda-forge/noarch/executing-2.1.0-pyhd8ed1ab_0.conda
      - conda: https://conda.anaconda.org/conda-forge/linux-64/expat-2.6.3-h5888daf_0.conda
      - conda: https://conda.anaconda.org/conda-forge/linux-64/ffmpeg-7.0.1-gpl_h3589c0a_102.conda
      - conda: https://conda.anaconda.org/conda-forge/noarch/filelock-3.16.0-pyhd8ed1ab_0.conda
      - conda: https://conda.anaconda.org/conda-forge/linux-64/flann-1.9.2-h54ed35b_1.conda
      - conda: https://conda.anaconda.org/conda-forge/linux-64/fmt-11.0.2-h434a139_0.conda
      - conda: https://conda.anaconda.org/conda-forge/noarch/font-ttf-dejavu-sans-mono-2.37-hab24e00_0.tar.bz2
      - conda: https://conda.anaconda.org/conda-forge/noarch/font-ttf-inconsolata-3.000-h77eed37_0.tar.bz2
      - conda: https://conda.anaconda.org/conda-forge/noarch/font-ttf-source-code-pro-2.038-h77eed37_0.tar.bz2
      - conda: https://conda.anaconda.org/conda-forge/noarch/font-ttf-ubuntu-0.83-h77eed37_2.conda
[... cut ...]
  url: https://conda.anaconda.org/conda-forge/linux-64/proxsuite-0.6.7-py312h68727a3_1.conda
  sha256: a40cf5455649bbcc2f378fb4ddbd7c4a4fffeffef7ec4471114b8252f4038f77
  md5: 30aeb56283f7ad795d145983e86de4c5
  depends:
  - __glibc >=2.17,<3.0.a0
  - eigen
  - libgcc >=13
  - libstdcxx >=13
  - numpy
  - python >=3.12,<3.13.0a0
  - python_abi 3.12.* *_cp312
  - scipy
  - simde
  license: BSD-2-Clause
  license_family: BSD
  size: 1654819
  timestamp: 1725915829070
- kind: conda
  name: pthread-stubs
  version: '0.4'
  build: h36c

traversaro, 2024-09-12:
Unfortunately you hit https://github.com/conda-forge/pybind11-feedstock/issues/77 . In a nutshell, pybind11 compiled bindings have a hidden dependency (not handled by conda) on the major GCC version used to compile them, and you are mixing blf compiled with GCC 12 with manif compiled with GCC 13. There is any reason you need casadi 3.6.5 and 3.6.6 is not ok? If indeed you need casadi 3.6.5, you can try with earlier build of manifpy until it works (I am on a mobile so I can't easily check which version of manifpy was compiled with GCC 12 and which one by GCC 13.

traversaro, 2024-09-13:
Ok, I have access to a laptop now. 

This are the GCC versions used by blf

| blf version |  GCC version | Reference |
|:----------:|:-------------:|:-----------:|
| 0.19.0=*_0 |  13  | https://github.com/conda-forge/bipedal-locomotion-framework-feedstock/pull/81 | 
| 0.18.0=*_16 | 13 |  https://github.com/conda-forge/bipedal-locomotion-framework-feedstock/pull/79 |
| 0.18.0=*_15 | 12 |  https://github.com/conda-forge/bipedal-locomotion-framework-feedstock/pull/79 |

These the one used by manifpy:

| manifpy version |  GCC version | Reference |
|:----------:|:-------------:|:-----------:|
| 0.0.5=*_2 |  13  | https://github.com/conda-forge/manif-feedstock/pull/50/files | 
| 0.0.5=*_1 | 13 |  https://github.com/conda-forge/manif-feedstock/pull/48 |
| 0.0.5=*_0 | 12 |  https://github.com/conda-forge/manif-feedstock/pull/48 |

So, if you need to use casadi 3.5.5, please pin manifpy to 0.0.5=*_0 .

xela-95, 2024-09-13:
Thank you so much @traversaro, pinning manifpy to 0.0.5=*_0 solved the issue!

Just as a side note, I was able to configure a conda environment by pinning the manifpy version just using conda, not using pixi since I was not able to pin the build version (either from CLI or updating `pixi.toml`).

traversaro, 2024-09-13:
> Just as a side note, I was able to configure a conda environment by pinning the manifpy version just using conda, not using pixi since I was not able to pin the build version (either from CLI or updating pixi.toml).

I guess something like: 

~~~
manifpy = {version = "0.0.5", build = "*_0"}
~~~

 Should work on pixi?

xela-95, 2024-09-13:
> I guess something like:
> 
> ```
> manifpy = {version = "0.0.5", build = "*_0"}
> ```
> 
> Should work on pixi?

Yep that syntax put in the `pixi.toml` worked!