gbionics/bipedal-locomotion-framework #900: CentroidalMPC.advance() gets stuck with Ipopt and Mumps

xela-95, 2024-10-07:
Hi, I'm using [comodo](https://github.com/ami-iit/comodo) to perform simulations on JaxSim, since it provides an easy to use interface to blf controllers. 

However I'm having some flaky errors (not repeating 100% of the times) in which the BipedalLocomotion::ReducedModelControllers::CentroidalMPC::[advance](https://ami-iit.github.io/bipedal-locomotion-framework/classBipedalLocomotion_1_1ReducedModelControllers_1_1CentroidalMPC.html#a769ca3af8290ee3a847226534013153b) method gets stuck and do not exit anymore (even with ctrl-C). When i go debugging those methods it just logs:

```
******************************************************************************
This program contains Ipopt, a library for large-scale nonlinear optimization.
 Ipopt is released as open source code under the Eclipse Public License (EPL).
         For more information visit https://github.com/coin-or/Ipopt
******************************************************************************
```

That method is called in `comodo` wrapper of the centroidal MPC `centroidalMPC.plan_trajectory()`: https://github.com/ami-iit/comodo/blob/1130009d6ef0cfa1944495209a15887388faa52d/src/comodo/centroidalMPC/centroidalMPC.py#L36-L57

Have you ever encountered this kind of issue before? What should I try to change?

### System info

| Library | Version |
|--------|--------|
| casadi | 3.6.5 |
| ipopt | 3.14.16 |
| mumps | 5.7.3 | 

CC @CarlottaSartore

traversaro, 2024-10-07:
Do collect a bit more information, probably you can attach a debugger to the Python process and get a backtrace of the thread that is hanging, to get a bit more context of what is happening. What is the usage of CPU cores? Can you report the full conda environment you are using? 

Can you also try to set `OMP_NUM_THREADS=1` ? We recently had a mumps packaging regression that seems similar to your problem, see https://github.com/conda-forge/mumps-feedstock/issues/125 and the linked issues.

traversaro, 2024-10-07:
Also https://github.com/conda-forge/mumps-feedstock/pull/126 seems related.

xela-95, 2024-10-07:
Thanks a lot @traversaro !! Setting `OMP_NUM_THREADS=1` seems to have solved the issue! I will monitor the situation and if this problem is still present I will provide further details, otherwise I will close this issue. 

> What is the usage of CPU cores?

When the code hangs, the CPU usage is 0%, so a deadlock is likely the issue.

> Can you report the full conda environment you are using?

I'm using pixi, below you can find my environment:

<details><summary>pixi list</summary>
<p>

```
Package                               Version             Build                        Size        Kind   Source
_libgcc_mutex                         0.1                 conda_forge                  2.5 KiB     conda  _libgcc_mutex-0.1-conda_forge.tar.bz2
_openmp_mutex                         4.5                 2_gnu                        23.1 KiB    conda  _openmp_mutex-4.5-2_gnu.tar.bz2
absl-py                               2.1.0               pyhd8ed1ab_0                 104.8 KiB   conda  absl-py-2.1.0-pyhd8ed1ab_0.conda
ace                                   8.0.1               he02047a_0                   5.2 MiB     conda  ace-8.0.1-he02047a_0.conda
adam-robotics                         0.3.0               pyh780aa2d_1                 43.6 KiB    conda  adam-robotics-0.3.0-pyh780aa2d_1.conda
aiohappyeyeballs                      2.4.3               pyhd8ed1ab_0                 18.8 KiB    conda  aiohappyeyeballs-2.4.3-pyhd8ed1ab_0.conda
aiohttp                               3.10.8              py310ha75aee5_0              708.4 KiB   conda  aiohttp-3.10.8-py310ha75aee5_0.conda
aiosignal                             1.3.1               pyhd8ed1ab_0                 12.4 KiB    conda  aiosignal-1.3.1-pyhd8ed1ab_0.tar.bz2
alembic                               1.13.3              pyhd8ed1ab_0                 155.9 KiB   conda  alembic-1.13.3-pyhd8ed1ab_0.conda
alsa-lib                              1.2.12              h4ab18f5_0                   542.8 KiB   conda  alsa-lib-1.2.12-h4ab18f5_0.conda
ampl-mp                               3.1.0               h2cc385e_1006                1.1 MiB     conda  ampl-mp-3.1.0-h2cc385e_1006.tar.bz2
anyio                                 4.6.0               pyhd8ed1ab_1                 105.9 KiB   conda  anyio-4.6.0-pyhd8ed1ab_1.conda
aom                                   3.9.1               hac33072_0                   2.6 MiB     conda  aom-3.9.1-hac33072_0.conda
argon2-cffi                           23.1.0              pyhd8ed1ab_0                 18.2 KiB    conda  argon2-cffi-23.1.0-pyhd8ed1ab_0.conda
argon2-cffi-bindings                  21.2.0              py310ha75aee5_5              33.6 KiB    conda  argon2-cffi-bindings-21.2.0-py310ha75aee5_5.conda
arrow                                 1.3.0               pyhd8ed1ab_0                 97.7 KiB    conda  arrow-1.3.0-pyhd8ed1ab_0.conda
assimp                                5.4.2               h2334b4d_0                   3.3 MiB     conda  assimp-5.4.2-h2334b4d_0.conda
asttokens                             2.4.1               pyhd8ed1ab_0                 28.2 KiB    conda  asttokens-2.4.1-pyhd8ed1ab_0.conda
async-lru                             2.0.4               pyhd8ed1ab_0                 15 KiB      conda  async-lru-2.0.4-pyhd8ed1ab_0.conda
async-timeout                         4.0.3               pyhd8ed1ab_0                 11.1 KiB    conda  async-timeout-4.0.3-pyhd8ed1ab_0.conda
attr                                  2.5.1               h166bdaf_1                   69.4 KiB    conda  attr-2.5.1-h166bdaf_1.tar.bz2
attrs                                 24.2.0              pyh71513ae_0                 54.7 KiB    conda  attrs-24.2.0-pyh71513ae_0.conda
babel                                 2.14.0              pyhd8ed1ab_0                 7.3 MiB     conda  babel-2.14.0-pyhd8ed1ab_0.conda
beautifulsoup4                        4.12.3              pyha770c72_0                 115.4 KiB   conda  beautifulsoup4-4.12.3-pyha770c72_0.conda
bipedal-locomotion-framework          0.18.0              ha770c72_15                  9.1 KiB     conda  bipedal-locomotion-framework-0.18.0-ha770c72_15.conda
bipedal-locomotion-framework-python   0.18.0              py310h248966e_15             602.8 KiB   conda  bipedal-locomotion-framework-python-0.18.0-py310h248966e_15.conda
black                                 24.8.0                                           5.1 MiB     pypi   black-24.8.0-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.manylinux_2_28_x86_64.http.whl
bleach                                6.1.0               pyhd8ed1ab_0                 128.1 KiB   conda  bleach-6.1.0-pyhd8ed1ab_0.conda
blis                                  0.9.0               h4ab18f5_2                   3.4 MiB     conda  blis-0.9.0-h4ab18f5_2.conda
blosc                                 1.21.6              hef167b5_0                   47.7 KiB    conda  blosc-1.21.6-hef167b5_0.conda
brotli                                1.1.0               hb9d3cd8_2                   18.8 KiB    conda  brotli-1.1.0-hb9d3cd8_2.conda
brotli-bin                            1.1.0               hb9d3cd8_2                   18.4 KiB    conda  brotli-bin-1.1.0-hb9d3cd8_2.conda
brotli-python                         1.1.0               py310hf71b8c6_2              341.5 KiB   conda  brotli-python-1.1.0-py310hf71b8c6_2.conda
bzip2                                 1.0.8               h4bc722e_7                   246.9 KiB   conda  bzip2-1.0.8-h4bc722e_7.conda
[... cut ...]
zipp                                  3.20.2              pyhd8ed1ab_0                 20.9 KiB    conda  zipp-3.20.2-pyhd8ed1ab_0.conda
zlib                                  1.3.1               h4ab18f5_1                   90.8 KiB    conda  zlib-1.3.1-h4ab18f5_1.conda
zstandard                             0.23.0              py310ha39cb0e_1              398.7 KiB   conda  zstandard-0.23.0-py310ha39cb0e_1.conda
zstd                                  1.5.6               ha6fb4c9_0                   541.8 KiB   conda  zstd-1.5.6-ha6fb4c9_0.conda

```

</p>
</details>

traversaro, 2024-10-07:
> I will monitor the situation and if this problem is still present I will provide further details, otherwise I will close this issue.

The `OMP_NUM_THREADS=1` is just a workaround, the issue should not be there in the first place. There are two interesting things that emerges from your `pixi list`:
* Somehow you are using the `blis` implementation of BLAS, that seems to have the same problem of openblas that was solved in https://github.com/conda-forge/mumps-feedstock/pull/126 for openblas, but not for blis.
* Why the solved selected blis as the blas implementation? I wonder if it is as it can't satisfy the constraint added in https://github.com/conda-forge/mumps-feedstock/pull/126 . Can you please provide the `pixi.toml` that generated that environment? Thanks!

traversaro, 2024-10-07:
By the way, I tried to reproduce the issue with https://raw.githubusercontent.com/ami-iit/bipedal-locomotion-framework/da3d948990fdab29daf8da9ccb90fb93a093804b/bindings/python/ReducedModelControllers/tests/test_centroidal_mpc.py failing, so if you can provide an easy way to reproduce the issue it would be great.

xela-95, 2024-10-07:
> * Why the solved selected blis as the blas implementation? I wonder if it is as it can't satisfy the constraint added in [openmp builds require openblas openmp builds conda-forge/mumps-feedstock#126](https://github.com/conda-forge/mumps-feedstock/pull/126) . Can you please provide the `pixi.toml` that generated that environment? Thanks!

Here is the `pixi.toml`:

```toml
[project]
authors = ["Carlotta Sartore <carlotta.sartore@iit.it>"]
channels = ["conda-forge"]
description = "Suite of parametrized controller with simulation environments for co-design of Humanoid Robots"
name = "comodo"
platforms = ["linux-64"]
version = "0.0.0"
repository = "https://github.com/ami-iit/comodo"
readme = "README.md"
license-file = "LICENSE"

[activation]
# To avoid issue with git-lfs, see https://github.com/prefix-dev/pixi/issues/2000
env = { GIT_LFS_SKIP_SMUDGE = "1" }

[dependencies]
# bipedal-locomotion-framework and manifpy have pinned build dependencies
# due to https://github.com/ami-iit/bipedal-locomotion-framework/issues/885
bipedal-locomotion-framework = { version = "==0.18.0", build = "ha770c72_15" }
manifpy = { version = "==0.0.5", build = "py310hf5d888b_0" }
idyntree = "*"
adam-robotics = "*"
mujoco = "*"
mujoco-python-viewer = "*"
matplotlib = ">=3.9.2,<4"
urllib3 = "*"
urchin = "*"
resolve-robotics-uri-py = "*"
notebook = "*"
rod = "*"
ipykernel = "*"
mesalib = "*"
optuna = ">=4.0.0,<5"
cmaes = ">=0.11.1,<0.12"
plotly = ">=5.24.1,<6"
casadi = "<=3.6.6"
nbformat = ">=5.10.4,<6"
python-kaleido = ">=0.2.1,<0.3"
scikit-learn = ">=1.5.2,<2"
# jaxsim = ">=0.4.2"

[pypi-dependencies]
jaxsim = { git = "https://github.com/ami-iit/jaxsim.git", rev="main", extras = ["all"]}
# jaxsim = { path = "../jaxsim", extras = ["all"] }
urdfmodifiers = { git = "https://github.com/CarlottaSartore/urdf-modifiers.git", rev = "scalar_modification" }
comodo = { path = "./", editable = true }
````

And here the output of `pixi info`:

```
pixi info                                                                                                              1 ✘  38s   13:08:03  
      Pixi version: 0.30.0
          Platform: linux-64
  Virtual packages: __unix=0=0
                  : __linux=6.8.0=0
                  : __glibc=2.39=0
                  : __cuda=12.4=0
                  : __archspec=1=skylake
         Cache dir: /home/acroci/.cache/rattler/cache
      Auth storage: /home/acroci/.rattler/credentials.json
  Config locations: No config files found

Project
------------
              Name: comodo
           Version: 0.0.0
     Manifest file: /home/acroci/repos/comodo/pixi.toml
      Last updated: 07-10-2024 13:07:27

Environments
------------
       Environment: default
          Features: default
          Channels: conda-forge
  Dependency count: 21
      Dependencies: bipedal-locomotion-framework, manifpy, idyntree, adam-robotics, mujoco, mujoco-python-viewer, matplotlib, urllib3, urchin, resolve-robotics-uri-py, notebook, rod, ipykernel, mesalib, optuna, cmaes, plotly, casadi, nbformat, python-kaleido, scikit-learn
 PyPI Dependencies: jaxsim, urdfmodifiers, comodo
  Target platforms: linux-64

```

traversaro, 2024-10-07:
> By the way, I tried to reproduce the issue with [...]

Sorry, this was not clear perhaps. The reproduction attempt was "failing" as the test was passing and so I "failed" in reproducing the issue.

xela-95, 2024-10-07:
> Sorry, this was not clear perhaps. The reproduction attempt was "failing" as the test was passing and so I "failed" in reproducing the issue.

Sure, this is the script I'm testing right now: https://github.com/xela-95/comodo/blob/viscoelastic-simulations/examples/tune_jaxsim_contact_model.py

In order to reproduce the issue just comment out https://github.com/xela-95/comodo/blob/8510ead3e1f3c3361c3852134dfeeb39025bdd2a/examples/tune_jaxsim_contact_model.py#L29 and run it. I usually run it with: `pixi run python tune_jaxsim_contact_model.py --trials 10`