gbionics/bipedal-locomotion-framework #912: YarpSensorBridge: Drop use of IGenericSensor and IAnalogSensor

traversaro, 2024-11-15:
Now that all the IMUs and FTs in the iCub and ergoCub robots we use have been migrated to multipleAnalogSensors interfaces, we can drop the use of IGenericSensor and IAnalogSensor in the YarpSensorBridge. Similarly to https://github.com/ami-iit/bipedal-locomotion-framework/issues/911, we could take the occasion to just drop any custom remapping logic from YarpSensorBridge, and use use a `multipleanalogsensorsremapper`.

GiulioRomualdi, 2024-11-15:
I agree! It will simplify the logic!

traversaro, 2024-11-15:
I renamed the title as I realized that we are also using IGenericSensors to read the cartesian wrenches, and that info is not published by WBD with a usual chain of C++ device interface ---> NWS ---> NWC , so let's leave them aside for now. I wonder if we could just expose those using the existing https://www.yarp.it/v3.4/classyarp_1_1dev_1_1ISixAxisForceTorqueSensors.html inteface. If you think about it, in the end external wrenches estimates could be tought of virtual FT sensors, so that could make sense, but that is OT here.