gbionics/bipedal-locomotion-framework #929: Failed to configure due to ``rclcpp``

S-Dafarra, 2025-01-31:
In a shared machine where ROS is used, we had some difficulties in configuring the project with the following error

[image]

In particular, the error is related to not finding the  ``iceoryx_binding_c`` project, which is searched by ``rclcpp`` in https://github.com/ami-iit/bipedal-locomotion-framework/blob/f4d03f581072a1287ed3d769a5c1fb119d12007d/cmake/BipedalLocomotionFrameworkDependencies.cmake#L31

I solved it temporarily by commenting out the line above

cc @msecode who found the issue

cc @GiulioRomualdi

traversaro, 2025-01-31:
So it was not sufficient to set `FRAMEWORK_USE_rclcpp` to `OFF`? That is strange. Probably some problem upstream in ROS. Anyhow, didn't we agree to be able to comment out the ROS inclusion in ergocub-torso's setup.sh ?

S-Dafarra, 2025-01-31:
> So it was not sufficient to set `FRAMEWORK_USE_rclcpp` to `OFF`?

Indeed, that option was not even appearing because it was failing first.

> Anyhow, didn't we agree to be able to comment out the ROS inclusion in ergocub-torso's setup.sh ?

I believe so, maybe there were some leftover

traversaro, 2025-01-31:
> > So it was not sufficient to set `FRAMEWORK_USE_rclcpp` to `OFF`?
> 
> Indeed, that option was not even appearing because it was failing first.

I proposed a mitigation for this in https://github.com/ami-iit/bipedal-locomotion-framework/pull/930 . Note that it would still fail, but it would not fail after explicitly setting `-DFRAMEWORK_USE_rclcpp:BOOL=OFF` in the CMake invocation.