gbionics/bipedal-locomotion-framework #943: Logged joint positions are automatically converted to radians.

FabioBergonti, 2025-02-27:
Before logging, joint positions are automatically converted from degrees to radians.

https://github.com/ami-iit/bipedal-locomotion-framework/blob/b7ee4915699d9f07a8a5a5fee6eb687845912297/src/RobotInterface/YarpImplementation/src/YarpRobotControl.cpp#L238-L266

This can be problematic if we want to use linear actuators!
It would be great if there were an option to specify which joints should be scaled by π/180.

traversaro, 2025-02-27:
The YARP controlboard interfaces exposes the joint type, so we could use that to decide if convert from degrees to rad or not.

traversaro, 2025-02-27:
See https://github.com/robotology/yarp-devices-ros2/blob/39f8fb95e81273fe8ab586354df773e4234ae046/src/devices/controlBoard_nws_ros2/ControlBoard_nws_ros2.cpp#L406-L413 for a related piece of code in the ROS 2 devices, that also need to convert degrees to radians but just for revolute joints.

GiulioRomualdi, 2025-03-10:
Fixed by #949

closed 2025-03-10 by GiulioRomualdi