gbionics/bipedal-locomotion-framework #944: Conda forge CI failure March

GiulioRomualdi, 2025-03-06:
The CI fails with the following error

``` 
 369 | inline fmt::basic_string_view<T> to_string_view(fmt::basic_format_string<T, Args...> fmt) {
      |                                                      ^~~~~~~~~~~~~~~~~~~
/usr/share/miniconda3/envs/test/include/spdlog/common.h:350:45: note: previous declaration 'constexpr spdlog::string_view_t spdlog::details::to_string_view(spdlog::string_view_t)'
  350 | SPDLOG_CONSTEXPR_FUNC spdlog::string_view_t to_string_view(spdlog::string_view_t str)
      |                                             ^~~~~~~~~~~~~~
/usr/share/miniconda3/envs/test/include/spdlog/common.h:369:54: error: 'basic_format_string' is not a member of 'fmt'
  369 | inline fmt::basic_string_view<T> to_string_view(fmt::basic_format_string<T, Args...> fmt) {
      |                                                      ^~~~~~~~~~~~~~~~~~~
/usr/share/miniconda3/envs/test/include/spdlog/common.h:369:75: error: expected primary-expression before ',' token
  369 | inline fmt::basic_string_view<T> to_string_view(fmt::basic_format_string<T, Args...> fmt) {
      |                                                                           ^
/usr/share/miniconda3/envs/test/include/spdlog/common.h:369:81: error: expected primary-expression before '...' token
  369 | inline fmt::basic_string_view<T> to_string_view(fmt::basic_format_string<T, Args...> fmt) {
      |                                                                                 ^~~
```

GiulioRomualdi, 2025-03-06:
This is the first CI with the error https://github.com/ami-iit/component_ifeel/actions/runs/13701592879/job/38316549128?pr=2010 actually 13 hours ago

GiulioRomualdi, 2025-03-06:
This is the difference between working CI https://github.com/ami-iit/bipedal-locomotion-framework/actions/runs/13667162197/job/38210583745 and not working CI https://github.com/ami-iit/bipedal-locomotion-framework/actions/runs/13689823797/job/38280885996#step:10:124

```diff
8c8
< aiohappyeyeballs          2.4.6              pyhd8ed1ab_0    conda-forge
---
> aiohappyeyeballs          2.4.8              pyhd8ed1ab_0    conda-forge
47c47
< fmt                       11.0.2               h434a139_0    conda-forge
---
> fmt                       11.1.0               h07f6e7f_0    conda-forge
374c374
< sdl3                      3.2.6                h3083f51_0    conda-forge
---
> sdl3                      3.2.8                h3083f51_0    conda-forge
```

GiulioRomualdi, 2025-03-06:
cc @dariosortino @traversaro

dariosortino, 2025-03-06:
@davidegorbani fyi

GiulioRomualdi, 2025-03-06:
Interesting issues/PRs:
- https://github.com/gabime/spdlog/issues/3302
- https://github.com/gabime/spdlog/pull/3301

davidegorbani, 2025-03-06:
CC @LudovicaDanovaro

GiulioRomualdi, 2025-03-06:
The fix appeared the first time on https://github.com/gabime/spdlog/releases/tag/v1.15.1
But in our case we have
```
    spdlog-1.14.1              |       hed91bc2_1         191 KB  conda-forge
```

GiulioRomualdi, 2025-03-06:
Let's see if https://github.com/ami-iit/bipedal-locomotion-framework/pull/945 fixes the problem

GiulioRomualdi, 2025-03-06:
Forcing spdlog to be at least v1.15.1` makes the environment not solvable 😭 
```
LibMambaUnsatisfiableError: Encountered problems while solving:
  - package ros-humble-rclcpp-16.0.2-py310h7c61026_2 requires ros-humble-rcl, but none of the providers can be installed

Could not solve for environment specs
The following packages are incompatible
├─ ros-humble-rclcpp is installable and it requires
│  └─ ros-humble-rcl, which requires
│     └─ ros-humble-rcl-logging-spdlog with the potential options
│        ├─ ros-humble-rcl-logging-spdlog 2.3.1 would require
│        │  └─ spdlog >=1.11.0,<1.12.0a0 , which can be installed;
│        ├─ ros-humble-rcl-logging-spdlog 2.3.1 would require
│        │  └─ spdlog >=1.12.0,<1.13.0a0 , which can be installed;
│        └─ ros-humble-rcl-logging-spdlog 2.3.1 would require
│           └─ spdlog >=1.14.1,<1.15.0a0 , which can be installed;
└─ spdlog >=1.15.1  is not installable because it conflicts with any installable versions previously reported.
```

traversaro, 2025-03-06:
See my comment in https://github.com/ami-iit/bipedal-locomotion-framework/pull/945#issuecomment-2704268268 . If you like the change from Humble to Jazzy, I can do it in the same PR. Otherwise the alternative is to do a rebuild of ros-humble, that is doable but takes some time.

GiulioRomualdi, 2025-03-09:
Fixed by https://github.com/ami-iit/bipedal-locomotion-framework/pull/945

closed 2025-03-09 by GiulioRomualdi